ros2/ros2 #1191: Duplicate compilation of files in ROS 2

sloretz, 2021-09-24:
Inspired by https://github.com/ros2/rosbag2/issues/872 I looked at ROS 2 files that get compiled multiple times. It looks like there's a lot of them. I opened a couple PRs for some easy ones to fix (https://github.com/ros2/rcl/pull/942 , https://github.com/ros2/rviz/pull/774).

In rosbag2's case it looks like a lot of the duplication comes from tests being created for each rmw implementation. Most of the files below look like test files to me too. Maybe splitting the ament CMake functions that create tests into two - one that creates the test executabe and another that creates the test itself could solve most of these? Only the second function would need to be run for each RMW implementation.

<details><summary>How to collect the data</summary>

```
colcon build --event-handlers console_cohesion+ --cmake-args -DCMAKE_VERBOSE_MAKEFILE=true > src/build_log_verbose_make.txt
```

```python3
import re

source_file_regex = re.compile("/home/osrf/ws/ros2/src/[^ ]+\.c(pp)?")

source_code_files = []
with open('../src/build_log_verbose_make.txt', 'r') as fin:
    for match in source_file_regex.finditer(fin.read()):
        source_code_files.append(match.group(0))

count_files = {}
for f in source_code_files:
    if f not in count_files:
        count_files[f] = 0
    count_files[f] += 1

for f, count in count_files.items():
    if count > 1:
        print(f'({count}) {f}')
```
</details>

<details><summary>List of duplicate files</summary>

```
(2) /home/osrf/ws/ros2/src/ros/urdfdom/urdf_parser/src/pose.cpp
(2) /home/osrf/ws/ros2/src/ros/urdfdom/urdf_parser/src/link.cpp
(2) /home/osrf/ws/ros2/src/ros/urdfdom/urdf_parser/src/model.cpp
(2) /home/osrf/ws/ros2/src/ros/urdfdom/urdf_parser/src/joint.cpp
(2) /home/osrf/ws/ros2/src/eclipse-cyclonedds/cyclonedds/src/core/ddsi/src/q_config.c
(2) /home/osrf/ws/ros2/src/eclipse-cyclonedds/cyclonedds/src/ddsrt/src/bswap.c
(2) /home/osrf/ws/ros2/src/eclipse-cyclonedds/cyclonedds/src/ddsrt/src/log.c
(2) /home/osrf/ws/ros2/src/eclipse-cyclonedds/cyclonedds/src/ddsrt/src/strtod.c
(2) /home/osrf/ws/ros2/src/eclipse-cyclonedds/cyclonedds/src/ddsrt/src/io.c
(2) /home/osrf/ws/ros2/src/eclipse-cyclonedds/cyclonedds/src/ddsrt/src/retcode.c
(2) /home/osrf/ws/ros2/src/eclipse-cyclonedds/cyclonedds/src/ddsrt/src/strtol.c
(2) /home/osrf/ws/ros2/src/eclipse-cyclonedds/cyclonedds/src/ddsrt/src/mh3.c
(2) /home/osrf/ws/ros2/src/eclipse-cyclonedds/cyclonedds/src/ddsrt/src/environ.c
(2) /home/osrf/ws/ros2/src/eclipse-cyclonedds/cyclonedds/src/ddsrt/src/expand_vars.c
(2) /home/osrf/ws/ros2/src/eclipse-cyclonedds/cyclonedds/src/ddsrt/src/hopscotch.c
(2) /home/osrf/ws/ros2/src/eclipse-cyclonedds/cyclonedds/src/ddsrt/src/xmlparser.c
(2) /home/osrf/ws/ros2/src/eclipse-cyclonedds/cyclonedds/src/ddsrt/src/fibheap.c
(2) /home/osrf/ws/ros2/src/eclipse-cyclonedds/cyclonedds/src/ddsrt/src/avl.c
(2) /home/osrf/ws/ros2/src/eclipse-cyclonedds/cyclonedds/src/ddsrt/src/circlist.c
(2) /home/osrf/ws/ros2/src/eclipse-cyclonedds/cyclonedds/src/ddsrt/src/atomics.c
(2) /home/osrf/ws/ros2/src/eclipse-cyclonedds/cyclonedds/src/ddsrt/src/cdtors.c
(2) /home/osrf/ws/ros2/src/eclipse-cyclonedds/cyclonedds/src/ddsrt/src/environ/posix/environ.c
(2) /home/osrf/ws/ros2/src/eclipse-cyclonedds/cyclonedds/src/ddsrt/src/heap/posix/heap.c
(2) /home/osrf/ws/ros2/src/eclipse-cyclonedds/cyclonedds/src/ddsrt/src/ifaddrs.c
(2) /home/osrf/ws/ros2/src/eclipse-cyclonedds/cyclonedds/src/ddsrt/src/random.c
(2) /home/osrf/ws/ros2/src/eclipse-cyclonedds/cyclonedds/src/ddsrt/src/random/posix/random.c
(2) /home/osrf/ws/ros2/src/eclipse-cyclonedds/cyclonedds/src/ddsrt/src/ifaddrs/posix/ifaddrs.c
(2) /home/osrf/ws/ros2/src/eclipse-cyclonedds/cyclonedds/src/ddsrt/src/sockets.c
(2) /home/osrf/ws/ros2/src/eclipse-cyclonedds/cyclonedds/src/ddsrt/src/rusage/posix/rusage.c
(2) /home/osrf/ws/ros2/src/eclipse-cyclonedds/cyclonedds/src/ddsrt/src/sockets/posix/gethostname.c
(2) /home/osrf/ws/ros2/src/eclipse-cyclonedds/cyclonedds/src/ddsrt/src/sockets/posix/socket.c
(2) /home/osrf/ws/ros2/src/eclipse-cyclonedds/cyclonedds/src/ddsrt/src/string.c
(2) /home/osrf/ws/ros2/src/eclipse-cyclonedds/cyclonedds/src/ddsrt/src/string/posix/strerror.c
(2) /home/osrf/ws/ros2/src/eclipse-cyclonedds/cyclonedds/src/ddsrt/src/sync/posix/sync.c
(2) /home/osrf/ws/ros2/src/eclipse-cyclonedds/cyclonedds/src/ddsrt/src/threads.c
[... 99 lines cut ...]
(3) /home/osrf/ws/ros2/src/ros2/rosbag2/rosbag2_transport/test/rosbag2_transport/test_record_all.cpp
(3) /home/osrf/ws/ros2/src/ros2/rosbag2/rosbag2_transport/test/rosbag2_transport/test_record_all_no_discovery.cpp
(3) /home/osrf/ws/ros2/src/ros2/rosbag2/rosbag2_transport/test/rosbag2_transport/test_play_topic_remap.cpp
(3) /home/osrf/ws/ros2/src/ros2/rosbag2/rosbag2_transport/test/rosbag2_transport/test_record.cpp
(3) /home/osrf/ws/ros2/src/ros2/rosbag2/rosbag2_transport/test/rosbag2_transport/test_play_next.cpp
(3) /home/osrf/ws/ros2/src/ros2/rosbag2/rosbag2_transport/test/rosbag2_transport/test_record_regex.cpp
(3) /home/osrf/ws/ros2/src/ros2/rosbag2/rosbag2_transport/test/rosbag2_transport/test_play_publish_clock.cpp
(3) /home/osrf/ws/ros2/src/ros2/rosbag2/rosbag2_transport/test/rosbag2_transport/test_play_services.cpp
(3) /home/osrf/ws/ros2/src/ros2/rosbag2/rosbag2_transport/test/rosbag2_transport/test_play_seek.cpp
(3) /home/osrf/ws/ros2/src/ros2/rosbag2/rosbag2_transport/test/rosbag2_transport/test_play_timing.cpp
(3) /home/osrf/ws/ros2/src/ros2/rosbag2/rosbag2_transport/test/rosbag2_transport/test_play.cpp
(3) /home/osrf/ws/ros2/src/ros2/rosbag2/rosbag2_transport/test/rosbag2_transport/test_play_loop.cpp
(2) /home/osrf/ws/ros2/src/ros2/rviz/rviz_common/src/rviz_common/display_factory.cpp
(2) /home/osrf/ws/ros2/src/ros2/rviz/rviz_common/src/rviz_common/frame_manager.cpp
(2) /home/osrf/ws/ros2/src/ros2/rviz/rviz_common/test/mock_display.cpp
(2) /home/osrf/ws/ros2/src/ros2/rviz/rviz_common/test/mock_property_change_receiver.cpp
(2) /home/osrf/ws/ros2/src/ros2/rviz/rviz_common/test/display_context_fixture.cpp
(2) /home/osrf/ws/ros2/src/ros2/rviz/rviz_common/test/ogre_testing_environment.cpp
(8) /home/osrf/ws/ros2/src/ros2/rviz/rviz_default_plugins/test/rviz_default_plugins/pointcloud_messages.cpp
(2) /home/osrf/ws/ros2/src/ros2/rviz/rviz_default_plugins/test/rviz_default_plugins/displays/marker/marker_messages.cpp
(8) /home/osrf/ws/ros2/src/ros2/rviz/rviz_default_plugins/test/rviz_default_plugins/page_objects/point_cloud_common_page_object.cpp
(3) /home/osrf/ws/ros2/src/ros2/rviz/rviz_default_plugins/test/rviz_default_plugins/page_objects/marker_display_page_object.cpp
(2) /home/osrf/ws/ros2/src/ros2/rviz/rviz_default_plugins/test/rviz_default_plugins/page_objects/pose_display_page_object.cpp
```
</details>

emersonknapp, 2021-09-27:
> Maybe splitting the ament CMake functions that create tests into two - one that creates the test executabe and another that creates the test itself could solve most of these? Only the second function would need to be run for each RMW implementation.

This sounds really nice! The triplicate build of the tests has always felt slow and unecessary given that it's all the same code and APIs - really just a swapped environment variable at runtime in most scenarios.

clalancette, 2024-09-10:
Over the last few years, I've actually fixed many examples of this.  I certainly won't claim that I've gotten rid of all of the duplicates, but I think enough has been done here to call this general bug closed.  If new ones pop up, we should of course fix them, but I'm going to close this out for now.  Feel free to reopen if you disagree with my assessment here.

closed 2024-09-10 by clalancette