ros2/ros2 #1194: Unable to build rosdl_generator_py package from source in Dashing

goxul, 2021-09-30:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 18.04
- Installation type:
  - From source ([link](https://docs.ros.org/en/dashing/Installation/Ubuntu-Development-Setup.html))

#### Steps to reproduce issue

```
colcon build --symlink-install
```

#### Expected behavior

Build without any error.

#### Actual behavior

>--- stderr: rosidl_generator_py
>Illegal instruction (core dumped)
make[2]: *** [rosidl_generator_py/rosidl_generator_py/_rosidl_generator_py_s.ep.rosidl_typesupport_fastrtps_c.c] Error 132
make[1]: *** [rosidl_generator_py_custom__py/CMakeFiles/rosidl_generator_py_custom__py.dir/all] Error 2
make: *** [all] Error 2

I am running this on a Jetson Xavier NX board by Nvidia. 

The output of `uname -a` is:

>Linux xavier-desktop 4.9.253-tegra #1 SMP PREEMPT Mon Jul 26 12:19:28 PDT 2021 aarch64 aarch64 aarch64 GNU/Linux

The output of  `cat /etc/os-release` is:

>NAME="Ubuntu"
VERSION="18.04.5 LTS (Bionic Beaver)"
ID=ubuntu
ID_LIKE=debian
PRETTY_NAME="Ubuntu 18.04.5 LTS"
VERSION_ID="18.04"
HOME_URL="https://www.ubuntu.com/"
SUPPORT_URL="https://help.ubuntu.com/"
BUG_REPORT_URL="https://bugs.launchpad.net/ubuntu/"
PRIVACY_POLICY_URL="https://www.ubuntu.com/legal/terms-and-policies/privacy-policy"
VERSION_CODENAME=bionic
UBUNTU_CODENAME=bionic

I tried running `sudo colcon build --symlink-install --packages-ignore rosidl_generator_py` as well, but it gets stuck on some other package which seems to need `rosidl_generator_py`. 

How do I fix this? I searched the issues and couldn't find anything related to this. Any help would be really nice, thank you.

fujitatomoya, 2021-09-30:
I am not sure what went wrong with this, but Dashing is already `E.O.L`. (see https://docs.ros.org/en/rolling/Releases.html)

goxul, 2021-09-30:
Fair enough, the board that I have has Ubuntu 18.04, hence I was trying to build Dashing. 

I tried to build Foxy on the same board. I saw that the binary installation is only available for 20.04, but would it be able to build on 18.04?

I, in fact tried it, and got the same error:

>[Processing: rosidl_generator_py, rviz_ogre_vendor]                                                                                 
[Processing: rosidl_generator_py, rviz_ogre_vendor]                                                                                 
--- stderr: rosidl_generator_py                                                                                                    
Illegal instruction (core dumped)
make[2]: *** [rosidl_generator_py/rosidl_generator_py/_rosidl_generator_py_s.ep.rosidl_typesupport_fastrtps_c.c] Error 132
make[1]: *** [rosidl_generator_py_custom__py/CMakeFiles/rosidl_generator_py_custom__py.dir/all] Error 2
make: *** [all] Error 2
---
Failed   <<< rosidl_generator_py [1min 52s, exited with code 2]

I tried skipping this package but I think some packages ahead are dependent on this. Is there any other option to install this package manually or via some other way?

clalancette, 2021-09-30:
Foxy should build on 18.04; it isn't officially supported, but it does generally work.

The `Illegal instruction` means that the process is receiving a SIGILL, which generally means that you are trying to execute code that is not for that CPU.  Are you trying to cross-compile this?

goxul, 2021-09-30:
No, I am not trying to cross-compile.

 I've SSHed into my board (this is the [board](https://www.nvidia.com/en-in/autonomous-machines/embedded-systems/jetson-xavier-nx/) if it helps) and that's where I am trying to build from source.  

The strange part was that I tried to install Dashing via binaries on it and it installed without errors. But when I tried to run `ros2 run demo talker`, it showed `Illegal instruction (core dumped)`. What could possible be the issue?

goxul, 2021-09-30:
If it helps, here's more info about where I am trying to build it:

>NVIDIA Jetson Xavier NX (Developer Kit Version)
L4T 32.6.1 [ JetPack 4.6 ]
Ubuntu 18.04.5 LTS
Kernel Version: 4.9.253-tegra
CUDA NOT_INSTALLED
CUDA Architecture: 7.2
OpenCV version: 4.1.1
OpenCV Cuda: NO
CUDNN: NOT_INSTALLED
TensorRT: NOT_INSTALLED
Vision Works: NOT_INSTALLED
VPI: NOT_INSTALLED
Vulcan: 1.2.70

The related overview can be found here - [link](https://developer.nvidia.com/embedded/linux-tegra-r3261)

goxul, 2021-09-30:
There were two packages that gave me an error `rosidl_generator_py` and `rclpy`. I skipped them and was able to finish building Foxy from source, but when I tried to execute source, it gave me this error

>not found: "/home/xavier/ros2_foxy/install/rosidl_generator_py/share/rosidl_generator_py/local_setup.bash"

I suppose it's a package which I can't skip. Is it possible to build it manually or something else?

closed 2021-10-06 by goxul

skymaze, 2022-01-24:
I had the same problem building foxy with JetPack 4.6 and found a workaround, hope it helps.

`export OPENBLAS_CORETYPE=ARMV8`

Then build.