ros2/ros2 #1211: fat binary releases too large

christianrauch, 2021-10-31:
The fat binary release is multiple times larger than the Debian packages. E.g. the `librclcpp.so` from the fat release is about 40MB in size, while the file in the Debian package is only about 2MB. There are many more libraries that are larger in the fat release than in their corresponding Debian package. Especially on a Raspberry Pi with a small SD card, this is a limiting factor.

## Bug report

**Required Info:**

- Operating System: Ubuntu 20.04
- Installation type: binaries
- Version or commit hash: 6.1 (20211013)
- DDS implementation: irrelevant
- Client library (if applicable): rclcpp

#### Steps to reproduce issue
1. download the fat arm64 release from https://github.com/ros2/ros2/releases/download/release-foxy-20211013/ros2-foxy-20211013-linux-focal-arm64.tar.bz2
2. show size of `librclcpp.so` via `ls -alh`, it's shown as `39M` on my system
3. compare with debian package version at `/opt/ros/foxy/lib/librclcpp.so`, it's shown as `1,9M`

#### Expected behavior
fat releases should have the same binary size than the debian packages

#### Actual behavior
fat releases are tenfold larger that official debian releases

gavanderhoorn, 2021-10-31:
I ran into this myself some time ago.

It looks like the fat archives are not stripped (which could be why they are called "fat"). The `.deb`s have their debug symbols stripped by default (and then made available separately).

Vanilla files from archive: 2.7 GB

After recursively stripping everything: 313 MB

Debugging becomes quite a bit more difficult after stripping though.

clalancette, 2021-10-31:
Exactly as @gavanderhoorn says; it's much larger because it has the symbols.  It's a tradeoff between usability and download size, and for the binaries here, we've decided to include everything.  Without the debugging symbols, any debugging becomes basically useless.

While we *could* consider having a "fat" and "slim" variety, that is more confusing for users (which one should they download?), and more work for a release.  So I'm inclined to close this as "wontfix", but I'll leave it open for now for more comments.

christianrauch, 2021-10-31:
> It looks like the fat archives are not stripped (which could be why they are called "fat").

I thought it's called "fat" because it is one single large archive instead of split packages. Is it really called "fat" because it contains debug symbols?

Does the archive only contain the symbols or does it contain additional debug information that will impact performance?

> The .debs have their debug symbols stripped by default (and then made available separately).

Is it not possible to do the same for the released archives? One archive with the runtime libraries and one with the debug information? This is quite common and I don't think it will confuse people. Most users of ROS know a bit or two about computers and programming.

> While we could consider having a "fat" and "slim" variety, that is more confusing for users (which one should they download?),

I think only a minority of ROS2 users are actually debugging the ROS2 core libraries. You can probably verify this by looking at the ROS2 repo usage and comparing how often `ros-foxy-rclcpp` and `ros-foxy-rclcpp-dbgsym` is installed. I bet that there are less than 10% of installations of `ros-foxy-rclcpp-dbgsym` than there are for `ros-foxy-rclcpp`.

Splitting the debug symbols from the archive is much more efficient for bandwidth and memory usage on embedded systems like the Raspberry Pi. Having them there for the less than 10% of users that actually debug core libraries causes the remaining 90% to use more bandwidth and storage space. I don't think "just in case" is a good argument for wasting resources. And if you really want to debug core libraries and fix their issues, you are probably going to check out a package by source anyway.

gavanderhoorn, 2021-10-31:
> > It looks like the fat archives are not stripped (which could be why they are called "fat").
> 
> I thought it's called "fat" because it is one single large archive instead of split packages. Is it really called "fat" because it contains debug symbols?

I was making an assumption here (note "which could be why"). It's likely that assumption is incorrect.

clalancette, 2021-11-01:
> I don't think "just in case" is a good argument for wasting resources.

This is all a tradeoff.  If we *don't* have the symbols in there, then someone will want it and have no way to get it.  If we do have the symbols in there, we end up using more bandwidth and disk space.  But the thing about bandwidth and disk space is that it is relatively cheap.  Users on desktop-class systems probably won't notice the difference between a 1GB archive and a 300MB archive.  If you are on something smaller, then you are probably going to have to build from source anyway.  And the debian packages are available as well (though I realize this isn't always a solution).

On the whole, I don't find the issue particularly problematic to devote time to optimizing it.  Others can comment if they disagree or want to spend time looking at a solution.

christianrauch, 2021-11-01:
> This is all a tradeoff. If we _don't_ have the symbols in there, then someone will want it and have no way to get it.

Sure it is a trade-off. But you have to consider how many users benefit of increasing the archive size by a multiple. You can still provide the debug symbols in a separate archive for those that really need them. I am not requesting to remove the debug symbols entirely, I am suggesting to split those into a separate archive.

> But the thing about bandwidth and disk space is that it is relatively cheap.

I guess you are talking only about "western cultures" here. I really dislike this mindset of "we have sufficient resources, let's not care about them" and I hope this mindset is only present in a minority of ROS developers. ROS is used by many people in many projects and having this mindset is not a good way forward. There are people that cannot effort to buy a new computer every 2 years or have access to a high-speed internet connection.

> If you are on something smaller, then you are probably going to have to build from source anyway.

Why would you build all of ROS from scratch on a Raspberry Pi if you can just download a 300MB archive?

To back this all up with facts: Do you have quantitative results on how often `dbgsym` packages are downloaded from the official ROS repo compared to the standard runtime version of packages? Is there a survey on how many people actually need the debug symbols in the archives? I doubt that there is a high percentage of ROS users that also use the debug symbols.

gavanderhoorn, 2021-11-01:
Without attaching ethics to this: @clalancette: reducing the size of the the archive is actually rather trivial (recursive `strip` on all binaries).

Are the fat archives uploaded manually, or is that already automated? If the latter, creating a second archive with the stripped binaries should not be too complex, right?

nuclearsandwich, 2021-11-05:
@christianrauch you've correctly pointed out that bandwidth and storage being "relatively cheap" is a relative statement and that the resources consumed by the ROS 2 release archives can be prohibitive for otherwise valid workloads. We haven't made any changes to the binary archive pipeline to accommodate bandwidth and storage constraints in the past because there has not previously been any report that they were too large. They've remained within the limitations for those of us producing them which is where the assessment of "good enough" comes from . So thanks for sharing that they don't currently meet your needs because we wouldn't know to consider doing anything differently without someone raising the issue.

> I think only a minority of ROS2 users are actually debugging the ROS2 core libraries. You can probably verify this by looking at the ROS2 repo usage and comparing how often `ros-foxy-rclcpp` and [...]

Looking at http://awstats.osuosl.org/reports/packages.ros.org/2021/10/awstats.packages.ros.org.downloads.html you're not far off.

>  reducing the size of the the archive is actually rather trivial (recursive `strip` on all binaries).
> Are the fat archives uploaded manually, or is that already automated? If the latter, creating a second archive with the stripped binaries should not be too complex, right?

These binaries are built on https://ci.ros.org using the packaging and ci_packaging jobs, then uploaded by the ROS Bosses to GitHub for ROS 2 patch releases.

All of these archives, with the exception of the Windows Debug one, are built using the CMake build type `RelWithDebInfo`. If you're interested in seeing stripped binary archives a good first step would be to contribute a pull request to the packaging jobs in https://github.com/ros2/ci which will perform the proposed recursive strip of debug symbols. Integrating support for stripping binaries in the CI scripts would at least give us the option of producing stripped release archives in addition to the existing archives.

@christianrauch are you using the CentOS archive for Foxy? If not, what limitations or preferences prohibit you from using the .deb packages on your target platform since they already contain the stripped binaries?

These packaging jobs are something of a historical appendage for Linux operating systems and I at least considered them something that we would stop producing once all of ROS 2's target platforms had a supported modular packaging format and we've discussed dropping the Linux ones in favor of the Ubuntu and RHEL packages even earlier.

> I was making an assumption here (note "which could be why"). It's likely that assumption is incorrect.

The name arises from the fact that the entire workspace is a single archive versus the per-package binaries you would normally get from a package management framework. Personally I'm not a big fan of the "fat" nomenclature and have tried at various times to rebrand them as "comprehensive", " all-in-one" or "workspace" archives without much success.

christianrauch, 2021-11-06:
> Looking at http://awstats.osuosl.org/reports/packages.ros.org/2021/10/awstats.packages.ros.org.downloads.html you're not far off.

Do I read that correctly that on the amd64 platform `ros-foxy-rclcpp` is downloaded more than 20.000 times and the corresponding `ros-foxy-rclcpp-dbgsym` only 40 times? That's less than 0.2%.

> @christianrauch are you using the CentOS archive for Foxy? If not, what limitations or preferences prohibit you from using the .deb packages on your target platform since they already contain the stripped binaries?

I was about to install that archive on a Raspberry Pi and did not want to go through the manual steps of adding the repo. I opened this issue because I thought that the huge archive is an oversight.

Seeing that this was intended, I am just arguing that having an about 15x larger archive size (comparing the macos and linux amd64 archives) is not reasonable for a less than 0.2% userbase, for which the debug binaries can be made optional. And defending this with the argument that there are plenty of resources is not a good move by a software project that wants to drive real-time robots.

gavanderhoorn, 2021-11-06:
> Seeing that this was intended, I am just arguing that having an about 15x larger archive size (comparing the macos and linux amd64 archives) is not reasonable for a less than 0.2% userbase, for [...]

no one is defending anything. It's already been established that this is something that can be solved, but will need someone to spend the effort.

> by a software project that wants to drive real-time robots.

not sure what you are trying to say here.