ros2/ros2 #1212: available_rmw_implementations is null

yiyibst, 2021-11-02:
I ported some components of ros2-foxy to our framework and when I try to compile ros2-demo with colcon build, I get the following error:

CMake Error at xx/sysroots/aarch64-bst-linux/usr/share/rmw_implementation/cmake/rmw_implementation-extras.cmake:54 (message):
  The RMW implementation has been specified as 'rmw_fastrtps_cpp' via the
  default when rmw_implementation was built, but it is not available at this
  time.
  Currently available middlewares:
  ''
Call Stack (most recent call first):
  xx/sysroots/aarch64-bst-linux/usr/share/rmw_implementation/cmake/rmw_implementationConfig.cmake:41 (include)
  xx/sysroots/aarch64-bst-linux/usr/share/rcl/cmake/ament_cmake_export_dependencies-extras.cmake:21 (find_package)
  xx/sysroots/aarch64-bst-linux/usr/share/rcl/cmake/rclConfig.cmake:41 (include)
  xx/sysroots/aarch64-bst-linux/usr/share/libstatistics_collector/cmake/ament_cmake_export_dependencies-extras.cmake:21 (find_package)
  xx/sysroots/aarch64-bst-linux/usr/share/libstatistics_collector/cmake/libstatistics_collectorConfig.cmake:41 (include)
  xx/sysroots/aarch64-bst-linux/usr/share/rclcpp/cmake/ament_cmake_export_dependencies-extras.cmake:21 (find_package)
  xx/aarch64-bst-linux/usr/share/rclcpp/cmake/rclcppConfig.cmake:41 (include)
  CMakeLists.txt:14 (find_package)

---
Failed   <<< cpp_pubsub [1.89s, exited with code 1]

Summary: 0 packages finished [2.13s]
  1 package failed: cpp_pubsub
  1 package had stderr output: cpp_pubsub

I want to use fastrtps as middleware and I have ported the fastrtps implementation. I don't understand why I get the error, do I need to set something? Or is there a missing component?

fujitatomoya, 2021-11-02:
is that possible to share the complete reproducible procedure with the package failed to compile?

yiyibst, 2021-11-03:
I used the https://github.com/ros/meta-rosd warrior branch to build the aarch4-based ros2 image and built the corresponding sdk for the ament compilation. The corresponding ros2 components added to the image are:
                  rcutils  \
                rcpputils \
                rosidl-typesupport-interface \
                rosidl-generator-c \
                rosidl-generator-cpp \
                rosidl-typesupport-introspection-c \
                rosidl-typesupport-introspection-cpp \
                rosidl-typesupport-c \
                rosidl-typesupport-cpp \
                unique-identifier-msgs \
                lifecycle-msgs \
                rcl-logging-noop \
                rcl-yaml-param-parser \
                builtin-interfaces \
                rcl-interfaces \
                rosgraph-msgs \
                composition-interfaces \
                action-msgs \
                std-msgs \
                rmw \
                rmw-implementation \
                rmw-implementation-cmake \
                rmw-fastrtps-cpp \
                rcl \
                rcl-action \
                rcl-lifecycle \
                rclcpp \
                rclcpp-lifecycle \
                ros2pkg \
                ros2topic \
                ros2run \
                ros2node \
                ros2service \
                ros2cli \
                ros2cli-dev \
                ament-cmake \
                ament-cmake-ros \
                ament-cmake-core \
               ament-cmake-export-definitions \
               ament-cmake-export-dependencies \
               ament-cmake-export-include-directories \
               ament-cmake-export-interfaces \
               ament-cmake-export-libraries \
               ament-cmake-export-link-flags \
               ament-cmake-export-targets \
               ament-cmake-libraries \
               ament-cmake-python \
               ament-cmake-target-dependencies \
               ament-package \
               ament-cmake-include-directories \
               ament-cmake-test \
               ament-cmake-version \
               fastrtps \
               ros-environment \
               rosidl-generator-dds-idl \
               fastrtps-cmake-module \
               rosidl-typesupport-fastrtps-cpp \
               rosidl-default-generators \
               rosidl-typesupport-fastrtps-c \

The corresponding ros2 component added in the sdk is:
            ros2cli \
                ros2pkg \
               ament-cmake \
               ament-cmake-core \
               ament-cmake-export-definitions \
               ament-cmake-export-dependencies \
               ament-cmake-export-include-directories \
               ament-cmake-export-interfaces \
               ament-cmake-export-libraries \
               ament-cmake-export-link-flags \
               ament-cmake-export-targets \
               ament-cmake-libraries \
               ament-cmake-python \
               ament-cmake-target-dependencies \
               ament-package \
               ros-environment \
               ros-workspace \
               ament-index-python \

Are there any libraries missing or do the environment variables need special settings?

fujitatomoya, 2021-11-03:
I cannot really tell you what's missing here, and https://github.com/ros/meta-rosd is `Page Not Found`. i would check the build environment at 1st to make sure all of the tool-chain is installed though.

yiyibst, 2021-11-04:
As you said, I set the environment of AMENT_PREFIX_PATH variable, and I am not reporting the above error anymore. But after colcon build, the error is reported as
Starting >>> cpp_pubsub
Failed <<< cpp_pubsub [2.19s, exited with code 1]

Summary: 0 packages finished [2.56s]
  1 package failed: cpp_pubsub
  1 package not processed
This one confuses me even more. There is no error message, I checked the log error and it just says Configuring incomplete, errors occurred!
I would like to know what toolchain is needed to build the ros2 package? I currently only added ros2cli and ation_cmake related ones.

clalancette, 2021-11-18:
The toolchains we generally support are gcc 9.3.0 on Ubuntu 20.04, MSVC 2019 on Windows, and Clang on macOS.  Other toolchains may work, but we don't test them and generally rely on community support for them.

At this point, this seems to be a question.  I think you might get more response over at https://answers.ros.org, so I'm going to close this and suggest you ask the question there.

closed 2021-11-18 by clalancette