ros2/ros2 #1213: ROS 2 Galactic : Problems with multiple message types on the same topic

Felix-Helsch, 2021-11-02:
## Bug report

**Required Info:**

- Operating System:
Ubuntu 20.04
- Installation type:
binaries
- Version or commit hash:
ROS2 Galactic
- DDS implementation:
Fast-RTPS or Eclipse Cyclone DDS
- Client library (if applicable):
rclpy

#### Steps to reproduce issue

Publish/Subscribe to multiple different issues on a topic

```
import rclpy
import std_msgs.msg as std_msg

def main(args=None):

    rclpy.init(args=args)

    node1 = rclpy.create_node('Node1')

    # works for versions <= Foxy, but not for Galactic
    sub1 = node1.create_subscription(std_msg.Int32, 'my_topic', lambda msg: print(msg), 1)
    sub2 = node1.create_subscription(std_msg.String, 'my_topic', lambda msg: print(msg), 1)

    # same issue for publishers
    pub1 = node1.create_publisher(std_msg.Int32, 'my_topic', 1)
    pub2 = node1.create_publisher(std_msg.String, 'my_topic', 1)

    rclpy.shutdown()
```

#### Expected behavior

Program runs without crashing. (I.e. it is possible to use different message types on the same topic)

#### Actual behavior

Crashes with error message:

```
Traceback (most recent call last):
  File ".../galactic_test", line 11, in <module>
    load_entry_point('...==0.0.0', 'console_scripts', 'galactic_test')()
  File "...galactic_test.py", line 12, in main
    sub2 = node1.create_subscription(std_msg.String, 'my_topic', lambda msg: print(msg), 1)
  File "/opt/ros/galactic/lib/python3.8/site-packages/rclpy/node.py", line 1363, in create_subscription
    subscription_capsule = _rclpy.rclpy_create_subscription(
rclpy._rclpy_pybind11.RCLError: Failed to create subscription: failed to create topic, at /tmp/binarydeb/ros-galactic-rmw-cyclonedds-cpp-0.22.3/src/rmw_node.cpp:2552, at /tmp/binarydeb/ros-galactic-rcl-3.1.2/src/rcl/subscription.c:105
```
(local path and package name is replaced by "..." in this message)
 
#### Additional information

We are working on a larger ROS 2 project which was using ROS 2 Dashing before. Currently we are in the process of upgrading our ROS version, but for the topics which use multiple different message types, we encountered the described issue.

We also investigated this a bit more and it seems that the issue is not occurring under Foxy. But under Galactic it seems to occur for Eclipse Cyclone DDS as well as Fast-RTPS.

If we missed anything during our testing and there is an easy fix for this, then please let us know :)

fujitatomoya, 2021-11-02:
related to https://github.com/ros2/ros2/issues/1095

gavanderhoorn, 2021-11-02:
@Felix-Helsch: could you describe a situation where having a single topic with multiple message types makes something possible which otherwise would not work? Or in other words: for what kind of application would not having a 1-type-per-topic approach make sense?

Felix-Helsch, 2021-11-03:
We designed our architecture so that the topic names are derived from the components which publish the messages. And then naturally some components publish multiple different messages.

For example we have a "Command Control Center" (GUI application) with a "Mode Selection" component where e.g. a "Driving Mode" and a "Cage Mode" (safety related feature) can be selected. And depending on that a different message type is published on the topic belonging to the component.

Of course it would be possible to create additional subscribers with different topics. But the flexibility to choose if we want to publish the message on the same topic was definitely useful.
(In our case  it also made it easier to get a clear mapping from our architecture to the topics)

gavanderhoorn, 2021-11-03:
this is off-topic of course.

Personally I'm not a fan of using a typed communication system and then 'violating' one of its core ideas (1 topic == 1 type), but perhaps that's the ambiguity were partly trying to solve here (or perhaps in #1095): is what you describe an acceptable (or perhaps intended) use of the infrastructure.

It seems unfortunate to me to "throw away" the possibility for a system to help you avoid making mistakes, which it seems being able to publish multiple different types of messages on the same topic does (ie: how would it know incompatible components are being connected to each other if "anything goes"?).

That's perhaps more of a design discussion or architectural considering than the technical issue you reported in your OP.

clalancette, 2021-11-03:
For what it is worth, I can't reproduce the failure on Foxy, Galactic, or the latest Rolling sources.  In all cases, the program runs and immediately shuts down with no failures.

@Felix-Helsch Besides running your test program, is there anything else that needs to be done to trigger the problem?

Felix-Helsch, 2021-11-05:
@clalancette : The code which I posted was included in a ros package, where the main method is used as entry point for an executable.

If you want to test it without ros package, you could modify the code to:
```
import rclpy
import std_msgs.msg as std_msg

def main(args=None):

    rclpy.init(args=args)

    node1 = rclpy.create_node('Node1')

    # works for versions <= Foxy, but not for Galactic
    sub1 = node1.create_subscription(std_msg.Int32, 'my_topic', lambda msg: print(msg), 1)
    sub2 = node1.create_subscription(std_msg.String, 'my_topic', lambda msg: print(msg), 1)

    # same issue for publishers
    pub1 = node1.create_publisher(std_msg.Int32, 'my_topic', 1)
    pub2 = node1.create_publisher(std_msg.String, 'my_topic', 1)

    rclpy.shutdown()
    
if __name__=='__main__':
    main()
```
Then source the ros installation and run the program:
```
source /opt/ros/galactic/setup.bash
python3 galactic_test.py
```
We just tested this again on a fresh Ubuntu 20.04 installation with a fresh Galactic installation using the debian packages and are always getting the described error. Then we also installed Foxy on the same machine and with sourcing Foxy instead of Galactic the program runs and exits without problem.

ZYblend, 2022-05-13:
Hi Felix-Helsch, have you resolved this problem. I am also encountering the same problem.

pranavpshah, 2024-02-09:
Hi @Felix-Helsch have you found a work around for this. Facing this same issue right and would be great to get some help!