ros2/ros2 #1215: Compiling ROS2 from source to debian packages

xander-m2k, 2021-11-04:
Currently am working on a ROS 2 project, and wanting to run ROS on a real-time kernel (hypervisor and priviliged). The only compatible distro's for that are Gentoo or Debian, so Debian was a quick choice.
So, to run ROS foxy on a debian you can compile with colcon and build it from source and use ROS2 fine. But I'm seeking to find a more robust and refined solution, namely compiling it to debian packages. 

To my frustration no build script or whatsoever exists online to build ROS2 on and for a custom distro.
Please help me by providing a script to build all the needed packages, and a guide to create rosdeps for all of those packages.

aprotyas, 2021-11-04:
If I'm not misunderstanding, this discourse post contains the information you need.

[discourse.ros.org - how to create debian package for ROS 2](https://discourse.ros.org/t/how-to-create-debian-package-for-ros-2/19673)

xander-m2k, 2021-11-04:
@aprotyas you're misunderstanding me indeed. I am talking about ROS foxy itself, and not a ROS package.

nuclearsandwich, 2021-11-05:
> To my frustration no build script or whatsoever exists online (not even for Ubuntu Focal itself) to build ROS2 on and for a custom distro.
> Please help me by providing a script to build all the needed packages, and a guide to create rosdeps for all of those packages.

To the best of my knowledge there's not a script for this operation because it is larger and more complicated than a single script can generally accommodate. The entire [ros_buildfarm](https://github.com/ros-infrastructure/ros_buildfarm) project exists to configure and run a Jenkins cluster for building and hosting all the individual packages in a ROS distribution in a coordinated fashion. 

Depending on your needs, setting up an entire build farm cluster is probably more overhead than you actually want but many of the suggestions in the thread linked by @aprotyas can be broadened to apply to an entire colcon install space if creating individual debs per package is not what you want.

My recommendation if none of the answers linked in that thread meet your needs is to ask a new question on [ROS Answers](https://answers.ros.org) detailing clearly what you want and which of the existing solutions you have determined not to be ideal.

---

Since this issue is more of a process question than a report of a bug or an enhancement request for ROS 2 itself I am going to close this issue.

closed 2021-11-05 by nuclearsandwich

xander-m2k, 2021-11-05:
@nuclearsandwich the thing is, if you want to use ROS on another linux distro, docker included, it would be really easy to just have all deb packages. Right now, this is just nearly impossible, not even to mention creating custom linux distro ros packages.

Nonetheless, it is very unfortunate that this does not exist.

aprotyas, 2021-11-05:
@xander-m2k one idea could be to have a script that walks through the source tree and invokes `bloom-generate rosdebian && fakeroot debian/rules binary` on each directory containing a `package.xml` file, right?

xander-m2k, 2021-11-05:
@aprotyas I did, but that causes all kinds of dependency errors unfortunately. I even made a custom rosdep.yaml, but it does not fully work.

aprotyas, 2021-11-05:
I see, my final comment here would be: you might want to check out this [ros_buildfarm/sourcedeb_joy.py](https://github.com/ros-infrastructure/ros_buildfarm/blob/master/ros_buildfarm/sourcedeb_job.py) script to see if it helps generate debians for you.

nuclearsandwich, 2021-11-05:
Another project to take a look at is https://github.com/ros-infrastructure/superflore/ which was built to do something analogous to "bloom-release everything" for Gentoo and OpenEmbedded (hey, maybe this will flip you on to Gentoo for realtime kernels). One long-standing wishlist item for the ROS infrastructure team is to re-use bloom's templating engine for superflore so that all platforms supported by Bloom are usable in superflore and superflore's platforms could be reworked into bloom templates instead of the internal API.  If that's a project you would be interested in contributing to then we can provide guidance but it's not at the top of our work queue right now.

> > one idea could be to have a script that walks through the source tree and invokes `bloom-generate rosdebian && fakeroot debian/rules binary` on each directory containing a `package.xml` file, right?

> I did, but that causes all kinds of dependency errors unfortunately. I even made a custom rosdep.yaml, but it does not fully work.

The migrate-rosdistro.py script in https://github.com/ros/rosdistro/tree/master/migration-tools does some dangerously clever work to avoid the need to bloom packages in a distribution topologically by pretending that they're all present at the beginning and then taking them out if doing an actual bloom release fails. That approach may help your homegrown solution.

xander-m2k, 2021-11-06:
@nuclearsandwich thanks for this great suggestion, I will definitely take a look at it :)