ros2/ros2 #1217: Question about preallocation for sensor_msgs/Image

advancenOO, 2021-11-09:
I could use uint8[<=N] in ros2 IDL to preallocate memory for arrays.
And I am wondering if I can preallocate for data(uint8[]) field of sensor_msgs/Image?

advancenOO, 2021-11-09:
I am using ROS2 eloquent with fastrtps, and I find preallocation for sensor_msgs/Image is 0 according to int8_multi_array__type_support.cpp.

clalancette, 2021-11-09:
As it stands, no that is not possible.  If you want a preallocated Image-like message, you'll have to define your own `.msg` file that has the appropriate amount of space for the image you want to transfer.

In the future, please open questions like this on https://answers.ros.org .  That's our central questions-and-answers site, and will archive the answer for future searching.  I'm going to close this one out for now.

closed 2021-11-09 by clalancette