ros2/ros2 #1219: ros2.repo file does not provide the specific commit id, is that ok?

crazyhank, 2021-11-12:
I found that there are no commit id information in ros2.repo file if you build source from scratch, for example:
```
ros2/rmw_fastrtps:
    type: git
    url: https://github.com/ros2/rmw_fastrtps.git
    version: galactic
```
so, I get the latest branch source code each time when I build, right?
If so, the problem is if I want to build galactic from source, my build result is different from binary installation? Is that OK?

clalancette, 2021-11-12:
There are two different branches for every major ROS release in this repository.  Let's assume for now that you are using Galactic.

If you want the "latest" Galactic sources, then you use https://github.com/ros2/ros2/blob/galactic/ros2.repos (that is, `ros2.repos` from the `galactic` branch).  That is all of the latest sources for Galactic.  It's what we typically recommend in our tutorials, as that has all of the latest fixes and is what is going to be in the next patch release.

If you want exactly the sources that are in the latest Galactic patch release, then you use https://github.com/ros2/ros2/blob/galactic-release/ros2.repos .  That has the exact tags that were used for the last Galactic patch release, but of course will not have the latest fixes.

In the future, please ask questions like this on https://answers.ros.org, which is our central Question-and-Answer site.  I'm going to close this out for now.

closed 2021-11-12 by clalancette

gavanderhoorn, 2021-11-12:
Reminds me of [How are they different: foxy and foxy-release branches](https://answers.ros.org/question/371996) on ROS Answers.