ros2/ros2 #1220: Paths not being exported

g-arjones, 2021-11-13:
I apologize if this is not the place to report this issue but I couldn't find a better one.

As far as I can tell, some paths (`PKG_CONFIG_PATH`, `CMAKE_MODULE_PATH`, `CMAKE_PREFIX_PATH`) are not being exported through the environment setup scripts and hooks. That makes integration with other build tools more painful than it should be.

```
arjones@dfcdfc78e327:~$ find /opt/ros/galactic/ -name "*.pc"
/opt/ros/galactic/lib/pkgconfig/orocos-kdl.pc
/opt/ros/galactic/lib/pkgconfig/orocos_kdl.pc
/opt/ros/galactic/lib/pkgconfig/libzstd.pc
/opt/ros/galactic/lib/x86_64-linux-gnu/pkgconfig/console_bridge.pc
/opt/ros/galactic/lib/x86_64-linux-gnu/pkgconfig/urdfdom.pc
/opt/ros/galactic/lib/x86_64-linux-gnu/pkgconfig/urdfdom_headers.pc
/opt/ros/galactic/opt/rviz_ogre_vendor/lib/pkgconfig/OGRE-Overlay.pc
/opt/ros/galactic/opt/rviz_ogre_vendor/lib/pkgconfig/OGRE.pc
/opt/ros/galactic/opt/rviz_ogre_vendor/lib/pkgconfig/OGRE-Paging.pc
/opt/ros/galactic/opt/rviz_ogre_vendor/lib/pkgconfig/OGRE-MeshLodGenerator.pc
/opt/ros/galactic/opt/rviz_ogre_vendor/lib/pkgconfig/OGRE-PCZ.pc
/opt/ros/galactic/opt/rviz_ogre_vendor/lib/pkgconfig/OGRE-RTShaderSystem.pc
/opt/ros/galactic/opt/rviz_ogre_vendor/lib/pkgconfig/OGRE-Terrain.pc
/opt/ros/galactic/opt/rviz_ogre_vendor/lib/pkgconfig/OGRE-HLMS.pc
/opt/ros/galactic/opt/rviz_ogre_vendor/lib/pkgconfig/OGRE-Volume.pc
/opt/ros/galactic/opt/rviz_ogre_vendor/lib/pkgconfig/OGRE-Property.pc
/opt/ros/galactic/opt/yaml_cpp_vendor/lib/pkgconfig/yaml-cpp.pc
arjones@dfcdfc78e327:~$ source /opt/ros/galactic/setup.bash 
arjones@dfcdfc78e327:~$ env | egrep "PKG_CONFIG_PATH|CMAKE_MODULE_PATH|CMAKE_PREFIX_PATH"
arjones@dfcdfc78e327:~$ 
```

Is this a bug or am I missing something?

clalancette, 2021-11-13:
As for the actual question: we don't support `pkg-config` in any way.  The fact that certain of our vendor packages have it (orocos, zstd, console_bridge, urdfdom, and OGRE) is purely an accident.  The main reason we don't support `pkg-config` is that it doesn't work on Windows, and Windows is one of our Tier 1 platforms.

In terms of the `CMAKE_MODULE_PATH` and `CMAKE_PREFIX_PATH`, I'm not sure what is going on with that.  We *do* set those on a build-from-source workspace, but not in a binary debian installation.  What we always set is `AMENT_PREFIX_PATH`, which is what the `ament_cmake` helper scripts use to find things.

But I'm not all that familiar with how `ament_cmake` works here.  Pinging @sloretz for some additional feedback.

g-arjones, 2021-11-13:
> As for the actual question: we don't support pkg-config in any way. The fact that certain of our vendor packages have it (orocos, zstd, console_bridge, urdfdom, and OGRE) is purely an accident. The [...]

That makes sense assuming you have hardcoded these packages include and libraries directories in the cmake configuration files of the packages that depend on them (i.e rviz_common). I didn't dig that deep but since you guys are building ROS2 on windows I think that's a fair assumption.

About `CMAKE_MODULE_PATH` and `CMAKE_PREFIX_PATH` though, I actually had to define those env vars myself to be able to properly integrate ROS2 with my build system... Hopefully, @sloretz will have some insights.

g-arjones, 2021-11-20:
@clalancette @sloretz Ping?

sloretz, 2021-11-22:
> In terms of the CMAKE_MODULE_PATH and CMAKE_PREFIX_PATH, I'm not sure what is going on with that. We do set those on a build-from-source workspace, but not in a binary debian installation. What we always set is AMENT_PREFIX_PATH, which is what the ament_cmake helper scripts use to find things.

> About CMAKE_MODULE_PATH and CMAKE_PREFIX_PATH though, I actually had to define those env vars myself to be able to properly integrate ROS2 with my build system

IIUC `CMAKE_PREFIX_PATH` and `CMAKE_MODULE_PATH` are set by `colcon` when building packages. I don't have a good understanding of how it happens, but I think a good place to start looking at it is here:

* https://github.com/colcon/colcon-cmake/blob/master/colcon_cmake/environment/cmake_module_path.py
* https://github.com/colcon/colcon-cmake/blob/master/colcon_cmake/environment/cmake_prefix_path.py
 
My personal opinion is the package setup scripts should set these instead of relying on `colcon`. I think `CMAKE_PREFIX_PATH` should be set for all `ament_cmake` packages ([I think environment hooks for all packages get set here](https://github.com/ament/ament_cmake/blob/84719051ef2b26070de484e60b8f060ae7a70b06/ament_cmake_core/cmake/environment_hooks/ament_cmake_environment_hooks_package_hook.cmake#L20)), and any packages also offering CMake find modules should register an environment hook to set `CMAKE_MODULE_PATH`.

g-arjones, 2021-11-22:
Yeah, I think this is very inconsistent (not to mention it makes integration harder). I really don't understand why these two variables are handled differently.

Would a PR that adds that to `ament_cmake_core` be welcome?