ros2/ros2 #1221: Transition Rolling to Ubuntu 22.04

nuclearsandwich, 2021-11-16:
Continuing from the broad announcement on Discourse. Here's a detailed checklist of items to take care of for the Rolling transition to Ubuntu 22.04.
- [ ] Create bootstrap repositories for `jammy` and `bullseye`
- [ ] Create ROS 2 repositories for `jammy` and `bullseye`
- [ ] Create osrf qemu-user-static images
  - [ ] osrf/ubuntu_arm64:jammy
  - [ ] osrf/debian_arm64:bullseye 
- [ ] Add Ubuntu Jammy support to [ros2/ci](https://github.com/ros2/ci)
- [ ] Prepare a Nightly Ubuntu Jammy debug build for the ROS 2 build farm in [ros2/ros_buildfarm_config](https://github.com/ros2/ros_buildfarm_config)
- [ ] Using results from the nightly debug job, begin resolving any issues with ROS 2 on Ubuntu Jammy
- [ ] Prepare updates to rosdep database to make sure that dependencies are met on Ubuntu Jammy
- [ ] Perform a test migration of ROS 2 Rolling to its new target platforms
- [ ] Update the default platforms for Rolling jobs in [ros2/ci](https://github.com/ros2/ci) and [ros2/ros_buildfarm_config](https://github.com/ros2/ros_buildfarm_config)
- [ ] Perform the migration of ROS 2 Rolling to its new target platforms

nbbrooks, 2022-01-26:
What is the latest (rough) timeline for this? The original discourse post talked about end of January, is this going to end up happening closer to the Humble release?

clalancette, 2022-01-26:
> What is the latest (rough) timeline for this? The original discourse post talked about end of January, is this going to end up happening closer to the Humble release?

Still shooting for the end of January.  Given the work we had to do, this may end up slipping into the first week of February, but likely no longer than that.

nbbrooks, 2022-01-26:
Thanks Chris, I just realized I was looking at the Dec notes, not Jan, where this was all written up. https://discourse.ros.org/t/ros-2-tsc-meeting-january-20th-2022/23986

nuclearsandwich, 2022-01-26:
I opened this issue and completely forgot to update it as I went. Some bootstrapping challenges with the Jammy debug job have arisen and devel jobs can't run without binary jobs at the foundation so the last several tasks will all tick over to Jammy in a short fell swoop.

clalancette, 2022-04-22:
@nuclearsandwich Can we close this one out now?

clalancette, 2022-05-03:
Rolling is well and truly on Jammy now (despite some lingering jobs left on Focal).  As such, I'm going to close this out, but feel free to reopen if this issue is still needed.

closed 2022-05-03 by clalancette