ros2/ros2 #1222: library_abs set to Not FOUND on Mac Big Sur intel

crossev2000, 2021-11-24:
## Bug report

**Required Info:**

- Operating System:
- Mac OS Big Sur version 11.5.2 Processor Intel Core i7
  - 
- Installation type:
  - From Source
- Version or commit hash:
  - 
- DDS implementation:
  - 
- Client library (if applicable):
  - 

#### Steps to reproduce issue

```
Attempting to build ROS2 with the following command
colcon build \
  --symlink-install \
  --merge-install \
  --event-handlers console_cohesion+ console_package_list+ \
  --packages-skip-by-dep python_qt_binding \
  --cmake-args \
    --no-warn-unused-cli \
    -DBUILD_TESTING=OFF \
    -DINSTALL_EXAMPLES=ON \
    -DCMAKE_OSX_SYSROOT=/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk \
    -DCMAKE_OSX_ARCHITECTURES="x86_64" \
    -DCMAKE_PREFIX_PATH=$(brew --prefix):$(brew --prefix qt@5)
```

#### Expected behavior

It should compile

#### Actual behavior

library_abs should be set to some directory?

#### Additional information

I am following the instructions provided on http://mamykin.com/posts/building-ros2-on-macos-big-sur-m1/
However, I did not switch the variables to "arm64" and kept them as  x86_64 since I am on an Intel-based Mac

clalancette, 2021-12-09:
Thanks for the report.

We don't currently target macOS Big Sur with any of our ROS 2 releases, so I'm not quite sure what is going wrong here.  If you find out what the issue is, please feel free to open a PR to either fix it, or to the documentation at https://github.com/ros2/ros2_documentation .

oysstu, 2021-12-14:
I can confirm that the same issue appears on archlinux. Could this be related to release 3.22.1 of cmake? Looks like there's some policy changes amongst the [latest commits](https://gitlab.kitware.com/cmake/cmake/-/commits/v3.22.1) that may apply (e.g. CMP0126).

XuShaohua, 2021-12-15:
In assimp-5.1.0/assimpConfig.cmake, ASSIMP_LIBRARY_DIRS variable is reset to empty.
While rviz/rviz_assimp_vendor/rviz_assimp_vendor-extras.cmake.in depends on this variable to find assimp library.

Two ways to fix this issue quickly:

Modify assimpConfig.cmake, set `ASSIMP_LIBRARY_DIRS` to real folder, on Debian it is `/usr/lib/x86_64-linux-gnu`

Or modify rviz_assimp_vendor-extras.cmake.in:
```diff
diff --git a/rviz_assimp_vendor/rviz_assimp_vendor-extras.cmake.in b/rviz_assimp_vendor/rviz_assimp_vendor-extras.cmake.in
index 8e41fe66..929f659c 100644
--- a/rviz_assimp_vendor/rviz_assimp_vendor-extras.cmake.in
+++ b/rviz_assimp_vendor/rviz_assimp_vendor-extras.cmake.in
@@ -25,7 +25,7 @@ foreach(library IN LISTS ASSIMP_LIBRARIES)
   if(IS_ABSOLUTE "${library}")
     list(APPEND rviz_assimp_vendor_LIBRARIES "${library}")
   else()
-    find_library(library_abs ${library} PATHS "${ASSIMP_LIBRARY_DIRS}" NO_DEFAULT_PATH)
+    find_library(library_abs assimp)
     list(APPEND rviz_assimp_vendor_LIBRARIES "${library_abs}")
   endif()
 endforeach()
```

pryre, 2021-12-19:
> Fix this issue quickly, by modifying rviz_assimp_vendor-extras.cmake.in:

This method also works as a quick fix for the recent update in Arch Linux

crossev2000, 2021-12-20:
Thanks for the responses. I have not had a chance to try either option because I am now getting an error related to [Fails to identify package that specifies python_requires. #33](https://github.com/colcon/colcon-python-setup-py/issues/33). Not sure why this error is happening so I need to do some more digging. I will let you know if this works once I can get back to this point

> In assimp-5.1.0/assimpConfig.cmake, ASSIMP_LIBRARY_DIRS variable is reset to empty (see https://github.com/assimp/assimp/blob/master/cmake-modules/assimp-plain-config.cmake.in). While rviz/rviz_assimp_vendor/rviz_assimp_vendor-extras.cmake.in depends on this variable to find assimp library.
> 
> Fix this issue quickly, by modifying rviz_assimp_vendor-extras.cmake.in:
> 
> ```diff
> diff --git a/rviz_assimp_vendor/rviz_assimp_vendor-extras.cmake.in b/rviz_assimp_vendor/rviz_assimp_vendor-extras.cmake.in
> index 8e41fe66..929f659c 100644
> --- a/rviz_assimp_vendor/rviz_assimp_vendor-extras.cmake.in
> +++ b/rviz_assimp_vendor/rviz_assimp_vendor-extras.cmake.in
> @@ -25,7 +25,7 @@ foreach(library IN LISTS ASSIMP_LIBRARIES)
>    if(IS_ABSOLUTE "${library}")
>      list(APPEND rviz_assimp_vendor_LIBRARIES "${library}")
>    else()
> -    find_library(library_abs ${library} PATHS "${ASSIMP_LIBRARY_DIRS}" NO_DEFAULT_PATH)
> +    find_library(library_abs assimp)
>      list(APPEND rviz_assimp_vendor_LIBRARIES "${library_abs}")
>    endif()
>  endforeach()
> ```
> 
> Or replace `find_library()` command:
> 
> ```diff
> diff --git a/rviz_assimp_vendor/rviz_assimp_vendor-extras.cmake.in b/rviz_assimp_vendor/rviz_assimp_vendor-extras.cmake.in
> index 8e41fe66..abd7bedc 100644
> --- a/rviz_assimp_vendor/rviz_assimp_vendor-extras.cmake.in
> +++ b/rviz_assimp_vendor/rviz_assimp_vendor-extras.cmake.in
> @@ -25,8 +25,8 @@ foreach(library IN LISTS ASSIMP_LIBRARIES)
>    if(IS_ABSOLUTE "${library}")
>      list(APPEND rviz_assimp_vendor_LIBRARIES "${library}")
>    else()
> -    find_library(library_abs ${library} PATHS "${ASSIMP_LIBRARY_DIRS}" NO_DEFAULT_PATH)
> -    list(APPEND rviz_assimp_vendor_LIBRARIES "${library_abs}")
> +    get_target_property(rviz_assimp_vendor_LIBRARY ${library} IMPORTED_LOCATION_NONE)
> +    list(APPEND rviz_assimp_vendor_LIBRARIES "${rviz_assimp_vendor_LIBRARY}")
>    endif()
>  endforeach()
>  set(rviz_assimp_vendor_LIBRARY_DIRS ${ASSIMP_LIBRARY_DIRS})
> ```

KenwoodFox, 2022-01-06:
> In assimp-5.1.0/assimpConfig.cmake, ASSIMP_LIBRARY_DIRS variable is reset to empty (see [assimp/assimp@`master`/cmake-modules/assimp-plain-config.cmake.in](https://github.com/assimp/assimp/blob/master/cmake-modules/assimp-plain-config.cmake.in?rgh-link-date=2021-12-15T01%3A37%3A27Z)). While rviz/rviz_assimp_vendor/rviz_assimp_vendor-extras.cmake.in depends on this variable to find assimp library.
> 
> Fix this issue quickly, by modifying rviz_assimp_vendor-extras.cmake.in:
> 
> ```diff
> diff --git a/rviz_assimp_vendor/rviz_assimp_vendor-extras.cmake.in b/rviz_assimp_vendor/rviz_assimp_vendor-extras.cmake.in
> index 8e41fe66..929f659c 100644
> --- a/rviz_assimp_vendor/rviz_assimp_vendor-extras.cmake.in
> +++ b/rviz_assimp_vendor/rviz_assimp_vendor-extras.cmake.in
> @@ -25,7 +25,7 @@ foreach(library IN LISTS ASSIMP_LIBRARIES)
>    if(IS_ABSOLUTE "${library}")
>      list(APPEND rviz_assimp_vendor_LIBRARIES "${library}")
>    else()
> -    find_library(library_abs ${library} PATHS "${ASSIMP_LIBRARY_DIRS}" NO_DEFAULT_PATH)
> +    find_library(library_abs assimp)
>      list(APPEND rviz_assimp_vendor_LIBRARIES "${library_abs}")
>    endif()
>  endforeach()
> ```
> 
> Or replace `find_library()` command:
> 
> ```diff
> diff --git a/rviz_assimp_vendor/rviz_assimp_vendor-extras.cmake.in b/rviz_assimp_vendor/rviz_assimp_vendor-extras.cmake.in
> index 8e41fe66..abd7bedc 100644
> --- a/rviz_assimp_vendor/rviz_assimp_vendor-extras.cmake.in
> +++ b/rviz_assimp_vendor/rviz_assimp_vendor-extras.cmake.in
> @@ -25,8 +25,8 @@ foreach(library IN LISTS ASSIMP_LIBRARIES)
>    if(IS_ABSOLUTE "${library}")
>      list(APPEND rviz_assimp_vendor_LIBRARIES "${library}")
>    else()
> -    find_library(library_abs ${library} PATHS "${ASSIMP_LIBRARY_DIRS}" NO_DEFAULT_PATH)
> -    list(APPEND rviz_assimp_vendor_LIBRARIES "${library_abs}")
> +    get_target_property(rviz_assimp_vendor_LIBRARY ${library} IMPORTED_LOCATION_NONE)
> +    list(APPEND rviz_assimp_vendor_LIBRARIES "${rviz_assimp_vendor_LIBRARY}")
>    endif()
>  endforeach()
>  set(rviz_assimp_vendor_LIBRARY_DIRS ${ASSIMP_LIBRARY_DIRS})
> ```

I have this very helpful patch applied (on arch linux) already, guess it must be in the upstream but i still have this error

```
CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
Please set them or make sure they are set and tested correctly in the CMake files:
library_abs
    linked by target "rviz_rendering" in directory /home/joe/.cache/paru/clone/ros2-galactic/src/ros2/src/ros2/rviz/rviz_rendering
```

Is there something more i could do?

traversaro, 2022-01-07:
I am afraid that I caused this issue in the effort of cleaning up upstream the CMake configuration files of assimp (that create several problems over the years) in https://github.com/assimp/assimp/pull/3455 .
 
In RoboStack (cc @Tobias-Fischer @wolfv) we already have a patch for this in https://github.com/RoboStack/ros-galactic/blob/b26bfc2ae0708821b4598f2fa6171db4e38ad3dc/patch/ros-galactic-rviz-assimp-vendor.patch, but in needs to be modified to be used upstream  .

traversaro, 2022-01-07:
The issue should be fixed by https://github.com/ros2/rviz/pull/817 . It would be great if anyone affected by this issue could confirm that this fixes the problem.

michaelisc, 2022-01-18:
I had the same problem on macOS Monterey 12.1 and tested the patch [ros2/rviz#817](https://github.com/ros2/rviz/pull/817). It solved the assimp issue but new issues arose:

1. Qt (installed with homebrew following the [instructions of the macOS setup](https://docs.ros.org/en/galactic/Installation/macOS-Development-Setup.html)) requires a C++17 compiler.

```
--- stderr: rviz_rendering                                                                                   
In file included from /Users/claudio/Code/ros2_galactic/build/rviz_rendering/include/rviz_rendering/moc_render_window.cpp:10:
In file included from /Users/claudio/Code/ros2_galactic/build/rviz_rendering/include/rviz_rendering/../../../../src/ros2/rviz/rviz_rendering/include/rviz_rendering/render_window.hpp:36:
In file included from /usr/local/opt/qt@5/lib/QtCore.framework/Headers/QObject:1:
In file included from /usr/local/opt/qt@5/lib/QtCore.framework/Headers/qobject.h:46:
In file included from /usr/local/include/QtCore/qobjectdefs.h:48:
In file included from /usr/local/include/QtCore/qnamespace.h:44:
/usr/local/include/QtCore/qglobal.h:96:6: error: "Qt requires a C++17 compiler"
#    error "Qt requires a C++17 compiler"
     ^
/usr/local/include/QtCore/qglobal.h:773:43: error: no template named 'is_arithmetic_v' in namespace 'std'; did you mean 'is_arithmetic'?
         typename = std::enable_if_t<std::is_arithmetic_v<T> && std::is_arithmetic_v<U> &&
                                     ~~~~~^
```
I fixed this by editing `.../src/ros2/rviz/rviz_rendering/CMakeLists.txt`:
```diff
- # Default to C++14
+ # Default to C++17
  if(NOT CMAKE_CXX_STANDARD)
-   set(CMAKE_CXX_STANDARD 14)
+   set(CMAKE_CXX_STANDARD 17)
  endif()
```
This solved the C++17 issue but led to:

2. Some Qt versioning issue:

```
--- stderr: rviz_rendering                                                                             
/Users/claudio/Code/ros2_galactic/build/rviz_rendering/include/rviz_rendering/moc_render_window.cpp:16:2: error: "This file was generated using the moc from 5.15.2. It"
#error "This file was generated using the moc from 5.15.2. It"
 ^
/Users/claudio/Code/ros2_galactic/build/rviz_rendering/include/rviz_rendering/moc_render_window.cpp:17:2: error: "cannot be used with the include files from this version of Qt."
#error "cannot be used with the include files from this version of Qt."
 ^
/Users/claudio/Code/ros2_galactic/build/rviz_rendering/include/rviz_rendering/moc_render_window.cpp:18:2: error: "(The moc has changed too much.)"
#error "(The moc has changed too much.)"
 ^
/Users/claudio/Code/ros2_galactic/build/rviz_rendering/include/rviz_rendering/moc_render_window.cpp:35:1: error: use of undeclared identifier 'Q_STATIC_BYTE_ARRAY_DATA_HEADER_INITIALIZER_WITH_OFFSET'
QT_MOC_LITERAL(0, 0, 28), // "rviz_rendering::RenderWindow"
^
/Users/claudio/Code/ros2_galactic/build/rviz_rendering/include/rviz_rendering/moc_render_window.cpp:29:5: note: expanded from macro 'QT_MOC_LITERAL'
    Q_STATIC_BYTE_ARRAY_DATA_HEADER_INITIALIZER_WITH_OFFSET(len, \
    ^
/Users/claudio/Code/ros2_galactic/build/rviz_rendering/include/rviz_rendering/moc_render_window.cpp:36:1: error: use of undeclared identifier 'Q_STATIC_BYTE_ARRAY_DATA_HEADER_INITIALIZER_WITH_OFFSET'
QT_MOC_LITERAL(1, 29, 11), // "renderLater"
^
/Users/claudio/Code/ros2_galactic/build/rviz_rendering/include/rviz_rendering/moc_render_window.cpp:29:5: note: expanded from macro 'QT_MOC_LITERAL'
    Q_STATIC_BYTE_ARRAY_DATA_HEADER_INITIALIZER_WITH_OFFSET(len, \
    ^
/Users/claudio/Code/ros2_galactic/build/rviz_rendering/include/rviz_rendering/moc_render_window.cpp:37:1: error: use of undeclared identifier 'Q_STATIC_BYTE_ARRAY_DATA_HEADER_INITIALIZER_WITH_OFFSET'
QT_MOC_LITERAL(2, 41, 0), // ""
^
/Users/claudio/Code/ros2_galactic/build/rviz_rendering/include/rviz_rendering/moc_render_window.cpp:29:5: note: expanded from macro 'QT_MOC_LITERAL'
    Q_STATIC_BYTE_ARRAY_DATA_HEADER_INITIALIZER_WITH_OFFSET(len, \
    ^
/Users/claudio/Code/ros2_galactic/build/rviz_rendering/include/rviz_rendering/moc_render_window.cpp:38:1: error: use of undeclared identifier 'Q_STATIC_BYTE_ARRAY_DATA_HEADER_INITIALIZER_WITH_OFFSET'
QT_MOC_LITERAL(3, 42, 9) // "renderNow"
^
/Users/claudio/Code/ros2_galactic/build/rviz_rendering/include/rviz_rendering/moc_render_window.cpp:29:5: note: expanded from macro 'QT_MOC_LITERAL'
    Q_STATIC_BYTE_ARRAY_DATA_HEADER_INITIALIZER_WITH_OFFSET(len, \
    ^
/Users/claudio/Code/ros2_galactic/build/rviz_rendering/include/rviz_rendering/moc_render_window.cpp:84:1: error: unknown type name 'QT_INIT_METAOBJECT'
QT_INIT_METAOBJECT const QMetaObject rviz_rendering::RenderWindow::staticMetaObject = { {
^
/Users/claudio/Code/ros2_galactic/build/rviz_rendering/include/rviz_rendering/moc_render_window.cpp:84:37: error: expected ';' after top level declarator
QT_INIT_METAOBJECT const QMetaObject rviz_rendering::RenderWindow::staticMetaObject = { {
                                    ^
                                    ;
9 errors generated.
make[2]: *** [CMakeFiles/rviz_rendering.dir/include/rviz_rendering/moc_render_window.cpp.o] Error 1
make[1]: *** [CMakeFiles/rviz_rendering.dir/all] Error 2
make: *** [all] Error 2
make: INTERNAL: Exiting with 6 jobserver tokens available; should be 4!
---
Failed   <<< rviz_rendering [13.4s, exited with code 2]
```

There I got stuck.

Tobias-Fischer, 2022-01-18:
Note that we have working builds for OSX in RoboStack if that is an option for you: https://github.com/RoboStack/ros-galactic

eigendude, 2022-03-31:
I've lived the saga in this PR. Starting with the assimp problem, I rejoiced when it was fixed, and then hit the "Qt requires a C++17 compiler" problem. The solution [above](https://github.com/ros2/ros2/issues/1222#issuecomment-1015373355) worked:

```patch
-# Default to C++14
+# Default to C++17
 if(NOT CMAKE_CXX_STANDARD)
-  set(CMAKE_CXX_STANDARD 14)
+  set(CMAKE_CXX_STANDARD 17)
 endif()
```

Next I got the same Qt versioning error. Cleaning my build folder didn't fix the problem, so I cleared my entire ROS 2 workspace / desktop build files, and rebuilt everything from scratch. With that patch and a 100% full rebuild everything works.

mr-inaank, 2022-04-01:
Hey @eigendude!
I am right where you are with regards to installing ROS2. I am a bit new to this, and very out of my depth. Could you please clarify the steps you took to 

>  I cleared my entire ROS 2 workspace / desktop build files, and rebuilt everything from scratch

Greatly appreciated! Thanks!

eigendude, 2022-04-05:
@mr-inaank The journey to compile ROS is long but rewarding :slightly_smiling_face: I just open sourced my ROS 2 stack, [OASIS](https://github.com/eigendude/OASIS). Check out the [oasis_tooling](https://github.com/eigendude/OASIS/tree/main/oasis_tooling) package. You can manually enter commands there as you reference the official docs, and probably find some tricks to work around the more difficult problems.

There's a lot of scripts in that package, so you can view the file [build_all.sh](https://github.com/eigendude/OASIS/blob/main/oasis_tooling/scripts/build_all.sh) to see the entry points and the order I run them in.

Another strategy is to find the CI config in the repo. From CI configs you can see the commands used to build and test the whole workflow of repos.

moyutianque, 2022-04-25:
> OASIS

Have you found solutions? I tried to rebuild with the patches mentioned above and other settings from [this link](http://mamykin.com/posts/building-ros2-on-macos-big-sur-m1/) but the qt version error for rviz_rendering still occurs.
`error: "This file was generated using the moc from 5.15.3 cannot be used with the include files from this version of Qt. (The moc has changed too much.)`

eigendude, 2022-04-26:
@moyutianque Good job writing up the procedure! FWIW for me I solved this by doing `brew uninstall qt@6`.