ros2/ros2 #1230: stderr: rviz_rendering . Can't build ROS galactic from sources. macOS Monterey.

lexeek, 2021-12-18:
## Bug report

**Required Info:**

- Operating System:
   `macOS Monterey`

- Installation type:
`   from source` 

- Version or commit hash:
.
- DDS implementation:
`default`
- Client library (if applicable):
.

#### Steps to reproduce issue

I just tried to execute the following command 
```
 colcon build --symlink-install --packages-skip-by-dep python_qt_binding
```

#### Expected behavior
`It should compile `

#### Actual behavior
`Error occurs`
```
Starting >>> rosidl_generator_dds_idl
--- stderr: rviz_rendering
CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
Please set them or make sure they are set and tested correctly in the CMake files:
library_abs
    linked by target "rviz_rendering" in directory /Users/lexeek/ros2_galactic/src/ros2/rviz/rviz_rendering

CMake Generate step failed.  Build files cannot be regenerated correctly.
---
Failed   <<< rviz_rendering [1.89s, exited with code 1]
Aborted  <<< rosidl_generator_dds_idl [0.40s]
Aborted  <<< rcutils [3.94s]
```
#### Additional information

I just followed the steps from the installation manual for [galactic version, building ROS2 on mac os](https://docs.ros.org/en/galactic/). 
----
## Feature request

#### Feature description

#### Implementation considerations

cqtju, 2021-12-18:
It looks like the same problem: https://github.com/ros2/ros2/issues/1222

clalancette, 2021-12-21:
Yes, agreed, it does look like a duplicate of #1222.  Closing as such.

closed 2021-12-21 by clalancette