ros2/ros2 #1232: Console extremely slow after sourcing ros2

RSchwan, 2021-12-22:
## Bug report

**Required Info:**

- Operating System:
  - macOS
- Installation type:
  - from source
- Version or commit hash:
  - galactic

#### Steps to reproduce issue
After `source ~/ros2_galactic/install/setup.bash` all commands run become extremely slow. This is especially bad when compiling packages with `colcon build`. Let's assume I already ran `colcon build`, hence every subsequent run of `colcon build` should be superfast. Indeed, running `colcon build` **without** sourcing ros2 finishes in **3.44s**. Running `colcon build` **after** sourcing ros2 takes **4min 19s**. Clearly, the sourced environment has a huge effect, and 12x longer compilation times are kind of unsatisfactory.

In this thread https://github.com/colcon/colcon-core/issues/193 they mention you shouldn't run `colcon build` in the sourced environment, but fresh compilations will fail since it can't find dependencies.

It also seems like this is a macOS specific issue, since I don't see this extreme slowdown on Ubuntu.

RSchwan, 2022-01-03:
I did some further investigations, and it turns out the environment variable `DYLD_LIBRARY_PATH` causes the slowdown. Running `unset DYLD_LIBRARY_PATH` fixes the slowdown, but this probably causes issues further down the road.

clalancette, 2022-01-04:
Yeah, this is a bit unfortunate.  This probably belongs linked to https://github.com/ros2/ros2/issues/1222 and https://github.com/ros2/ros2/issues/1148 for our general problems with ROS 2 on macOS.