ros2/ros2 #1233: Private action name does not work with composed nodes

AndyZe, 2021-12-30:
I made a fairly detailed [answers.ros.org question here](https://answers.ros.org/question/393575/in-ros2-action-name/). In short, the ~ that is supposed to prepend the node name to an action does not seem to work for Composed Nodes.

This makes it difficult (for example) to launch 2 action servers for a dual-arm robot.

fujitatomoya, 2022-01-03:
@AndyZe 

thanks for opening issue.

i am not really sure what exactly the problem is though. i might be missing something.

> /hybrid_planning_manager/hybrid_planning/local_planning_action
> /local_planner/hybrid_planning/local_planning_action
> The problem being local_planning_action has 2 different namespaces.

why this is the problem? given private namespace with `~`, it prepends the nodename before action name as expected?

AndyZe, 2022-01-04:
Sorry for being somewhat confused myself and writing a confusing issue.

I guess the real problem is, I don't see a convenient way to launch 2 actions servers as Composed Nodes and get the action namespacing correct.

I tried the `namespace` arg of the ComposableNode. I don't remember what effect it had exactly but it didn't namespace the action like I expected. (The client and the server should be namespaced the same way, IMO) I also tried the `namespace` arg of the individual ComposableNodes.

clalancette, 2022-01-20:
> I tried the `namespace` arg of the ComposableNode

This seems to work fine for me on Rolling (for topics):

```
import launch
from launch_ros.actions import ComposableNodeContainer
from launch_ros.descriptions import ComposableNode

def generate_launch_description():
    """Generate launch description with multiple components."""
    container = ComposableNodeContainer(
            name='image_container',
            namespace='',
            package='rclcpp_components',
            executable='component_container',
            composable_node_descriptions=[
                ComposableNode(
                    package='image_tools',
                    plugin='image_tools::Cam2Image',
                    name='cam2image',
                    namespace='ns',
                    parameters=[{'width': 320, 'height': 240, 'burger_mode': True}]),
                ComposableNode(
                    package='image_tools',
                    plugin='image_tools::ShowImage',
                    name='showimage',
                    namespace='ns')
            ],
            output='both',
    )

    return launch.LaunchDescription([container])
```

That is, when I launch that, the image generator node is called `/ns/cam2image` and the node that displays the image is called `/ns/showimage`.  Also the topic that they both use is called `/ns/image`.  Can you try the above launch and confirm that it works in your setup?  If it does, can you provide a further example of an action that is not working for you?

closed 2022-01-20 by AndyZe