ros2/ros2 #1234: Windows enhancement

OgreTransporter, 2022-01-11:
- Operating System:
  - Windows 10
- Installation type:
  - binaries and from source 
- Version or commit hash:
  - foxy and following versions
- DDS implementation:
  - default

## Feature request
I would be happy if the Windows setup for ROS2 is adapted a bit. There are many hard coded paths programmed in like Python (see #657). This is very problematic when working on machines without admin rights. Would it be possible to set all these paths via a configuration file? For example, I also need to use Chocolatey portable and use a different target directory than the Windows default.

clalancette, 2022-02-17:
This is definitely a problem in the current Windows setup.  It's hard to fix because of the way we build the binary tarballs for Windows.

We are working towards supporting Windows via conda packages instead.  That will get rid of this whole problem, and make Windows more like installing packages on Linux (`conda install ros-rolling-desktop`).  But we aren't there yet, and probably won't be for Humble.

michaelaeriksen, 2022-07-23:
Any plans to support Windows 11 and Visual Studio 2022? It would also be great if you supported `vcpkg` 

I have ROS2 building on Windows 11 + Visual Studio 2022 and vcpkg, but it requires a few hacks. It would be great if it was supported out of the box.

gongjun136, 2025-11-12:
> Any plans to support Windows 11 and Visual Studio 2022? It would also be great if you supported `vcpkg`
> 
> I have ROS2 building on Windows 11 + Visual Studio 2022 and vcpkg, but it requires a few hacks. It would be great if it was supported out of the box.

I also really hope to have support in this area. Have you solved this problem?

clalancette, 2026-10-26:
We switched to pixi now, so this should be done.  Please feel free to open a new issue if you are still having problems.

closed 2026-10-26 by clalancette