ros2/ros2 #1235: Topic remapping does not work in component nodes

AndyZe, 2022-01-19:
Distro: Rolling

I've been trying to remap the /joint_states topic for 2 "ComposableNodes" in a dual-arm system. The remapping does not seem to work.

The relevant bit of the launch file looks like this. Note the remapping.

```
            ComposableNode(
                package='moveit_hybrid_planning',
                plugin='moveit::hybrid_planning::LocalPlannerComponent',
                name='local_planner',
                namespace='/LeftArm',
                parameters=[
                    left_common_hybrid_planning_param,
                    left_local_planner_param,
                    robot_description,
                    robot_description_semantic,
                    bmlc.robot_description_kinematics,
                ],
                remappings=[
                    ("/LeftArm/joint_states", "/joint_states")
                ]
            ),
```

`ros2 node info /LeftArm/local_planner_private_140...` shows that the node is still subscribing to `/LeftArm/joint_states`. So I don't think the remapping has taken effect.

I have tried switching the order in the tuple like this, just to be sure. No luck.

```
                remappings=[
                    ("/joint_states", "/LeftArm/joint_states")
                ]
```

Screenshot attached.

[image]

AndyZe, 2022-01-19:
I'm not sure what the difference is between these 2 nodes:  /LeftArm/local_planner and /LeftArm/local_planner_1404999. But, `ros2 node info /LeftArm/local_planner` also is not helpful:

```
/LeftArm/local_planner
  Subscribers:
    /LeftArm/global_trajectory: moveit_msgs/msg/MotionPlanResponse
    /parameter_events: rcl_interfaces/msg/ParameterEvent
  Publishers:
    /LeftArm/position_traj_controller/joint_trajectory: trajectory_msgs/msg/JointTrajectory
    /parameter_events: rcl_interfaces/msg/ParameterEvent
    /rosout: rcl_interfaces/msg/Log
  Service Servers:
    /LeftArm/load_map: moveit_msgs/srv/LoadMap
    ...
```

clalancette, 2022-01-19:
This launch file remaps just fine for me on Rolling:

```
import launch
from launch_ros.actions import ComposableNodeContainer
from launch_ros.descriptions import ComposableNode

def generate_launch_description():
    """Generate launch description with multiple components."""
    container = ComposableNodeContainer(
            name='image_container',
            namespace='',
            package='rclcpp_components',
            executable='component_container',
            composable_node_descriptions=[
                ComposableNode(
                    package='image_tools',
                    plugin='image_tools::Cam2Image',
                    name='cam2image',
                    remappings=[('/image', '/burgerimage')],
                    parameters=[{'width': 320, 'height': 240, 'burger_mode': True}]),
                ComposableNode(
                    package='image_tools',
                    plugin='image_tools::ShowImage',
                    name='showimage',
                    remappings=[('/image', '/burgerimage')])
            ],
            output='both',
    )

    return launch.LaunchDescription([container])
```

```
$ ros2 topic list
/burgerimage
/flip_image
/parameter_events
/rosout
```

Can you try that launch file out on your system?

AndyZe, 2022-01-20:
That does work. OK, I'll just close the issue unless I can come up with a better test. (It's a weird system with a mix of ROS1 and ROS2 in Docker containers.)

AndyZe, 2022-01-20:
Do you think this could have something to do with my component node having a namespace? Maybe the remapping should be something like:

```
                remappings=[
                    ("joint_states", "/joint_states")
```
                    
                    or
                    
```
                remappings=[
                    ("~/joint_states", "/joint_states")
```
                    
To account for the namespace.

clalancette, 2022-01-20:
> Do you think this could have something to do with my component node having a namespace? Maybe the remapping should be something like:

Yes.  Remappings are applied *after* the namespacing.  So if you want to both remap and namespace, you need to do:

```
    namespace='ns',
    remappings=[('/ns/joint_states', '/remapped_joint_states')]
```

AndyZe, 2022-01-20:
Yes, that works

clalancette, 2022-01-20:
OK.  At this point, this seems like a configuration issue somewhere.  I'm going to close this out, and suggest you re-ask the question on https://answers.ros.org with a lot of specifics on what you are trying to do and what is failing.  Hopefully someone can point out where things are going wrong.

closed 2022-01-20 by clalancette

AndyZe, 2022-01-20:
To bring some closure to this, it works properly when the namespace of the component container is removed. Don't ask me why.

It worked fine with @clalancette 's simple examples whether there was a component container namespace or not.

¯\_(ツ)_/¯

```
    left_hybrid_planning_container = ComposableNodeContainer(
        name='hybrid_planning_container',
        namespace='/LeftArm', --> ''
```

AndyZe, 2022-01-20:
Another thing I noticed is that after deleting the namespace of the ComposableNodeContainer, this node disappeared:

`/LeftArm/local_planner_private_1404999`

zflat, 2022-01-20:
It looks like there is some relation where the topics need to be remapped in the container node as well as the composable node?

clalancette, 2022-01-22:
> It looks like there is some relation where the topics need to be remapped in the container node as well as the composable node?

No, that shouldn't be.  The container node has no topics; they all belong to the composable nodes.