ros2/ros2 #1236: Could not find a package configuration file provided by "ament_cmake"

nikboy111, 2022-01-23:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 20.04
- Installation type:
  - ROS2 Foxy following the offical documentation and TurtleBot3 following the emanual: https://emanual.robotis.com/docs/en/platform/turtlebot3/quick-start/#pc-setup

#### Expected behavior
Successful build of turtlebot3 packages.

#### Actual behavior
After building the package I'm having the following error:

```
Starting >>> dynamixel_sdk
Starting >>> turtlebot3_msgs
Starting >>> turtlebot3_description
Starting >>> dynamixel_sdk_custom_interfaces
--- stderr: dynamixel_sdk                                          
CMake Error at CMakeLists.txt:18 (find_package):
  By not providing "Findament_cmake.cmake" in CMAKE_MODULE_PATH this project
  has asked CMake to find a package configuration file provided by
  "ament_cmake", but CMake did not find one.

  Could not find a package configuration file provided by "ament_cmake" with
  any of the following names:

    ament_cmakeConfig.cmake
    ament_cmake-config.cmake

  Add the installation prefix of "ament_cmake" to CMAKE_PREFIX_PATH or set
  "ament_cmake_DIR" to a directory containing one of the above files.  If
  "ament_cmake" provides a separate development package or SDK, be sure it
  has been installed.

---
Failed   <<< dynamixel_sdk [0.11s, exited with code 1]
Aborted  <<< dynamixel_sdk_custom_interfaces [0.10s]
Aborted  <<< turtlebot3_description [0.10s]
Aborted  <<< turtlebot3_msgs [0.10s]
```
After this section i couldn't build the package:

> git clone -b foxy-devel https://github.com/ROBOTIS-GIT/DynamixelSDK.git
> git clone -b foxy-devel https://github.com/ROBOTIS-GIT/turtlebot3_msgs.git
> git clone -b foxy-devel https://github.com/ROBOTIS-GIT/turtlebot3.git
> cd ~/turtlebot3_ws
> colcon build --symlink-install

Searching lots of different troubleshoot like [this ](https://github.com/ros2/ros2/issues/710) always got me the advice of having to source the /opt/ros/foxy/setup.bash or */setup.sh, but this doesn't help me fix the problem. 
If I source the setup.sh I'm getting this error:
```
Starting >>> dynamixel_sdk
Starting >>> turtlebot3_msgs
Starting >>> turtlebot3_description
Starting >>> dynamixel_sdk_custom_interfaces
--- stderr: turtlebot3_msgs                                        
CMake Error at CMakeLists.txt:18 (find_package):
  By not providing "Findaction_msgs.cmake" in CMAKE_MODULE_PATH this project
  has asked CMake to find a package configuration file provided by
  "action_msgs", but CMake did not find one.

  Could not find a package configuration file provided by "action_msgs" with
  any of the following names:

    action_msgsConfig.cmake
    action_msgs-config.cmake

  Add the installation prefix of "action_msgs" to CMAKE_PREFIX_PATH or set
  "action_msgs_DIR" to a directory containing one of the above files.  If
  "action_msgs" provides a separate development package or SDK, be sure it
  has been installed.

---
Failed   <<< turtlebot3_msgs [0.11s, exited with code 1]
Aborted  <<< turtlebot3_description [0.10s]
Aborted  <<< dynamixel_sdk [0.12s]
Aborted  <<< dynamixel_sdk_custom_interfaces [0.10s]
```
I don't know what else to do. Thank you so much in advance! 
Regards Nik

clalancette, 2022-01-24:
@nikboy111 Please open a question on https://answers.ros.org, which is our central question and answer site.  You should be able to get help there.

closed 2022-01-24 by clalancette

admajors00, 2022-11-04:
I had the exact same problem I found the answer here that worked.

https://www.reddit.com/r/ROS/comments/vrvyim/errno_13/[Reddit](https://www.reddit.com/r/ROS/comments/vrvyim/errno_13/) 

"This will happen mostly because you do not have permissions to edit the workspace directory. Try giving read/write permission of the workspace to all users using

```
cd <your workspace>
 sudo chmod 777 -R .
```
Then build using colcon build in the current directory."

Magendreh, 2023-05-07:
I had the same issue with ROS2 humble on Windows 10.
The directory of my humble installation was the set to read-only. 
After changing it, the error was gone.

jinyin-z19, 2025-01-09:
Thanks a lot, [admajors00](https://github.com/admajors00)!