ros2/ros2 #1238: [Post update error (foxy->galactic)] Invalid port mapping: port number(s) of out range

merosss, 2022-02-04:
## Bug report
I just updated my ROS2 version, from foxy to galactic, and I am getting this error when trying to run any of my nodes (in this case my node is called "_kinematic_control_node_"):

`1643896270.439717 [0] kinematic: Invalid port mapping: port number(s) of out range: multicast discovery 0000000000, multicast data 0000000000, unicast discovery 0000000000, unicast data 0000000000`

It looks to me like a DDS problem. I already checked if the multicast was working correctly with `ros2 multicast send/receive` and I had no problems.

I cannot find anything online regarding this problem, and it seems quite obscure to me, any hints?

**The only thing that solves the problem is rolling back to FastRTPS DDS.**

_**Code**_ (the `VehicleController` inherits from `rclcpp::Node`):

```cpp
int main(int argc, char* argv[])
{
    rclcpp::init(argc, argv);

    std::string filename = "kcl_ulisse.conf";
    auto vehicleController = std::make_shared<ulisse::VehicleController>(filename);

    rclcpp::executors::MultiThreadedExecutor exe;
    exe.add_node(vehicleController);
    exe.spin();

    rclcpp::shutdown();

    return 0;
}
```

_**GDB error stack**_:

```sh
Thread 1 "kinematic_contr" received signal SIGSEGV, Segmentation fault.
0x00007ffff6f3e0e3 in config_fini () from /opt/ros/galactic/lib/x86_64-linux-gnu/libddsc.so.0
(gdb) backtrace
#0  0x00007ffff6f3e0e3 in config_fini () from /opt/ros/galactic/lib/x86_64-linux-gnu/libddsc.so.0
#1  0x00007ffff6f82a6f in ?? () from /opt/ros/galactic/lib/x86_64-linux-gnu/libddsc.so.0
#2  0x00007ffff6f83010 in dds_create_domain () from /opt/ros/galactic/lib/x86_64-linux-gnu/libddsc.so.0
#3  0x00007ffff70334aa in ?? () from /opt/ros/galactic/lib/librmw_cyclonedds_cpp.so
#4  0x00007ffff7033ee8 in rmw_create_node () from /opt/ros/galactic/lib/librmw_cyclonedds_cpp.so
#5  0x00007ffff7d782e4 in rcl_node_init () from /opt/ros/galactic/lib/librcl.so
#6  0x00007ffff7ed7738 in rclcpp::node_interfaces::NodeBase::NodeBase(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::shared_ptr<rclcpp::Context>, rcl_node_options_t const&, bool, bool) ()
   from /opt/ros/galactic/lib/librclcpp.so
#7  0x00007ffff7ecfa24 in rclcpp::Node::Node(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, rclcpp::NodeOptions const&) () from /opt/ros/galactic/lib/librclcpp.so
#8  0x00007ffff7ed11f8 in rclcpp::Node::Node(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, rclcpp::NodeOptions const&) () from /opt/ros/galactic/lib/librclcpp.so
#9  0x00005555555d464d in ulisse::VehicleController::VehicleController (this=0x5555558f4030, 
    conf_filename="kcl_ulisse.conf")
    at /home/wanderfra/ros2_ws/src/ulisse_core/ulisse_ctrl/src/ulisse_ctrl/kinematic_vehicle_controller.cpp:22
#10 0x00005555555d3fab in __gnu_cxx::new_allocator<ulisse::VehicleController>::construct<ulisse::VehicleController, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >&> (this=0x7fffffffd38f, 
    __p=0x5555558f4030) at /usr/include/c++/9/ext/new_allocator.h:147
#11 0x00005555555d3d92 in std::allocator_traits<std::allocator<ulisse::VehicleController> >::construct<ulisse::VehicleController, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >&> (__a=..., 
    __p=0x5555558f4030) at /usr/include/c++/9/bits/alloc_traits.h:484
#12 0x00005555555d3a5a in std::_Sp_counted_ptr_inplace<ulisse::VehicleController, std::allocator<ulisse::VehicleController>, (__gnu_cxx::_Lock_policy)2>::_Sp_counted_ptr_inplace<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >&> (this=0x5555558f4020, __a=...) at /usr/include/c++/9/bits/shared_ptr_base.h:548
--Type <RET> for more, q to quit, c to continue without paging--
#13 0x00005555555d352a in std::__shared_count<(__gnu_cxx::_Lock_policy)2>::__shared_count<ulisse::VehicleController, std::allocator<ulisse::VehicleController>, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >&> (this=0x7fffffffd508, __p=@0x7fffffffd500: 0x0, __a=...) at /usr/include/c++/9/bits/shared_ptr_base.h:679
#14 0x00005555555d2f92 in std::__shared_ptr<ulisse::VehicleController, (__gnu_cxx::_Lock_policy)2>::__shared_ptr<std::allocator<ulisse::VehicleController>, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >&> (this=0x7fffffffd500, __tag=...) at /usr/include/c++/9/bits/shared_ptr_base.h:1344
#15 0x00005555555d2767 in std::shared_ptr<ulisse::VehicleController>::shared_ptr<std::allocator<ulisse::VehicleController>, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >&> (this=0x7fffffffd500, 
    __tag=...) at /usr/include/c++/9/bits/shared_ptr.h:359
#16 0x00005555555d2383 in std::allocate_shared<ulisse::VehicleController, std::allocator<ulisse::VehicleController>, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >&> (__a=...)
    at /usr/include/c++/9/bits/shared_ptr.h:702
#17 0x00005555555d1e66 in std::make_shared<ulisse::VehicleController, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >&> () at /usr/include/c++/9/bits/shared_ptr.h:718
#18 0x00005555555d02cd in main (argc=1, argv=0x7fffffffd6c8)
    at /home/wanderfra/ros2_ws/src/ulisse_core/ulisse_ctrl/src/kinematic_control_node.cpp:13

```

**Required Info:**

- Operating System:
  - Ubuntu 20.04 (tested also inside VirtualBox6 on a Win10 host machine, same behaviour).
- Installation type:
  - Binaries
- DDS implementation:
  - CycloneDDS
- Client library (if applicable):
  - rclcpp

clalancette, 2022-02-04:
Yeah, agreed that it looks like some kind of DDS problem.  Are you using a custom configuration for CycloneDDS?

merosss, 2022-02-04:
I don't think so, I never tinkered with DDS parameters.
I was only using QoS sensor configuration  in some Subscribers (`rclcpp::QoSInitialization::from_rmw(rmw_qos_profile_sensor_data)`, but to get QoS problems out of the way I removed all of them, leaving a simple integer for the queue size, but the problem persisted.

clalancette, 2022-02-04:
Hm, that's a really weird one then.  @eboasson Any thoughts from you on what might be happening?

eboasson, 2022-02-04:
> 1643896270.439717 [0] kinematic: Invalid port mapping: port number(s) of out range: multicast discovery 0000000000, multicast data 0000000000, unicast discovery 0000000000, unicast data 0000000000

Port numbers are 7400 + 250 *domainId + ..., and those constants are all configurable. Almost no-one changes the constants, and if it loads the configuration without error, then it should — clearly "should" is applicable here — not end up with everything equal to 0 (it prints so many zeros because that was the easiest way of handling 65 bits arithmetic). Furthermore, normally if they are explicitly set to 0, you get this error message but not the crash.

I've seen the combination of this error message and segmentation fault only when starting Cyclone with a binary configuration that is all-zero. The function you need for that isn't called by the RMW layer ...

So yes, it is quite mysterious. Is there any chance you can reproduce it in a debugger so we can see a little bit more of what's happening, what its state is, and where it is happening?

gavanderhoorn, 2022-02-04:
ABI problem?

@merosss mentions "updating [..] from foxy to galactic" (also on [ROS Answers](https://answers.ros.org/question/395566), but it's unclear whether the workspace was also cleaned and rebuilt.

Seeing `SEGFAULT`s in gdb is often the case.

merosss, 2022-02-04:
I will try to pack a minimal example to reproduce the issue asap.

For the ABI suggestion I deleted and rebuilt all the workspace multiple times after updating. I also cleaned the whole workspace after setting the RMW_IMPLEMENTATION variable, even if I don't actually know how this setting influences the building phase (but I presume it doesn't).

clalancette, 2026-10-26:
There's been no follow-up here for almost 5 years, so I'm assuming this is no longer an issue.  Closing, but if you run into an issue again please feel free to open a new issue.

closed 2026-10-26 by clalancette