ros2/ros2 #1239: Failed   <<< rosidl_typesupport_connext_cpp

smitalpatil234, 2022-02-04:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 18.04
- Installation type:
  - binaries
- Version or commit hash:
  - ROS2
- DDS implementation:
  - Fast-RTPS 
- Client library (if applicable):
  - rclcpp

#### Steps to reproduce issue

```

```

#### Expected behavior
Complete installation of all ROS2 packages
#### Actual behavior
/usr/bin/ld: skipping incompatible /opt/rti_connext_dds-6.0.1/lib/armv8Linux4.4gcc5.4.0/libnddscd.so when searching for -lnddscd
/usr/bin/ld: cannot find -lnddscd
/usr/bin/ld: skipping incompatible /opt/rti_connext_dds-6.0.1/lib/armv8Linux4.4gcc5.4.0/libnddscored.so when searching for -lnddscored
/usr/bin/ld: cannot find -lnddscored
/usr/bin/ld: skipping incompatible /opt/rti_connext_dds-6.0.1/lib/armv8Linux4.4gcc5.4.0/libnddscppd.so when searching for -lnddscppd
/usr/bin/ld: cannot find -lnddscppd
/usr/bin/ld: skipping incompatible /opt/rti_connext_dds-6.0.1/lib/armv8Linux4.4gcc5.4.0/librticonnextmsgcppd.so when searching for -lrticonnextmsgcppd
/usr/bin/ld: cannot find -lrticonnextmsgcppd
collect2: error: ld returned 1 exit status
make[2]: *** [librosidl_typesupport_connext_cpp.so] Error 1
make[1]: *** [CMakeFiles/rosidl_typesupport_connext_cpp.dir/all] Error 2
make: *** [all] Error 2
---
**Failed   <<< rosidl_typesupport_connext_cpp [4.43s, exited with code 2]
Aborted  <<< rosidl_typesupport_introspection_cpp [1.34s]
Aborted  <<< spdlog_vendor [17.7s]**
                                   
Summary: 41 packages finished [59.4s]
  1 package failed: rosidl_typesupport_connext_cpp
  2 packages aborted: rosidl_typesupport_introspection_cpp spdlog_vendor
  41 packages had stderr output: ament_cmake ament_cmake_core ament_cmake_export_definitions ament_cmake_export_dependencies ament_cmake_export_include_directories ament_cmake_export_interfaces ament_cmake_export_libraries ament_cmake_export_link_flags ament_cmake_export_targets ament_cmake_gmock ament_cmake_gtest ament_cmake_include_directories ament_cmake_libraries ament_cmake_pytest ament_cmake_python ament_cmake_ros ament_cmake_target_dependencies ament_cmake_test ament_cmake_version ament_index_python ament_package connext_cmake_module domain_coordinator libyaml_vendor rcutils rmw_implementation_cmake rosidl_adapter rosidl_cmake rosidl_generator_c rosidl_generator_dds_idl rosidl_parser rosidl_runtime_c rosidl_runtime_cpp rosidl_typesupport_connext_cpp rosidl_typesupport_interface rosidl_typesupport_introspection_c spdlog_vendor test_interface_files tracetools_launch tracetools_read tracetools_trace
  53 packages not processed
Colcon build failed
Return code is 2
Traceback (most recent call last):
  File "/usr/local/lib/python3.6/dist-packages/conans/client/command.py", line 2170, in run
    method(args[0][1:])
  File "/usr/local/lib/python3.6/dist-packages/conans/client/command.py", line 382, in create
    profile_build=profile_build)
  File "/usr/local/lib/python3.6/dist-packages/conans/client/conan_api.py", line 93, in wrapper
    return f(api, *args, **kwargs)
  File "/usr/local/lib/python3.6/dist-packages/conans/client/conan_api.py", line 387, in create
    test_build_folder, test_folder, conanfile_path, recorder=recorder)
  File "/usr/local/lib/python3.6/dist-packages/conans/client/cmd/create.py", line 91, in create
    recorder=recorder)
  File "/usr/local/lib/python3.6/dist-packages/conans/client/manager.py", line 81, in deps_install
    graph_lock, keep_build=keep_build)
  File "/usr/local/lib/python3.6/dist-packages/conans/client/installer.py", line 314, in install
    graph_lock, remotes, build_mode, update)
  File "/usr/local/lib/python3.6/dist-packages/conans/client/installer.py", line 448, in _build
    self._handle_node_cache(node, keep_build, processed_package_refs, remotes)
  File "/usr/local/lib/python3.6/dist-packages/conans/client/installer.py", line 510, in _handle_node_cache
    pref = self._build_package(node, output, keep_build, remotes)
  File "/usr/local/lib/python3.6/dist-packages/conans/client/installer.py", line 549, in _build_package
    pref = builder.build_package(node, keep_build, self._recorder, remotes)
  File "/usr/local/lib/python3.6/dist-packages/conans/client/installer.py", line 223, in build_package
[... 1232 lines cut ...]
  File "/usr/local/lib/python3.6/dist-packages/conans/client/installer.py", line 549, in _build_package
    pref = builder.build_package(node, keep_build, self._recorder, remotes)
  File "/usr/local/lib/python3.6/dist-packages/conans/client/installer.py", line 223, in build_package
    self._build(conanfile, pref)
  File "/usr/local/lib/python3.6/dist-packages/conans/client/installer.py", line 145, in _build
    run_build_method(conanfile, self._hook_manager, reference=pref.ref, package_id=pref.id)
  File "/usr/local/lib/python3.6/dist-packages/conans/client/conanfile/build.py", line 16, in run_build_method
    conanfile.build()
  File "/home/host/.conan/data/rcl/3.1.4/lib/tag/export/conanfile.py", line 174, in build
    sys.exit()
SystemExit

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/local/bin/conan", line 8, in <module>
    sys.exit(run())
  File "/usr/local/lib/python3.6/dist-packages/conans/conan.py", line 11, in run
    main(sys.argv[1:])
  File "/usr/local/lib/python3.6/dist-packages/conans/client/command.py", line 2340, in main
    error = command.run(args)
  File "/usr/local/lib/python3.6/dist-packages/conans/client/command.py", line 2177, in run
    self._out.error("Exiting with code: %d" % exc.code)
TypeError: %d format: a number is required, not NoneType
`

clalancette, 2022-02-07:
We don't support Connext 6.0.1 on Ubuntu 18.04, or in ROS 2 Foxy, or on armv8.  You can see what we support here: https://www.ros.org/reps/rep-2000.html#foxy-fitzroy-may-2020-may-2023 , and we only support 5.3.1.

That said, we are planning on support Connext 6.0.1 with ROS 2 Humble.  So if that is what you need, you'll have better luck with Humble (which will be released in May).

I'm going to close this out for now, but feel free to keep commenting.

closed 2022-02-07 by clalancette

smitalpatil234, 2022-02-10:
Please delete this issue
It was posted by mistake