ros2/ros2 #1242: Very slow publishing of large messages

xander-m2k, 2022-02-08:
## Bug report

**Required Info:**

- Operating System:
  - Debian Buster
- Installation type:
  - Binaries (compiled from scratch) -> just compiling ROS with colcon should result in exactly the same ROS distro version.
- Version or commit hash:
  - Foxy/ros2.repos
- DDS implementation:
  - Fast-RTPS/DDS
- Client library (if applicable):
  - N/A

#### Steps to reproduce issue

```
#include "rclcpp/rclcpp.hpp"
#include "sensor_msgs/msg/image.hpp"
#include <queue>
#include "cv_bridge/cv_bridge.h"
#include <opencv2/opencv.hpp>

using namespace std::chrono_literals;

class ImagePublisher : public rclcpp::Node
{
private:
    rclcpp::Publisher<sensor_msgs::msg::Image>::SharedPtr image_pub_;
    rclcpp::TimerBase::SharedPtr timer_;
    std::thread pub_thread;
    std::queue<cv::Mat> imageQ;
    std::mutex im_mtx;

public:
    ImagePublisher() : Node("image_pub")
    {
        image_pub_ = create_publisher<sensor_msgs::msg::Image>("image_pub", 10);
        pub_thread = std::thread(std::bind(&ImagePublisher::publisher, this));
        timer_ = this->create_wall_timer(500ms, std::bind(&ImagePublisher::timer_callback, this));
        rclcpp::shutdown();
    }

    void publisher(){
        while(rclcpp::ok()){
            std::this_thread::sleep_for(std::chrono::milliseconds(1));
            if(!imageQ.empty()){
                RCLCPP_DEBUG_STREAM(rclcpp::get_logger("image_pub"),"[ImagePub]: Received data in publisher ");
                im_mtx.lock();
                cv::Mat img = imageQ.front();
                imageQ.pop();
                im_mtx.unlock();
                cv_bridge::CvImage img_bridge;

                sensor_msgs::msg::Image::SharedPtr msg(new sensor_msgs::msg::Image());
                msg =  cv_bridge::CvImage(msg->header, sensor_msgs::image_encodings::BGR8, img).toImageMsg();
                image_pub_->publish(*msg.get());
            }
        }
    }

    void timer_callback(){
        cv::Mat image = cv::imread("<file_location>", cv::IMREAD_COLOR);
        {
            std::lock_guard<std::mutex> lck(im_mtx);
            imageQ.push(image);
        }
    }

};

int main(int argc, char **argv){
    rclcpp::init(argc, argv);
    auto image_pub = std::make_shared<ImagePublisher>();
    rclcpp::spin(image_pub);

    rclcpp::shutdown();
    std::cout<<"shut down is called"<< std::endl;
    return 1;
}
```
I normally have this in a .h/.cpp architecture, but I have rewritten it a bit for this issue, so it fits in one file.
At <file_location> set the location for your image. The image should be about ~5MB (roughly 2000px * 2000px in 8-bit).

#### Expected behavior
Image should be published immediately, without delay.
#### Actual behavior
I have large delays with the publishing of these images. It takes about 30-60 seconds for the images to get published. Something is really slowing down, and I cannot find out what it is.
#### Additional information
Things I already have tried:
- Using CycloneDDS, results in the same
- Increasing the UDP memory buffer of Linux, same result
- Use other QoS profiles (history_depth = 1000 || rclcpp::SensorDataQoS() || best effort), played around with those a lot, because I thought that was where the issue is, turns out it does not affect anything.
- Shrinking the image size by 10x, now the image gets published immediately, no problems with an image of about 500kB.
- Prior to this code we had written this in Python, rewritten it to C++, did not solve any issues.

I am kind of running out of options here
----
## Feature request

#### Feature description

#### Implementation considerations

xander-m2k, 2022-02-08:
> To add to what @fujitatomoya mentioned, and just to make sure, have you tried measuring latency/throughput without using `ros2 topic`? E.g., using a C++ subscription. Take a look at [this discussion](https://discourse.ros.org/t/ros2-speed/20162), especially [this comment](https://discourse.ros.org/t/ros2-speed/20162/19).

As a quick reply to this message:
Yes I have. When the message do come through I notice a massive decrease in publishing speed when subscribing to it with an actual node. Latency was fairly decent when it worked, around 0,3s. But most of the time the message does not even come through.

The other message you sent I need to take a deeper look at tomorrow!

xander-m2k, 2022-02-08:
> @xander-m2k thanks for making the issue.
> 
> Does this publish the data via UDP, is this requirement for application? if that can be okay with intra-process or inter-process in the localhost, there would be a couple of ways to have more performance.
> 
> * [Efficient intra-process communication](https://docs.ros.org/en/foxy/Tutorials/Intra-Process-Communication.html) available on Foxy. this one is agnostic from rmw implementation since this is supported on rclcpp.
> * [ROS 2 Zero Copy](https://design.ros2.org/articles/zero_copy.html), unfortunately Fast-DDS Data Sharing is not supported with Foxy, but Galactic. If your requirement is with Foxy, this could not be an option.
> * (ROS 2 CLI cannot use both of them described above, since it is based on rclpy python implementation)
> 
> besides, rmw_cyclonedds can support zero copy as well, https://github.com/ros2/rmw_cyclonedds/blob/master/shared_memory_support.md. but this is also supported from Galactic or later.
> 
> hope this helps you a bit! Best,

I am publishing the data via UDP to another system. The process is not only on the localhost. 
Thanks a lot for this information though, these are interesting remarks.
Unfortunately for my application, the communication is between two operating systems, connected via UDP.
I already mentioned that I tried altering the UDP buffer settings, but I will clarify this more with that I am using two OS'es.

xander-m2k, 2022-02-09:
I have debugged a lot more by now and discovered the following things:
- The rest of my program was slightly interrupting publishing, but I managed to fix that. Did not solve any issue however on the subscribing part (no data is being received on the other OS).
- I have gradually increased the image size with factors of 0.1x and discovered that the limit lies at around 807 x 1076, which is roughly ~2.5MB (2604996 bytes). The localhost is fine, all messages are received when subscribing to it locally. I am only having problems with receiving messages over UDP. 
- Packets are received on the other OS side (looked into that with wireshark), but it looks like after 2.5MB the full UDP packet/datapacket transfer is not able to complete and reassemble fully.

EduPonz, 2022-02-09:
Hi @xander-m2k ,

I guess you are using RELIABLE QoS, right? Sometimes with large RTPS datagrams spanning across several UDP datagrams (that is, a lot of RTPS fragments, each taking a full 65 kB UDP) you may run into problems if the underlying IP frame are getting lost, since losing a single IP frame would mean losing a full 65 kB RTPS frag, meaning that the reliability will kick in for that fragment. In that scenario if you're consistently losing at least one IP frame per UDP datagram, then you'll never receive anything at all. Can you verify with Wireshark that you're not receiving all the IP frames for the UDP datagrams?

In that case, it usually helps to set the `maxMessageSize` of the transport layer to something a bit smaller than the MTU (see [here](https://fast-dds.docs.eprosima.com/en/latest/fastdds/xml_configuration/transports.html#transport-descriptors) and [here](https://fast-dds.docs.eprosima.com/en/latest/fastdds/ros2/ros2_configure.html)). This way, you end up with tons of very small RTPS fragments, but each of them only takes one IP frame, so if you lose an IP frame (one RTPS frag), the reliability will take care of it, and it is less likely that you consistently lose the same IP frame. This will of course be less performative than an ideal network with no losses, but at least you'll be able to receive something.

xander-m2k, 2022-02-09:
Thanks a lot @EduPonz! I will try again tomorrow!

xander-m2k, 2022-02-10:
@EduPonz thank you so much for helping, that did the job! I did however change the publish mode to `ASYNCHRONOUS`, just so I did not have to use an .xml file on the other OS.

xander-m2k, 2022-02-10:
Closing because the issue is solved.

EduPonz, 2022-02-12:
Hi @xander-m2k ,

I do not see how the configuration would affect the RPC mechanism. 

> does this behaviour improve after the ROS Foxy version?

Fast DDS in general improves with every version, both performance and feature wise. That being said, mind that the fact that the XML made communication possible means that the problem did not originate in Fast DDS but in your network instead. What you were experiencing was that your receiver machine could not reconstruct any UDP datagram, meaning that, somewhere in your network, you were loosing IP frames with every single UDP datagram. Under those circumstances it is impossible to get the RTPS reliability to work, since every RTPS fragment that is resent cannot be reconstructed on the other side. What the XML does is:

* Limiting the size of an RTPS fragment to something that fits in a single IP frame. This way, when an IP frame is lost (an it seems in your network that is likely going to happen), the RTPS reliability can effectively solve the problem, as the probability of losing the same fragment more than once or twice is very small
* The previous has the side effect that the middleware delivers data at a slower pace (since it needs to break each of your samples in more fragments), which may prevent buffer overrides in you network equipment.

In general, this problem of losing IP frames means that some network buffer is being overridden. This can happen in your machines or in any network link in the middle (router, switches). For example, we've observed this behaviour when sending on a machine using a 10 Gb/s interface and receiving in a 1 Gb/s interface. In that case, the 1 Gb/s output port on the switch was being overflowed with the data from the 10 Gb/s input port, mainly because the data was moved from the 10 Gb/s to the 1 Gb/s at a faster rate than what the 1 Gb/s could handle.

Without more information, it is difficult to say something more concrete; you could take a look at wireshark traces from both sending and receiving ends looking for the data corresponding to the service topics (request and replay) to try to assess whether the issue is the same or not. In any case, Fast DDS documentation has a use-case for [large data](https://fast-dds.docs.eprosima.com/en/v2.1.0/fastdds/use_cases/large_data/large_data.html?highlight=flow%20controller#large-data-rates) that may come in handy, specially the section about [Flow Controllers](https://fast-dds.docs.eprosima.com/en/v2.1.0/fastdds/use_cases/large_data/large_data.html?highlight=flow%20controller#flow-controllers), which limit the amount of data that can be published in a given period. Mind that in the issue I described in the previous paragraph, putting less data per time unit in the network may yield better results.

A completely different approach to solving the issue would be to use a different network transport, namely [TCP](https://fast-dds.docs.eprosima.com/en/v2.1.0/fastdds/transport/tcp/tcp.html), which you can also configure using XML. To ease the discovery setup (avoid setting initial peers everywhere), you could leverage the [Discovery Server](https://fast-dds.docs.eprosima.com/en/v2.1.0/fastdds/discovery/discovery_server.html#discovery-server-settings) configured to use TCP as transport layer. To further ease the configuration, you may find interesting to read about the eProsima's [DDS Router](https://eprosima-dds-router.readthedocs.io/en/latest/), more specifically the [ROS 2 Discovery Server example](https://eprosima-dds-router.readthedocs.io/en/latest/rst/examples/ros_discovery_server_example.html#ros2-discovery-server-example). By using TCP as transport layer, you let the underlying protocol (TCP) take care of the reliability and the adjusting of the publication pace.

Be advice that both flow controllers, TCP transport, Discovery Server, and DDS Router are features/products you'd not find outside Fast DDS.

I know I've added a lot of reading material 😖, I really think that some of these options will do the trick for you!

xander-m2k, 2022-02-12:
Thankyou @EduPonz,

I'm don't know if I am allowed to talk about the product I am using, because it's for the company I work for.
Basically, the connection is an Ethernet PCI bus between two OS'es running on a real-time hypervisor. It's hard to believe that it is related to the hardware because they are configured exactly the same on both sides, and it is real-time. I did notice though, that the last frame (with using the standard 65kB) was not reassembled (at least it did not say so in the Wireshark capture), so it has to be a buffer. The weird thing is, I increased pretty much every UDP buffer there is (or at least the ones that can be configured), on both sides, but this didn't do anything for me.

As you say, if the XML made the communication possible, it has to originate from my hardware/software config outside of Fast DDS. I did solve it by just separating this Node into another (Linux) `.service` file. But, this is an important remark, because I now know the problem is on my side and not the DDS side.

Thank you for your time, advice and for providing the alternatives that are still possible, I will investigate what works best for me.

I will close this issue now.

closed 2022-02-12 by xander-m2k

monticle, 2024-07-17:
using a correct gateway address solved my problem