ros2/ros2 #1243: Symlink-install created softlinks in build folders instead of install, why?

TheConstructAi, 2022-02-09:
## Bug report

**Required Info:**

- Operating System:
Ubuntu 20
- Installation type:
bin / ROS2 Galactic
- DDS implementation:
  - Cyclone

#### Steps to reproduce the issue

mkdir -p ~/ros2_ws/src
cd ~/ros2_ws/src
git clone https://bitbucket.org/theconstructcore/ros2_build_tests.git
cd ~/ros2_ws
source /opt/ros/galactic/setup.bash
colcon build --symlink-install
source install/setup.bash
ros2 run build_test_pkg script1_exe

#### Expected behavior

I would expect that the symlinks are inside the install folder which is the one form where we use the files

#### Actual behavior

It's sourcing the build folder, which I don't understand its use then

#### Additional information

## Feature request

That symlink-install is what is advertised, which is to create symlinks of all the files at least the ones stated in the setup.py

#### Feature description
It would solve a lot of confusion and not have to compile each time we change a YAML file or a launch file

clalancette, 2022-02-09:
This does seem weird, but I'm not all that familiar with how this is supposed to work.  Regardless, this seems like a problem in colcon itself so I'll suggest opening an issue over at https://github.com/colcon/colcon-core.

Since we can't transfer issues between organizations, I'm going to close this out.

closed 2022-02-09 by clalancette

TheConstructAi, 2022-02-10:
Thganks @clalancette , I've posted the issue here: https://github.com/colcon/colcon-core/issues/482

Just for all the users that probably have this same issue.

miguelriemoliveira, 2023-01-27:
I just read through this issue. I have the same problem.

Without the symlinks working properly, we are forced to have to recompile even when we just changed a yaml, or rviz configuration file, or some python code.

Coming from ROS1, I must say it is really annoying and throws away one of the big advantages of developing in python vs cpp, which is not having to wait for the compilation to finish in order to test the changes in the code.

I hope there is a solution to this. If I find one I will post it here.