ros2/ros2 #1244: Foxy, XML launch files: set_env behaviour changed in last release

Neofish22, 2022-02-16:
## Bug report

Latest version of Foxy introduced various changes to XML launch files, most of which had definite errors when launched, but `set_env` behaviour seems to have changed. It now "leaks" - `set_env` now affects places that are sequentially before the tag, including overriding those that are set deeper. Now one can only override the last `set_env` with an `env` in `node`/`executable`.

**Required Info:**

When having bug (corresponds to actual result):
- Operating System:
  - `Linux taco 5.13.0-28-generic #31~20.04.1-Ubuntu SMP Wed Jan 19 14:08:10 UTC 2022 x86_64 x86_64 x86_64 GNU/Linux`
- Version or commit hash:
  - Foxy, might be release 8. Packages: [ros-foxy-package-versions.txt](https://github.com/ros2/ros2/files/8069144/ros-foxy-package-versions.txt)

When not having bug (corresponds to expected result)
- Operating System:
  - `Linux xyzkvh 5.13.0-28-generic #31~20.04.1-Ubuntu SMP Wed Jan 19 14:08:10 UTC 2022 x86_64 x86_64 x86_64 GNU/Linux`
- Version or commit hash:
  - Foxy, any release before current. Packages: [ros-foxy-pkg-working.txt](https://github.com/ros2/ros2/files/8077526/ros-foxy-pkg-working.txt)
- Note: These come from a colleague who has a quite old system, but I had kept mine fairly up-to-date, the bug did not happen before either the latest release or maybe the one before if it was within a month.

Both:
  - Installation type:
    - Debian packages installed through `apt`.
  - DDS implementation:
    - RTI Connext 6 with rmw_connextdds.
  - Client library (if applicable):
    - Using rclcpp and rclpy.

#### Steps to reproduce issue
1. Create demo package.
```
cd
mkdir -p env_bug_ws/src
cd env_bug_ws/src
source /opt/ros/foxy/local_setup.bash
ros2 pkg create --build-type ament_cmake --node-name env_bug_node env_bug
```
2. Update/add files accordingly:
  a) `src/env_bug_node.cpp`:
```
#include <cstdlib>
#include <iostream>

const char* str_or_null(const char* str)
{
  return str != nullptr ? str : "<nullptr>";
}

int main(int argc, char ** argv)
{
  (void) argc;
  (void) argv;

  auto* my_env_var = std::getenv("my_env_var");

  std::cerr << "my_env_var = '" << str_or_null(my_env_var) << "'" << std::endl;
  return 0;
}
```
  b) `launch/outer.launch.ros2.xml`:
```
<launch>
  <set_env name="my_env_var" value="1"/>
  <group>
    <set_env name="my_env_var" value="2"/>
    <include file="$(find-pkg-share env_bug)/launch/inner.launch.ros2.xml"/>
    <set_env name="my_env_var" value="3"/>
  </group>
  <group>
    <set_env name="my_env_var" value="4"/>
  </group>
</launch>
```
  c) `launch/inner.launch.ros2.xml`:
```
<launch>
  <set_env name="my_env_var" value="5"/>
  <node pkg="env_bug" exec="env_bug_node">
    <!--<env name="my_env_var" value="6"/>-->
  </node>
</launch>
```
  d) `CMakeLists.txt`:
```
...
install(DIRECTORY launch
  DESTINATION share/${PROJECT_NAME})
...
```
3. Build and run:
```
colcon build --symlink-install
. install/setup.bash
my_env_var=7 ros2 launch env_bug outer.launch.ros2.xml
```

#### Expected behavior
Either:
- Use of `set_env` does not "leak" as it was before,
- Or, include notes on this change in release notes.

For included demo, expected result:
```
[env_bug_node-1] my_env_var = '2'
```

#### Actual behavior
- `set_env` "leaks".

For included demo, actual result:
```
[env_bug_node-1] my_env_var = '4'
```

#### Additional information

We should update our launch files to use the Python method, but have not done so yet. Noticed that after package upgrades that `required` was removed, which we had just left in place even though I believe it has not been operational in Foxy at all. `doc` was also removed it seems. However, since these have errors and had no functional change, they are easy to resolve. As far as I can see, none of these changes are documented in Foxy's release notes (I guess they probably are in Rolling's).

A change to the way `set_env` works has been very interesting, as we use it for `ROS_DOMAIN_ID` to improve the way we separate nodes in our system, and as such is also quite difficult to debug. This exposes how fragile doing this is, so I think when it comes to updating (rewriting in Python) the launch files, I/we will ensure to use something more robust, for example, use `arg` to pass the IDs around and convert it to `env` inside `node`/`executable` (if that even makes sense in that launch system).

Finally, I have yet to confirm this, but it seems this leaking issue might extend further in the tree of launch files than demonstrated above. This behaviour seems less deterministic; I do not experience it, a colleague sometimes does, and a testing computer we have always does. To illustrate here is an example:

- `package1/launch/start_1...` no `set_env`s, this one launched by user.
  - `package2/launch/start_2...` `set_env` A
  - `package3/launch/start_3...` `set_env` B
    - `package3/launch/start_4...` no `set_env`s, expect nodes to start with B, but get A.

jacobperron, 2022-02-18:
The change in `set_env` behavior sounds like a bug to me, but I'll take a closer look.

>  I/we will ensure to use something more robust, for example, use arg to pass the IDs around and convert it to env inside node/executable (if that even makes sense in that launch system).

I think this would work, though I would say it'd still be nice to be able to `set_env`. I presume the same issue exists in Galactic and Rolling, so it would be nice to fix the behavior and then backport to Foxy.

jacobperron, 2022-02-22:
https://github.com/ros2/launch/pull/567 seems to have introduced the behavior change.

The issue exists in Galactic and Rolling as well.

jacobperron, 2022-02-22:
Here's what I've concluded after my initial investigation: scoping environment variables to group actions has never formally been supported. It just happened to work in earlier versions of Foxy (and only with frontend launch files) because of a bug that was fixed in https://github.com/ros2/launch/pull/468.

You can confirm by installing Foxy patch 6; or specifically [launch 0.10.7](https://github.com/ros2/launch/tree/0.10.7), and then trying the following two "equivalent" launch files:

```xml
<launch>
  <set_env name="my_env_var" value="1"/>
  <node pkg="foo" exec="foo" output="screen" />
  <group>
    <set_env name="my_env_var" value="2"/>
  </group>
</launch>
```

```py
import launch
from launch.actions import SetEnvironmentVariable
from launch.actions import GroupAction
from launch_ros.actions import Node

def generate_launch_description():
    return launch.LaunchDescription([
        SetEnvironmentVariable(name='my_env_var', value='1'),
        Node(package='foo', executable='foo', output='screen'),
        GroupAction([
            SetEnvironmentVariable(name='my_env_var', value='2'),
        ]),
    ])
```

The Python launch file starts the node with `my_env_var=2`, but the XML launch files starts the node with `my_env_var=1` because of a bug [here](https://github.com/ros2/launch/blob/80de7ab6f3f7d2f70eb03ae11fb3b36eee41a65a/launch_xml/launch_xml/entity.py#L82-L83):

```py
            return_list = filter(lambda x: x.tag == name, self.__xml_element)
            if not return_list:
```

If the result of the `filter` is empty, the condition `not return_list` actually evaluates to `False` (but I believe the intention of the author was to have it evaluate to `True`). I think the same bug existed in YAML launch files too. This logic was fixed in https://github.com/ros2/launch/pull/468. So now the behavior of `<set_env>` in frontend launch files is the same as Python launch files.

All that being said, I think the behavior is not intuitive. It makes more sense to me to have the environment be scoped to the group tag (if [`scoped=True`](https://github.com/ros2/launch/blob/21986e064cbe74cfe2f26be7d7e8a566c45e41c2/launch/launch/actions/group_action.py#L69)). I'll open a ticket to add support for this to Rolling.

For Foxy, I think we have two options:

1. Revert the offending commit that changed behavior for frontend launch files.
- Pro: this fixes a regression in behavior and is relatively easy to do.
- Con: the behavior is not consistent with Galactic, so when you migrate to a newer version of ROS 2 your launch files will break (again).
2. Add support for scoping environment variables to groups to Rolling, and then backport this feature to Foxy.
- Pro: fixes the regression and is consistent with newer ROS distros.
- Con: backporting this will break people who may be relying on the current behavior in Python launch files.

I'm leaning towards option 1, simply to avoid potentially breaking another set of user launch files.
Pinging @ros2/team for feedback.

Either way, we should add nodes to the relevant release pages to give users a heads up about any behavior changes.

jacobperron, 2022-02-22:
Feature request: https://github.com/ros2/launch/issues/597

wjwwood, 2022-02-23:
If we think that option 2 is "fixing a bug" then I'm ok with that as a back port. It is slightly more disruptive, but I don't consider depending on the behavior of a bug as something worth preserving, but that's just my opinion.

jacobperron, 2022-02-23:
I'm not sure that the current behavior is a bug; but it is not intuitive, IMO.

Neofish22, 2022-02-23:
Thanks for your focus on this. :) Since both options would technically solve the issue affecting me I'm happy with either.

That said I'll give my 2 cents as an external person using ROS2; I think option 2 is better on paper, but the con is potentially a big issue; wouldn't want people to be in the same situation as my team is now (finding behaviour changes with an update). Perhaps the `scoped=True` option you mention would allow them to opt in/out, and an "easy" fix after a breaking update?

jacobperron, 2022-02-23:
> Perhaps the scoped=True option you mention would allow them to opt in/out, and an "easy" fix after a breaking update?

Good idea; I think that would work.

jacobperron, 2022-02-24:
After a discussion with the ROS 2 team, there's a consensus for option 2. While we wait for the scoping feature to be implemented and backported, I'm adding a note for Foxy patch 7: https://github.com/ros2/ros2_documentation/pull/2330

jacobperron, 2022-03-03:
Here's a proposed patch to support scoping environment variables in a GroupAction: https://github.com/ros2/launch/pull/601
After that is merged, we can backport to Foxy.

jacobperron, 2022-09-23:
The change for Foxy has been merged and will be in the upcoming patch release.
Here's a release note for the behavior change: https://github.com/ros2/ros2_documentation/pull/2904

closed 2022-09-23 by jacobperron