ros2/ros2 #1245: The repository 'http://packages.ros.org/ros2/ubuntu impish Release' does not have a Release file.

aohanhongzhi, 2022-03-06:
## Bug report

**Required Info:**

- Operating System:
  ubuntu 21.10
- Installation type:
  repository package
https://docs.ros.org/en/galactic/Installation/Ubuntu-Install-Binary.html#add-the-ros-2-apt-repository
- Version or commit hash:
  - 
- DDS implementation:
  - 
- Client library (if applicable):
  - 

#### Steps to reproduce issue

```

```

#### Expected behavior

#### Actual behavior

```
eric-ubuntu% sudo apt update
Hit:1 http://dl.google.com/linux/chrome/deb stable InRelease
Hit:2 https://mirrors.tuna.tsinghua.edu.cn/ubuntu impish InRelease                                                 
Get:3 http://packages.microsoft.com/repos/code stable InRelease [10.4 kB]                                          
Get:4 https://mirrors.tuna.tsinghua.edu.cn/ubuntu impish-updates InRelease [110 kB]                                
Get:5 https://packages.microsoft.com/repos/edge stable InRelease [7,342 B]                                         
Get:6 https://mirrors.tuna.tsinghua.edu.cn/ubuntu impish-backports InRelease [101 kB]                              
Get:7 https://mirrors.tuna.tsinghua.edu.cn/ubuntu impish-security InRelease [110 kB]                               
Hit:8 http://cn.archive.ubuntu.com/ubuntu impish InRelease                                                         
Hit:9 http://security.ubuntu.com/ubuntu impish-security InRelease                                                  
Hit:10 http://ppa.launchpad.net/linuxuprising/shutter/ubuntu impish InRelease                                      
Ign:11 http://packages.ros.org/ros2/ubuntu impish InRelease                                                        
Hit:12 http://cn.archive.ubuntu.com/ubuntu impish-updates InRelease                                                
Err:13 http://packages.ros.org/ros2/ubuntu impish Release                                                          
  404  Not Found [IP: 64.50.236.52 80]
Hit:14 https://dl.yarnpkg.com/debian stable InRelease                                                              
Hit:15 http://cn.archive.ubuntu.com/ubuntu impish-backports InRelease
Ign:16 https://www.scootersoftware.com bcompare4 InRelease
Hit:17 https://www.scootersoftware.com bcompare4 Release
Reading package lists... Done
E: The repository 'http://packages.ros.org/ros2/ubuntu impish Release' does not have a Release file.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.

```

#### Additional information

----
## Feature request

#### Feature description

#### Implementation considerations

aprotyas, 2022-03-06:
Impish isn't a supported platform. From the page that you linked:
> We currently support Ubuntu Linux Focal Fossa (20.04) 64-bit x86 and 64-bit ARM.

clalancette, 2022-03-06:
That's right, we don't support non-LTS Ubuntu releases with ROS 2.

In the future, please open questions like this on https://answers.ros.org, which is our central questions and answers site.  I'm going to close this out for now.

closed 2022-03-06 by clalancette