ros2/ros2 #1257: Segfaults on Windows while running tests in some python packages.

nuclearsandwich, 2022-04-14:
## Bug report

**Required Info:**

- Operating System: Windows
- Installation type: source
- Version or commit hash: vcs export --exact repos file: https://gist.github.com/nuclearsandwich/41a8181c97bc7c794148e47270f073b0
- DDS implementation:
  - Fast-RTPS (default), CycloneDDS, ConnextDDS 6.0.1
- Client library (if applicable): rclpy

#### Steps to reproduce issue

Nightly CI jobs which failed:
- https://ci.ros2.org/view/nightly/job/nightly_win_rel/2271/
- https://ci.ros2.org/view/nightly/job/nightly_win_rep/2583/

The [corresponding Windows Debug build](https://ci.ros2.org/view/nightly/job/nightly_win_deb/2327/) did not exhibit these failures despite an identical vcs-export-exact log.

#### Actual behavior

Several different packages exhibit different segfaults when running tests.

```
Windows fatal exception: access violation

Thread 0x00001968 (most recent call first):
  File "C:\Python38\lib\selectors.py", line 314 in _select
  File "C:\Python38\lib\selectors.py", line 323 in select
  File "C:\Python38\lib\asyncio\base_events.py", line 1823 in _run_once
  File "C:\Python38\lib\asyncio\base_events.py", line 570 in run_forever
  File "C:\ci\ws\install\Lib\site-packages\launch\utilities\signal_management.py", line 135 in run_background_loop
  File "C:\Python38\lib\threading.py", line 870 in run
  File "C:\Python38\lib\threading.py", line 932 in _bootstrap_inner
  File "C:\Python38\lib\threading.py", line 890 in _bootstrap

Thread 0x00001138 (most recent call first):
  File "C:\ci\ws\install\Lib\site-packages\rclpy\node.py", line 175 in __init__
  File "C:\ci\ws\install\Lib\site-packages\rclpy\__init__.py", line 172 in create_node
  File "C:\ci\ws\install\Lib\site-packages\launch_ros\ros_adapters.py", line 60 in start
  File "C:\ci\ws\install\Lib\site-packages\launch_ros\ros_adapters.py", line 52 in __init__
  File "C:\ci\ws\install\Lib\site-packages\launch_ros\ros_adapters.py", line 119 in get_ros_adapter
  File "C:\ci\ws\install\Lib\site-packages\launch_ros\ros_adapters.py", line 136 in get_ros_node
  File "C:\ci\ws\install\Lib\site-packages\launch_ros\actions\load_composable_nodes.py", line 224 in execute
  File "C:\ci\ws\install\Lib\site-packages\launch\action.py", line 108 in visit
  File "C:\ci\ws\install\Lib\site-packages\launch\utilities\visit_all_entities_and_collect_futures_impl.py", line 38 in visit_all_entities_and_collect_futures
  File "C:\ci\ws\install\Lib\site-packages\launch\utilities\visit_all_entities_and_collect_futures_impl.py", line 45 in visit_all_entities_and_collect_futures
  File "C:\ci\ws\install\Lib\site-packages\launch\utilities\visit_all_entities_and_collect_futures_impl.py", line 45 in visit_all_entities_and_collect_futures
  File "C:\ci\ws\install\Lib\site-packages\launch\utilities\visit_all_entities_and_collect_futures_impl.py", line 45 in visit_all_entities_and_collect_futures
  File "C:\ci\ws\install\Lib\site-packages\launch\launch_service.py", line 250 in __process_event
  File "C:\ci\ws\install\Lib\site-packages\launch\launch_service.py", line 230 in _process_one_event
  File "C:\Python38\lib\asyncio\events.py", line 81 in _run
  File "C:\Python38\lib\asyncio\base_events.py", line 1851 in _run_once
  File "C:\Python38\lib\asyncio\base_events.py", line 570 in run_forever
  File "C:\Python38\lib\asyncio\windows_events.py", line 316 in run_forever
  File "C:\Python38\lib\asyncio\base_events.py", line 603 in run_until_complete
  File "C:\ci\ws\src\ros2\launch_ros\test_launch_ros\test\test_launch_ros\actions\test_composable_node_container.py", line 43 in _assert_launch_no_errors
  File "C:\ci\ws\src\ros2\launch_ros\test_launch_ros\test\test_launch_ros\actions\test_composable_node_container.py", line 70 in test_composable_node_container
  File "C:\Python38\lib\site-packages\_pytest\python.py", line 192 in pytest_pyfunc_call
  File "C:\Python38\lib\site-packages\pluggy\_callers.py", line 39 in _multicall
  File "C:\Python38\lib\site-packages\pluggy\_manager.py", line 80 in _hookexec
  File "C:\Python38\lib\site-packages\pluggy\_hooks.py", line 265 in __call__
  File "C:\Python38\lib\site-packages\_pytest\python.py", line 1761 in runtest
  File "C:\Python38\lib\site-packages\_pytest\runner.py", line 166 in pytest_runtest_call
  File "C:\Python38\lib\site-packages\pluggy\_callers.py", line 39 in _multicall
  File "C:\Python38\lib\site-packages\pluggy\_manager.py", line 80 in _hookexec
  File "C:\Python38\lib\site-packages\pluggy\_hooks.py", line 265 in __call__
  File "C:\Python38\lib\site-packages\_pytest\runner.py", line 259 in <lambda>
  File "C:\Python38\lib\site-packages\_pytest\runner.py", line 338 in from_call
  File "C:\Python38\lib\site-packages\_pytest\runner.py", line 258 in call_runtest_hook
  File "C:\Python38\lib\site-packages\_pytest\runner.py", line 219 in call_and_report
  File "C:\Python38\lib\site-packages\_pytest\runner.py", line 130 in runtestprotocol
  File "C:\Python38\lib\site-packages\_pytest\runner.py", line 111 in pytest_runtest_protocol
  File "C:\Python38\lib\site-packages\pluggy\_callers.py", line 39 in _multicall
  File "C:\Python38\lib\site-packages\pluggy\_manager.py", line 80 in _hookexec
  File "C:\Python38\lib\site-packages\pluggy\_hooks.py", line 265 in __call__
  File "C:\Python38\lib\site-packages\_pytest\main.py", line 347 in pytest_runtestloop
  File "C:\Python38\lib\site-packages\pluggy\_callers.py", line 39 in _multicall
  File "C:\Python38\lib\site-packages\pluggy\_manager.py", line 80 in _hookexec
  File "C:\Python38\lib\site-packages\pluggy\_hooks.py", line 265 in __call__
  File "C:\Python38\lib\site-packages\_pytest\main.py", line 322 in _main
  File "C:\Python38\lib\site-packages\_pytest\main.py", line 268 in wrap_session
  File "C:\Python38\lib\site-packages\_pytest\main.py", line 315 in pytest_cmdline_main
  File "C:\Python38\lib\site-packages\pluggy\_callers.py", line 39 in _multicall
  File "C:\Python38\lib\site-packages\pluggy\_manager.py", line 80 in _hookexec
  File "C:\Python38\lib\site-packages\pluggy\_hooks.py", line 265 in __call__
  File "C:\Python38\lib\site-packages\_pytest\config\__init__.py", line 164 in main
  File "C:\Python38\lib\site-packages\_pytest\config\__init__.py", line 187 in console_main
  File "C:\Python38\lib\site-packages\pytest\__main__.py", line 5 in <module>
  File "C:\Python38\lib\runpy.py", line 87 in _run_code
  File "C:\Python38\lib\runpy.py", line 194 in _run_module_as_main
```

sloretz, 2022-04-25:
Happened again last night

https://ci.ros2.org/view/nightly/job/nightly_win_rel/2282/consoleText

sloretz, 2022-04-26:
I'm unable to reproduce on Windows with `--retest-until-fail 25`. I'll try building with sanitizers on Linux to see if that catches something.

sloretz, 2022-04-26:
So far nothing caught using address sanitizer (ignoring leaks) on Linux. It might be intermittent, or something address sanitizer can't catch, or specific to Windows.

<details><summary>Invocation and output of test_load_composable_nodes.py</summary>

```console
$ ASAN_OPTIONS=detect_leaks=0 LD_PRELOAD="/usr/lib/gcc/x86_64-linux-gnu/11/libasan.so /usr/lib/gcc/x86_64-linux-gnu/11/libstdc++.so" pytest-3 src/ros2/launch_ros/test_launch_ros/test/test_launch_ros/actions/test_load_composable_nodes.py  -s
============================================================================================== test session starts ==============================================================================================
platform linux -- Python 3.10.4, pytest-6.2.5, py-1.10.0, pluggy-0.13.0
rootdir: /home/osrf/ws/ros2/src/ros2/launch_ros/test_launch_ros, configfile: pytest.ini
plugins: launch-testing-ros-0.19.2, launch-pytest-1.0.1, launch-testing-1.0.1, ament-xmllint-0.12.3, ament-pep257-0.12.3, ament-mypy-0.12.3, ament-flake8-0.12.3, ament-copyright-0.12.3, ament-lint-0.12.3, colcon-core-0.8.2, rerunfailures-10.2, mock-3.6.1, cov-3.0.0, repeat-0.9.1, timeout-2.1.0
collected 6 items                                                                                                                                                                                               

src/ros2/launch_ros/test_launch_ros/test/test_launch_ros/actions/test_load_composable_nodes.py [DEBUG] [launch.launch_context]: emitting event synchronously: 'launch.events.IncludeLaunchDescription'
[INFO] [launch]: All log files can be found below /home/osrf/.ros/log/2022-04-25-17-35-34-313585-shane-689771
[INFO] [launch]: Default logging verbosity is set to DEBUG
[DEBUG] [launch]: processing event: '<launch.events.include_launch_description.IncludeLaunchDescription object at 0x7f66f6f9c580>'
[DEBUG] [launch]: processing event: '<launch.events.include_launch_description.IncludeLaunchDescription object at 0x7f66f6f9c580>' ✓ '<launch.event_handlers.on_include_launch_description.OnIncludeLaunchDescription object at 0x7f66f6f9caf0>'
[DEBUG] [launch_ros.actions.load_composable_nodes]: Calling the '/mock_component_container/_container/load_node' service with request 'composition_interfaces.srv.LoadNode_Request(package_name='foo_package', plugin_name='bar_plugin', node_name='test_node_name', node_namespace='/test_node_namespace', log_level=0, remap_rules=[], parameters=[], extra_arguments=[])'
[DEBUG] [launch_ros.actions.load_composable_nodes]: Received response 'composition_interfaces.srv.LoadNode_Response(success=True, error_message='', full_node_name='/test_node_namespace/test_node_name', unique_id=1)'
[INFO] [launch_ros.actions.load_composable_nodes]: Loaded node '/test_node_namespace/test_node_name' in container '/mock_component_container'
[DEBUG] [launch.launch_context]: emitting event: 'launch.events.Shutdown'
[DEBUG] [launch]: processing event: '<launch.events.shutdown.Shutdown object at 0x7f66f6f9cb50>'
[DEBUG] [launch]: processing event: '<launch.events.shutdown.Shutdown object at 0x7f66f6f9cb50>' ✓ '<launch.event_handlers.on_shutdown.OnShutdown object at 0x7f66f6f9df00>'
[DEBUG] [launch]: processing event: '<launch.events.shutdown.Shutdown object at 0x7f66f6f9cb50>' ✓ '<launch.event_handlers.on_shutdown.OnShutdown object at 0x7f66f6f9cc10>'
.[DEBUG] [launch.launch_context]: emitting event synchronously: 'launch.events.IncludeLaunchDescription'
[INFO] [launch]: All log files can be found below /home/osrf/.ros/log/2022-04-25-17-35-34-313585-shane-689771
[INFO] [launch]: Default logging verbosity is set to DEBUG
[DEBUG] [launch]: processing event: '<launch.events.include_launch_description.IncludeLaunchDescription object at 0x7f66f6f9d300>'
[DEBUG] [launch]: processing event: '<launch.events.include_launch_description.IncludeLaunchDescription object at 0x7f66f6f9d300>' ✓ '<launch.event_handlers.on_include_launch_description.OnIncludeLaunchDescription object at 0x7f66f6f9dea0>'
[DEBUG] [launch_ros.actions.load_composable_nodes]: Calling the '/mock_component_container/_container/load_node' service with request 'composition_interfaces.srv.LoadNode_Request(package_name='foo_package', plugin_name='bar_plugin', node_name='test_node_name', node_namespace='/test_node_namespace', log_level=0, remap_rules=['test_topic1:=test_remap_topic1', 'test/topic/two:=test/remap_topic2'], parameters=[], extra_arguments=[])'
[DEBUG] [launch_ros.actions.load_composable_nodes]: Received response 'composition_interfaces.srv.LoadNode_Response(success=True, error_message='', full_node_name='/test_node_namespace/test_node_name', unique_id=1)'
[INFO] [launch_ros.actions.load_composable_nodes]: Loaded node '/test_node_namespace/test_node_name' in container '/mock_component_container'
[DEBUG] [launch.launch_context]: emitting event: 'launch.events.Shutdown'
[DEBUG] [launch]: processing event: '<launch.events.shutdown.Shutdown object at 0x7f66f6f9cf10>'
[DEBUG] [launch]: processing event: '<launch.events.shutdown.Shutdown object at 0x7f66f6f9cf10>' ✓ '<launch.event_handlers.on_shutdown.OnShutdown object at 0x7f66f6f9f2e0>'
[DEBUG] [launch]: processing event: '<launch.events.shutdown.Shutdown object at 0x7f66f6f9cf10>' ✓ '<launch.event_handlers.on_shutdown.OnShutdown object at 0x7f66f6f9c550>'
.[DEBUG] [launch.launch_context]: emitting event synchronously: 'launch.events.IncludeLaunchDescription'
[... cut ...]
=============================================================================================== warnings summary ================================================================================================
test/test_launch_ros/actions/test_load_composable_nodes.py: 15 warnings
  Warning: There is no current event loop

-- Docs: https://docs.pytest.org/en/stable/warnings.html
======================================================================================== 6 passed, 15 warnings in 21.81s ========================================================================================
```
</details>

clalancette, 2022-05-02:
More cases of this over the weekend: https://ci.ros2.org/view/nightly/job/nightly_win_rep/2601/ and https://ci.ros2.org/view/nightly/job/nightly_win_rel/2289/