ros2/ros2 #1258: Debug Build failed on Windows for rviz_orge_vendor

mce16, 2022-04-15:
## Bug report

**Required Info:**

- Operating System:
  - 
- Installation type:
  - 
- Version or commit hash:
  - 
- DDS implementation:
  - 
- Client library (if applicable):
  - 

#### Steps to reproduce issue

```

```

#### Expected behavior
Build success

#### Actual behavior
 CMake Error at D:/MyGit/ros2-debug-build/ros2-foxy/build/rviz_ogre_vendor/ogre-v1.12.1-prefix/src/ogre-v1.12.1-stamp/ogre-v1.12.1-build-Debug.cmake:49 (message):
    Command failed: 1
  
     'C:/Program Files/CMake/bin/cmake.exe' '--build' '.' '--config' 'Debug'

  OgreLight.cpp
  OgreLodStrategy.cpp
D:\MyGit\ros2-debug-build\ros2-foxy\build\rviz_ogre_vendor\ogre-v1.12.1-prefix\src\ogre-v1.12.1\OgreMain\src\OgreLodStrategy.cpp(89,17): error C2039: 'binary_function': is not a member of 'std' [D:\MyGit\ros2-debug-build\ros2-foxy\build\rviz_ogre_vendor\ogre-v1.12.1-prefix\src\ogre-v1.12.1-build\OgreMain\OgreMain.vcxproj]
C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.29.30133\include\array(20): message : see declaration of 'std' [D:\MyGit\ros2-debug-build\ros2-foxy\build\rviz_ogre_vendor\ogre-v1.12.1-prefix\src\ogre-v1.12.1-build\OgreMain\OgreMain.vcxproj]
D:\MyGit\ros2-debug-build\ros2-foxy\build\rviz_ogre_vendor\ogre-v1.12.1-prefix\src\ogre-v1.12.1\OgreMain\src\OgreLodStrategy.cpp(89,32): error C2504: 'binary_function': base class undefined [D:\MyGit\ros2-debug-build\ros2-foxy\build\rviz_ogre_vendor\ogre-v1.12.1-prefix\src\ogre-v1.12.1-build\OgreMain\OgreMain.vcxproj]
D:\MyGit\ros2-debug-build\ros2-foxy\build\rviz_ogre_vendor\ogre-v1.12.1-prefix\src\ogre-v1.12.1\OgreMain\src\OgreLodStrategy.cpp(89,32): error C2143: syntax error: missing ',' before '<' [D:\MyGit\ros2-debug-build\ros2-foxy\build\rviz_ogre_vendor\ogre-v1.12.1-prefix\src\ogre-v1.12.1-build\OgreMain\OgreMain.vcxproj]
D:\MyGit\ros2-debug-build\ros2-foxy\build\rviz_ogre_vendor\ogre-v1.12.1-prefix\src\ogre-v1.12.1\OgreMain\src\OgreLodStrategy.cpp(104,17): error C2039: 'binary_function': is not a member of 'std' [D:\MyGit\ros2-debug-build\ros2-foxy\build\rviz_ogre_vendor\ogre-v1.12.1-prefix\src\ogre-v1.12.1-build\OgreMain\OgreMain.vcxproj]
C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.29.30133\include\array(20): message : see declaration of 'std' [D:\MyGit\ros2-debug-build\ros2-foxy\build\rviz_ogre_vendor\ogre-v1.12.1-prefix\src\ogre-v1.12.1-build\OgreMain\OgreMain.vcxproj]
D:\MyGit\ros2-debug-build\ros2-foxy\build\rviz_ogre_vendor\ogre-v1.12.1-prefix\src\ogre-v1.12.1\OgreMain\src\OgreLodStrategy.cpp(104,32): error C2504: 'binary_function': base class undefined [D:\MyGit\ros2-debug-build\ros2-foxy\build\rviz_ogre_vendor\ogre-v1.12.1-prefix\src\ogre-v1.12.1-build\OgreMain\OgreMain.vcxproj]
D:\MyGit\ros2-debug-build\ros2-foxy\build\rviz_ogre_vendor\ogre-v1.12.1-prefix\src\ogre-v1.12.1\OgreMain\src\OgreLodStrategy.cpp(104,32): error C2143: syntax error: missing ',' before '<' [D:\MyGit\ros2-debug-build\ros2-foxy\build\rviz_ogre_vendor\ogre-v1.12.1-prefix\src\ogre-v1.12.1-build\OgreMain\OgreMain.vcxproj]
  OgreLodStrategyManager.cpp
  OgreLog.cpp
  OgreLogManager.cpp
  OgreManualObject.cpp
  OgreMaterial.cpp

#### Additional information

----
## Feature request

#### Feature description

#### Implementation considerations

clalancette, 2022-05-05:
There's not quite enough information here, but in order to build in debug mode you need to build *everything* in debug mode, and have set your environment up properly (as in https://docs.ros.org/en/foxy/Installation/Windows-Development-Setup.html#extra-stuff-for-debug-mode).  With that setup, you should be able to run:

```
python_d d:\python\colcon.exe build --merge-install --cmake-args -DCMAKE_BUILD_TYPE=Debug --packages-up-to rviz_ogre_vendor
```

That should make it build.  If that is not working for you, can you provide more information about the steps you used to set things up?

mce16, 2022-05-06:
I have managed to build it eventually only after a change in the CMakeList;

`std::binary_function` is not available in c++17.

So if you have installed compiler like visual studio 2019, most probably it will default to c++17 or higher, so the build will fail due to rviz_ogre_vendor Windows build was not set to use C++14, and the error shown in the build log was due to the use of C++17.

mce16, 2022-05-09:
I have opened a pulled request here;

[https://github.com/ros2/rviz/pull/860](https://github.com/ros2/rviz/pull/860)

clalancette, 2022-05-09:
> I have opened a pulled request here;

Thank you!