ros2/ros2 #1259: Support changing the default namespace through the ROS_NAMESPACE environment variable

aprotyas, 2022-04-15:
## Feature request

#### Support changing the default namespace through the `ROS_NAMESPACE` environment variable.

My use case: I'm developing a course that relies on using said environment variable because all students access ROS through a shared server, and this set-and-forget (in a bashrc file, say) approach was very useful for students.

Alternatively: is there an equivalent mechanism to push down nodes into a default namespace that does not require appending the `__ns` argument to each node as prescribed in https://design.ros2.org/articles/static_remapping.html?

For context, I asked [this question](https://answers.ros.org/question/396785/ros_namespace-environment-variable-in-ros-2/) in answers.ros.org and learned about the [`PushRosNamespace` launch action](https://github.com/ros2/launch_ros/blob/06dcb3ff1bfc0967df58fc8ee559de7b9f3d0838/launch_ros/launch_ros/actions/push_ros_namespace.py). While this is helpful, it does not quite meet the same need - won't always have a launch file, for example.

gbiggs, 2022-05-05:
I agree with @sloretz that it appears to me to be a perfect example of `ROS_DOMAIN_ID` being useful.

MiguelCompany, 2022-05-06:
I agree that `ROS_DOMAIN_ID` would be useful, but depending on the RMW implementation, it may have a limit (for DDS ones, it will be around 230), which may be an issue depending on the number of students.

Being an environment variable that is supported in ROS 1, supporting `ROS_NAMESPACE` in ROS 2 will also improve the experience of users migrating to the new version.

JMTC

fujitatomoya, 2022-05-06:
i would ask the teacher if he/she needs to check or access all endpoints created by students. probably it would not be useful to switch `ROS_DOMAIN_ID` for each student since it is exclusive. with `ROS_NAMESPACE`, we can provide the same experience for students and it would be helpful for teacher to see the entire activity with root namespace?

alpylmz, 2022-06-20:
Are there any updates on this? I think it must be implemented as it is easy and will help lots of use cases in migration. I also could find this feature in [https://design.ros2.org/articles/static_remapping.html](static_remapping), so to match with the design documents it should be added.

alpylmz, 2022-06-20:
I have a final week now, but I will gladly look to it after that. Thank you!

alpylmz, 2022-07-08:
Hi, I am planning to start working for this feature in a couple of days. Any help is appreciated as this will be my first contribution to ros2.