ros2/ros2 #1265: Galactic fails to build from source. Undefined variable __malloc_hook

git-user44, 2022-05-04:
## Bug report
Attempting to build ROS2 from source fails. __malloc_hook is undefined. Looking at the code, there are comments about ignoring warnings about deprecated code. Well, ignore it no longer. It has gone, or at least, it no longer exists in Fedora Linux

Appears to be referenced in two source files.
~/ros2_galactic/src/ros2/realtime_support/tlsf_cpp/test/test_tlsf.cpp
~/ros2_galactic//src/ros2/demos/pendulum_control/src/pendulum_demo.cpp

- Operating System:
 Fedora 35

- Installation type:
 Build from source (at least - attempting to.)

- Version or commit hash:

#### Steps to reproduce issue

mkdir -p ~/ros2_galactic/src
cd ~/ros2_galactic
wget https://raw.githubusercontent.com/ros2/ros2/galactic/ros2.repos
vcs import src < ros2.repos
sudo rosdep init
rosdep update
rosdep install --from-paths src --ignore-src -y --skip-keys "fastcdr rti-connext-dds-5.3.1 urdfdom_headers"
colcon build --symlink-install

#### Expected behaviour
I would expect the source code to build cleanly.

#### Actual behaviour

#### Additional information

clalancette, 2022-05-04:
Yes, this has been fixed in Humble, but not in Galactic.  I guess we could backport the fixes that are in https://github.com/ros2/realtime_support/pull/109 .  I *think* there was another fix somewhere else in the ROS 2 stack, but I can't remember at the moment.  If you'd like to open a PR backporting those fixes, I'd be happy to review.

git-user44, 2022-05-04:
Sorry. I will wait for the next release - any day soon I guess.

I will close this issue.

closed 2022-05-04 by git-user44