ros2/ros2 #1268: Sourcing setup.bash doesn't work in ROS2 foxy

Salman-F, 2022-05-13:
- Operating System:
  - Ubuntu 20.04
- Installation type:
  - from source
- Version or commit hash:
  - ROS 2 Foxy Fitzroy
- DDS implementation:
  - Fast-RTPS

#### Steps to reproduce issue
Open new terminal.
```
source /opt/ros/foxy/setup.bash
```

#### Expected behavior
- ros2 command should be detected

#### Actual behavior
- ros2 command is not detected. Can't use ros2 (to start turtlesim for example)

#### Additional information
If i use: 
```
. ~/ros2_foxy/install/local_setup.bash
```
i can use the ros2 command. Is there any disadvantage on using this command?
And does it mean that something went wrong with my installation because i cant run the first command?
I finished the whole installation and the example at the end also worked, thats why i am confused.

Thanks in advance :)

clalancette, 2022-05-13:
Yes, something is definitely wrong with your setup.

What were the exact commands you used to install Foxy?

Salman-F, 2022-05-13:
Hi, i followed this installation guide: https://docs.ros.org/en/foxy/Installation/Ubuntu-Development-Setup.html.
I was a little confused with this step.
[image]
I thought i dont have to execute it because i didnt install ROS2 on another way (either via Debians or the binary distribution), but the next step didnt work because there was no install folder in the created ros2_foxy folder. After i executed i could finish my installation.

- i also finished this tutorial: https://docs.ros.org/en/foxy/Tutorials/Turtlesim/Introducing-Turtlesim.html while using 
```
. ~/ros2_foxy/install/local_setup.bash
```
to init ros2 in every shell.

Salman-F, 2022-05-23:
My Sollution was to uninstall ROS2 from source and installing it again with the Debian installation. Everythin is working now normally.

closed 2022-05-23 by Salman-F