ros2/ros2 #1270: ROS2 Foxy commands are unresponsive

harun-loodos, 2022-05-18:
After both clean Ubuntu and ROS installation, ROS2 commands became unresponsive. Commands are not working and freeze indefinitely on the terminal. Below, you can find the lists for both working and not-working commands and their corresponding outputs if exist.

The commands I **can** run:
- `ros2 daemon start`
  - Failed to confirm that the daemon started successfully
- `ros2 multicast send/receive`
  -  Sending one UDP multicast datagram... / Waiting for UDP multicast datagram... \n Received from 192.168.0.44:57770: 'Hello World!'
- `ros2 <command> --help` commands

The commands I **can not** run:
- `ros2 topic/action/node/service list`
- `ros2 run/launch commands

**Required Info:**

- Operating System: Ubuntu 20.04.4
- Installation type: binaries
  - `sudo apt install ros-foxy-desktop`
- Version or commit hash:
  - latest release on debian packages (probably Foxy Path Release 7)
- DDS implementation:
  - rmw_fastrtps_cpp
- Client library (if applicable):
  - rclcpp/rclpy

#### Expected behavior

#### Actual behavior

#### Additional information

clalancette, 2022-05-18:
Hm, this is quite a weird one.  This seems to be working fine for me, and even seems to be working fine for you in a separate machine.

A few different things to try/check out:
* Check out your networking setup, and make sure it is configured properly and has multicast enabled.  Many ROS 2 commands won't work properly without multicast.
* Try running some of the `ros2` commands with `--no-daemon`, like: `ros2 topic list --no-daemon`.
* Look for hardware errors in `dmesg`, and/or reboot the machine and try the tests again.
* Try running this test again on the same machine, but in a docker container.  That can at least tell us if something is misconfigured on the host.

gavanderhoorn, 2022-05-18:
Please note this has a duplicate/related post on ROS Answers: [ROS2 (Foxy) commands stopped working](https://answers.ros.org/question/400377).

tfoote, 2022-05-23:
This looks like a local issue needing debugging more than an issue to track so I'm going to close this in favor of the answers.ros.org post. Please iterate there to get more help finding the issue with this specific machine.

closed 2022-05-23 by tfoote

newMiodek, 2024-11-16:
I had the same issue (or at least one that looked similar). After a fresh Ubuntu (20.04.6) install I tried following the [tutorial](https://docs.ros.org/en/foxy/Installation/Ubuntu-Install-Debians.html#try-some-examples) with Foxy. After sourcing the setup I tried to run
`ros2 run demo_nodes_cpp talker`
but nothing happened. ROS became unresponsive and I couldn't even stop the process with `Ctrl+C`.

At first I thought that everything might be working, but at a very slow pace, because I installed Ubuntu on an HDD and everything was noticeably slow. So I tried to install Ubuntu on another drive - an SSD. Unfortunately I ended up with the same result.

Then I tried running everything inside a docker container which I set up on my Debian system (on the same PC), but this didn't help either. I also tried the bare-bones install but that was not it.

At some point I tried running
`ros2 run turtlesim turtlesim_node`
and something new happened - before stopping like in the earlier scenarios, the program printed a warning message:
`WARNING: CPU random generator seem to be failing, disable hardware random number generation`
`WARNING: RDRND generated: 0xffffffff 0xffffffff 0xffffffff 0xffffffff`
This was an important clue.

I recalled that I already saw messages like this before - every time I booted Debian or the newly installed Ubuntu systems a series of messages saying pretty much the same thing appeared. Ubuntu called it a "funky smelling output".

The next thing I tried was to set up Ubuntu inside of a VM on Windows (still same PC). This time the output smelled fine - Ubuntu gave no warnings. And when I tried to run the demo talker **it finally worked**.

I additionally checked what would happen if I set up an Ubuntu VM on my Debian system which also has an issue with `RDRAND`, and ROS did **not** work.

So I searched around on what I can do with this `RDRAND` thing, because I was becoming more and more convinced that it was the cause of my problems. Someone suggested a BIOS update. I updated it, the `RDRAND` warnings disappeared and ROS finally worked on my Ubuntu installation.

I'm posting this here because the discussion on answers.ros.org is now read-only and it has no equivalent on the robotics stack exchange.

harun-loodos, 2024-11-16:
Ooh, I never thought about it. I have an AMD Ryzen 9 3960X processor and now that you say it, I did also get
```
WARNING: CPU random generator seem to be failing, disable hardware random number generation
WARNING: RDRND generated: 0xffffffff 0xffffffff 0xffffffff 0xffffffff
```
warnings on boot. I tried to update the bios but for some reason I couldn't do it so I had given up. I had given up on this ROS setup as well because nothing seemed to be working and just installed ROS on another machine.

Now it makes sense. And it is understandable that ROS needs a proper working low level CPU functions and if they are somehow disabled, it causes issues. I will try again updating the bios. Thank you @newmiodek