ros2/ros2 #1272: Humble Binary-Install breaks Ubuntu 22.04

krsche, 2022-05-18:
## Bug report

**Required Info:**

- Operating System: Ubuntu 22.04
  - 
- Installation type: binaries
  - 
- Version or commit hash: Humble - ros2-humble-desktop
  - 
- DDS implementation: ?
  - 
- Client library (if applicable): -
  - 

#### Steps to reproduce issue
Follow the installation instructions for humble from the documentation until the step where you actually install it: `sudo apt install ros-humble-desktop`  

One minor deviation from the instructions:  
- LC_ALL was empty: `LC_ALL=`, but everything else was UTF8, default Ubuntu 22.04 install.

__More Info about my system:__  
- Fairly fresh Ubuntu 22.04 install on an XPS15 9510 w/ Nvidia Rtx 3050ti (driver version 510)
- Nvidia GPU was disabled (via `prime-select intel`)

#### Expected behavior
Install via apt works without breaking my OS :)

#### Actual behavior
- Initially install seemed to run through fine, until I heard the 'device disconnected' sound
- checked the terminal where the install ran and exit code was 0 -> looks good, __until...__ I noticed that:
  - my wifi was gone
  - settings window doesn't open anymore and the shutdown option disappeared from the dropdown in the top-right (only log out was possible)
  - `ip link list` reported that my wifi device was down and I couldn't bring it up anymore
- I then decided to reboot (via terminal) to see if that helps but I was greeted by a kernel panic on boot

Rendered my distro unusable and I wasn't able to recover it yet.

#### Additional information
> Info
>
> I'll get the apt logs in a second and will add them here

- Booting into emergency mode on an older kernel version (-27) worked
- Purging the nvidia drivers didn't help

nuclearsandwich, 2022-05-18:
Thanks for the report @krsche we're investigating.

Did you install 22.04 fresh or upgrade from an earlier version of Ubuntu?

As I think of other questions I'll update.

nkoenig, 2022-05-18:
I had a similar experience. I'm on a fresh install of Ubuntu 22.04, and attempted to install `ros-humble-desktop`. During the install my X server died. After a reboot, I'm only presented with a shell login  and no X server.

I tried to re-install `ros-humble-desktop` in my shell session. Apt told me to first run `sudo dpkg --configure -a` to correct the problem introduced by `dpkg` being interrupted. This ran fine.

Then an attempt to run `sudo apt install ros-humble-desktop` gave me the error:
```
udev: Breaks systemd
```

krsche, 2022-05-18:
Just added the logs to the top post. It can clearly be seen that lots of seemingly important stuff got removed.  

I haven't really checked the output of what is to be installed / removed before confirming it, since the list is super long and I know that the ros installation is big. Probably should've done that :)

I assume the problem with the kernel panic is related to the removal of initramfs and it not configuring correctly. But for now I'm planning with installing a clean Ubuntu again. I made a backup of `/` though, so if any other logs would help you let me know.

@nuclearsandwich it was fresh install, no upgrade

nuclearsandwich, 2022-05-18:
@krsche can you pull `/var/log/apt/history.log` from your previous installation.

We're still trying to figure out which actual packages in the chain are causing it but performing an `apt dist-upgrade -y` after installing Ubuntu Jammy and before installing any packages from ROS seems to maintain a working machine.

We're trying to figure out which packages that desktop depends on are causing the conflicts in an un-upgraded Jammy installation.

nuclearsandwich, 2022-05-18:
@cottsay just pointed out that this must affect ros-rolling-desktop as well since they share many dependencies on this platform.

tfoote, 2022-05-19:
I have successfully isolated the problem to an interaction between udev and init 

See https://github.com/tfoote/udevdebug

nuclearsandwich, 2022-05-19:
I've opened https://github.com/ros2/ros2_documentation/pull/2581 and https://github.com/ros2/ros2_documentation/pull/2582 by way of resolving this issue through advice.

I'm also working on a Launchpad issue to see if this is something that could be protected upstream.

nuclearsandwich, 2022-05-19:
https://bugs.launchpad.net/ubuntu/+source/systemd/+bug/1974196

clalancette, 2022-06-02:
Between the bug open with Canonical, and the mitigations we put in elsewhere, I think we can call this fixed.  I'm going to close it out, but feel free to reopen if you disagree.

maxpolzin, 2022-06-22:
After a fresh install of Ubuntu Server 22.04 on a Raspberry Pi 3 B+, I am still getting this error when following the installation instructions from here: https://docs.ros.org/en/humble/Installation/Ubuntu-Install-Debians.html

`
ubuntu@ubuntu:~$ sudo apt install ros-humble-ros-base
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 apt : Depends: libapt-pkg6.0 (>= 2.4.5) but it is not going to be installed
       Depends: libsystemd0 but it is not installable
 bsdutils : PreDepends: libsystemd0 but it is not installable
 init : PreDepends: systemd-sysv but it is not going to be installed
 libprocps8 : Depends: libsystemd0 (>= 209) but it is not installable
 util-linux : PreDepends: libsystemd0 but it is not installable
              PreDepends: libudev1 (>= 183) but it is not installable
E: Error, pkgProblemResolver::Resolve generated breaks, this may be caused by held packages.
`
I might have made an error during the installation and start with a fresh installation again soon.

coalman321, 2022-06-23:
This behavior is now showing up with Humble desktop, but with a few more conflicting packages related to QT versions being missing as well. I am configuring a new machine that has been apt upgraded before attempting to install.  

Here is the output of the installation of desktop:
`sudo apt upgrade && sudo apt install ros-humble-desktop

Reading package lists...
Building dependency tree...
Reading state information...
Calculating upgrade...
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.

Reading package lists...
Building dependency tree...
Reading state information...
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 apt : Depends: libapt-pkg6.0 (>= 2.4.5) but it is not going to be installed
       Depends: libsystemd0 but it is not installable
 bsdutils : PreDepends: libsystemd0 but it is not installable
 dconf-gsettings-backend : Depends: dconf-service (< 0.40.0-3.1~)
                           Depends: dconf-service (>= 0.40.0-3)
 init : PreDepends: systemd-sysv
 libcolord2 : Depends: libudev1 (>= 196) but it is not installable
 libdbus-1-3 : Depends: libsystemd0 but it is not installable
               Recommends: dbus
 libhwloc15 : Depends: libudev1 (>= 183) but it is not installable
 libignition-cmake2-dev : Depends: cmake
 libopenmpi-dev : Depends: openmpi-bin (>= 3.0.0-1)
                  Recommends: libcoarrays-openmpi-dev but it is not going to be installed
 libopenni2-0 : Depends: libudev1 (>= 183) but it is not installable
 libpulse0 : Depends: libsystemd0 but it is not installable
 libqt5designer5 : Depends: libqt5gui5 (>= 5.14.1) but it is not installable or
                            libqt5gui5-gles (>= 5.14.1) but it is not installable
 libqt5designercomponents5 : Depends: libqt5gui5 (>= 5.14.1) but it is not installable or
                                      libqt5gui5-gles (>= 5.14.1) but it is not installable
 libqt5help5 : Depends: libqt5gui5 (>= 5.0.2) but it is not installable or
                        libqt5gui5-gles (>= 5.0.2) but it is not installable
 libqt5printsupport5 : Depends: libqt5gui5 (>= 5.14.1) but it is not installable or
                                libqt5gui5-gles (>= 5.14.1) but it is not installable
 libqt5quickshapes5 : Depends: libqt5gui5 (>= 5.14.1) but it is not installable or
                               libqt5gui5-gles (>= 5.14.1) but it is not installable
                      Depends: libqt5quick5 (>= 5.14.1) but it is not installable or
                               libqt5quick5-gles (>= 5.14.1) but it is not installable
 libqt5quicktest5 : Depends: libqt5gui5 (>= 5.12.2) but it is not installable or
                             libqt5gui5-gles (>= 5.12.2) but it is not installable
                    Depends: libqt5quick5 (>= 5.0.2) but it is not installable or
                             libqt5quick5-gles (>= 5.0.2) but it is not installable
 libqt5quickwidgets5 : Depends: libqt5gui5 (>= 5.10.0) but it is not installable or
                                libqt5gui5-gles (>= 5.10.0) but it is not installable
                       Depends: libqt5quick5 (>= 5.15.1) but it is not installable or
                                libqt5quick5-gles (>= 5.15.1) but it is not installable
 libqt5svg5 : Depends: libqt5gui5 (>= 5.15.3+dfsg~) but it is not installable or
                       libqt5gui5-gles (>= 5.15.3+dfsg~) but it is not installable
 libqt5webkit5 : Depends: libqt5gui5 (>= 5.14.1) but it is not installable or
                          libqt5gui5-gles (>= 5.14.1) but it is not installable
                 Depends: libqt5quick5 (>= 5.15.3+dfsg~) but it is not installable or
                          libqt5quick5-gles (>= 5.15.3+dfsg~) but it is not installable
 libqt5widgets5 : Depends: libqt5gui5 (>= 5.15.1) but it is not installable or
                           libqt5gui5-gles (>= 5.15.1) but it is not installable
 libudev-dev : Depends: libudev1 (= 249.11-0ubuntu3) but it is not installable
 libusb-1.0-0 : Depends: libudev1 (>= 183) but it is not installable
 libvtk9-qt-dev : Depends: libqt5opengl5-dev but it is not installable
 libvtk9.1-qt : Depends: libqt5gui5 (>= 5.11.0~rc1) but it is not installable or
                         libqt5gui5-gles (>= 5.11.0~rc1) but it is not installable
 mpi-default-bin : Depends: openmpi-bin
[... 28 lines cut ...]
                                      libqt5gui5-gles (>= 5.14.1) but it is not installable
 ros-humble-rviz-common : Depends: libqt5gui5 (>= 5.2.0) but it is not installable or
                                   libqt5gui5-gles (>= 5.2.0) but it is not installable
                          Depends: libqt5gui5 but it is not installable
                          Depends: libqt5opengl5 but it is not installable
 ros-humble-rviz-default-plugins : Depends: libqt5gui5 (>= 5.2.0) but it is not installable or
                                            libqt5gui5-gles (>= 5.2.0) but it is not installable
                                   Depends: libqt5gui5 but it is not installable
                                   Depends: libqt5opengl5 but it is not installable
 ros-humble-rviz-rendering : Depends: libqt5gui5 (>= 5.5.0) but it is not installable or
[... cut ...]
 shim-signed : Depends: grub-efi-amd64-signed but it is not going to be installed or
                        grub-efi-arm64-signed but it is not installable
               Depends: grub2-common (>= 2.04-1ubuntu24)
 util-linux : PreDepends: libsystemd0 but it is not installable
              PreDepends: libudev1 (>= 183) but it is not installable
E: Error, pkgProblemResolver::Resolve generated breaks, this may be caused by held packages.
`

@clalancette or @nuclearsandwich anything on your end with this? It renders Humble un-installable right now.

clalancette, 2022-06-23:
> @clalancette or @nuclearsandwich anything on your end with this?

I just tried installing `ros-humble-desktop` in a fresh Ubuntu 22.04 container, and I didn't run into any issues.

Are you exactly following the instructions at https://docs.ros.org/en/humble/Installation/Ubuntu-Install-Debians.html ?  Do you have any other third-party apt repositories enabled?  How did you initially install Ubuntu 22.04?

nuclearsandwich, 2022-06-23:
If you are having trouble installing ROS 2 on your system. The appropriate way to request support is by asking a question on [ROS Answers](https://answers.ros.org/). So far all of these problems, including the one first reported in this issue, have not originated within ROS 2 and bumping a closed issue with new, different problems, is not a reliable way to get support.

If the discussion in ROS Answers, does reveal an actual bug in ROS 2 or its packaging and distribution, it would of course make sense to open a new issue with details but should not be the first step.

coalman321, 2022-06-23:
The system was installed fresh from a USB stick flashed with the canonical amd64 image. The only other non-stock repository is the docker repository for installing docker engine onto the computer. Otherwise, the instructions in  https://docs.ros.org/en/humble/Installation/Ubuntu-Install-Debians.html were followed step by step. 

> If you are having trouble installing ROS 2 on your system. The appropriate way to request support is by asking a question on [ROS Answers](https://answers.ros.org/). So far all of these problems, [...]

The reason I brought this here, was that it seemed to be related to the issue outlined above. Sorry if that should not have been the case.

jgoppert, 2022-06-23:
I have the same exact issue, fresh install today from a usb stick and posted a ros answers question: https://answers.ros.org/question/402781/ros-humble-ubuntu-2204-apt-install-issue/

rshanor, 2022-06-23:
Similar issue with `ros-humble-ros-base` on a fresh 22 install.

```
sudo apt install ros-humble-ros-base
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 apt : Depends: libapt-pkg6.0 (>= 2.4.5) but it is not going to be installed
       Depends: libsystemd0 but it is not installable
 bsdutils : PreDepends: libsystemd0 but it is not installable
 init : PreDepends: systemd-sysv
 ros-humble-ament-cmake : Depends: cmake
 ros-humble-ament-cmake-core : Depends: cmake
 shim-signed : Depends: grub-efi-amd64-signed but it is not going to be installed or
                        grub-efi-arm64-signed but it is not installable
               Depends: grub2-common (>= 2.04-1ubuntu24)
 util-linux : PreDepends: libsystemd0 but it is not installable
              PreDepends: libudev1 (>= 183) but it is not installable
E: Error, pkgProblemResolver::Resolve generated breaks, this may be caused by held packages.
```

jakkaj, 2022-06-26:
I too am facing a similar issue in a VS Code 22.04 DevContainer. It was working in the past...

```
The following packages have unmet dependencies:
#0 23.86  apt : Depends: libapt-pkg6.0 (>= 2.4.5) but it is not going to be installed
#0 23.86  libcolord2 : Depends: libudev1 (>= 196) but it is not installable
#0 23.86  libfido2-1 : Depends: libudev1 (>= 183) but it is not installable
#0 23.86  libhwloc15 : Depends: libudev1 (>= 183) but it is not installable
#0 23.86  libopenni2-0 : Depends: libudev1 (>= 183) but it is not installable
#0 23.86  libqt5designer5 : Depends: libqt5gui5 (>= 5.14.1) but it is not installable or
#0 23.86                             libqt5gui5-gles (>= 5.14.1) but it is not installable
#0 23.86  libqt5designercomponents5 : Depends: libqt5gui5 (>= 5.14.1) but it is not installable or
#0 23.86                                       libqt5gui5-gles (>= 5.14.1) but it is not installable
#0 23.86  libqt5help5 : Depends: libqt5gui5 (>= 5.0.2) but it is not installable or
#0 23.86                         libqt5gui5-gles (>= 5.0.2) but it is not installable
#0 23.86  libqt5printsupport5 : Depends: libqt5gui5 (>= 5.14.1) but it is not installable or
#0 23.86                                 libqt5gui5-gles (>= 5.14.1) but it is not installable
#0 23.86  libqt5quickshapes5 : Depends: libqt5gui5 (>= 5.14.1) but it is not installable or
#0 23.86                                libqt5gui5-gles (>= 5.14.1) but it is not installable
#0 23.86                       Depends: libqt5quick5 (>= 5.14.1) but it is not installable or
#0 23.86                                libqt5quick5-gles (>= 5.14.1) but it is not installable
#0 23.86  libqt5quicktest5 : Depends: libqt5gui5 (>= 5.12.2) but it is not installable or
#0 23.86                              libqt5gui5-gles (>= 5.12.2) but it is not installable
#0 23.86                     Depends: libqt5quick5 (>= 5.0.2) but it is not installable or
#0 23.86                              libqt5quick5-gles (>= 5.0.2) but it is not installable
#0 23.86  libqt5quickwidgets5 : Depends: libqt5gui5 (>= 5.10.0) but it is not installable or
#0 23.86                                 libqt5gui5-gles (>= 5.10.0) but it is not installable
#0 23.86                        Depends: libqt5quick5 (>= 5.15.1) but it is not installable or
#0 23.86                                 libqt5quick5-gles (>= 5.15.1) but it is not installable
#0 23.86  libqt5svg5 : Depends: libqt5gui5 (>= 5.15.3+dfsg~) but it is not installable or
#0 23.86                        libqt5gui5-gles (>= 5.15.3+dfsg~) but it is not installable
#0 23.86  libqt5webkit5 : Depends: libqt5gui5 (>= 5.14.1) but it is not installable or
#0 23.86                           libqt5gui5-gles (>= 5.14.1) but it is not installable
#0 23.86                  Depends: libqt5quick5 (>= 5.15.3+dfsg~) but it is not installable or
#0 23.86                           libqt5quick5-gles (>= 5.15.3+dfsg~) but it is not installable
#0 23.86  libqt5widgets5 : Depends: libqt5gui5 (>= 5.15.1) but it is not installable or
[2022-06-26T09:14:54.930Z] #0 23.86                            libqt5gui5-gles (>= 5.15.1) but it is not installable
#0 23.86  libudev-dev : Depends: libudev1 (= 249.11-0ubuntu3) but it is not installable
#0 23.86  libusb-1.0-0 : Depends: libudev1 (>= 183) but it is not installable
#0 23.86  libvtk9-qt-dev : Depends: libqt5opengl5-dev but it is not installable
#0 23.86  libvtk9.1-qt : Depends: libqt5gui5 (>= 5.11.0~rc1) but it is not installable or
#0 23.86                          libqt5gui5-gles (>= 5.11.0~rc1) but it is not installable
#0 23.86  python3-pyqt5 : Depends: libqt5gui5 (>= 5.1.0) but it is not installable
#0 23.86                  Depends: libqt5gui5 (>= 5.15.1) but it is not installable or
#0 23.86                           libqt5gui5-gles (>= 5.15.1) but it is not installable
#0 23.86  python3-pyqt5.qtsvg : Depends: libqt5gui5 (>= 5.0.2) but it is not installable or
#0 23.86                                 libqt5gui5-gles (>= 5.0.2) but it is not installable
#0 23.86  python3-pyside2.qtgui : Depends: libqt5gui5 (>= 5.15) but it is not installable
#0 23.86                          Depends: libqt5gui5 (>= 5.15.1) but it is not installable or
#0 23.86                                   libqt5gui5-gles (>= 5.15.1) but it is not installable
#0 23.86  python3-pyside2.qtsvg : Depends: libqt5gui5 (>= 5.15) but it is not installable or
#0 23.86                                   libqt5gui5-gles (>= 5.15) but it is not installable
#0 23.86  python3-pyside2.qtwidgets : Depends: libqt5gui5 (>= 5.15) but it is not installable or
#0 23.86                                       libqt5gui5-gles (>= 5.15) but it is not installable
#0 23.86  qhelpgenerator-qt5 : Depends: libqt5gui5 (>= 5.0.2) but it is not installable or
#0 23.86                                libqt5gui5-gles (>= 5.0.2) but it is not installable
#0 23.86  qt5-assistant : Depends: libqt5gui5 (>= 5.11.0~rc1) but it is not installable or
#0 23.86                           libqt5gui5-gles (>= 5.11.0~rc1) but it is not installable
#0 23.86  qt5-qmltooling-plugins : Depends: libqt5gui5 (>= 5.9.0~beta) but it is not installable or
#0 23.86                                    libqt5gui5-gles (>= 5.9.0~beta) but it is not installable
#0 23.86                           Depends: libqt5quick5 (>= 5.12.2) but it is not installable or
#0 23.86                                    libqt5quick5-gles (>= 5.12.2) but it is not installable
[... cut ...]
#0 23.86                              Depends: libqt5opengl5 but it is not installable
#0 23.86  ros-humble-turtlesim : Depends: libqt5gui5 (>= 5.2.0) but it is not installable or
#0 23.86                                  libqt5gui5-gles (>= 5.2.0) but it is not installable
#0 23.86                         Depends: libqt5gui5 but it is not installable
#0 23.87  util-linux : PreDepends: libudev1 (>= 183) but it is not installable
#0 23.87 E: Error, pkgProblemResolver::Resolve generated breaks, this may be caused by held packages.
------
```

tfoote, 2022-06-30:
@jakkaj Can you point to the specific container you used as well as the exact commands you ran so we can reproduce? Have you tried a `dist-upgrade` before you try to install instead of just an `upgrade`? What does `apt-cache policy libudev1` show?

mw46d, 2022-07-06:
same problem (Started with an Xubuntu 22.04 iso -> apt update && apt-dist-upgrade :-( 
```
root@ros2-ubuntu2204:~# apt-cache policy libudev1
libudev1:
  Installed: 249.11-0ubuntu3
  Candidate: 249.11-0ubuntu3
  Version table:
     249.11-0ubuntu3.4 1 (phased 10%)
        500 http://us.archive.ubuntu.com/ubuntu jammy-updates/main amd64 Packages
 *** 249.11-0ubuntu3 500
        500 http://us.archive.ubuntu.com/ubuntu jammy/main amd64 Packages
        100 /var/lib/dpkg/status

```

tfoote, 2022-07-08:
It looks like there's a fix coming from upstream too: https://bugs.launchpad.net/ubuntu/+source/systemd/+bug/1974196

ruturajsambhusvt, 2023-06-08:
I am still facing the same issue on Ubuntu 22.04. I was following the steps here -> https://docs.ros.org/en/iron/Installation/Ubuntu-Install-Debians.html

[ros2_issue.txt](https://github.com/ros2/ros2/files/11692833/ros2_issue.txt)

mw46d, 2023-06-08:
Why do you play with humble & iron at the same time?!

mw46d, 2023-06-08:
OK, update your `python3-catkin-pkg` package. There is an older version in the Ubuntu repos (which breaks things) and a newer version in the ros2 repos, which works;-)
```
marcow@feather6:~$ apt policy python3-catkin-pkg
python3-catkin-pkg:
  Installed: 0.5.2-100
  Candidate: 0.5.2-100
  Version table:
 *** 0.5.2-100 500
        500 http://packages.ros.org/ros2/ubuntu jammy/main amd64 Packages
        100 /var/lib/dpkg/status
     0.4.24-2 500
        500 http://us.archive.ubuntu.com/ubuntu jammy/universe amd64 Packages
        500 http://us.archive.ubuntu.com/ubuntu jammy/universe i386 Packages
marcow@feather6:~$ dpkg -l python3-catkin-pkg
Desired=Unknown/Install/Remove/Purge/Hold
| Status=Not/Inst/Conf-files/Unpacked/halF-conf/Half-inst/trig-aWait/Trig-pend
|/ Err?=(none)/Reinst-required (Status,Err: uppercase=bad)
||/ Name               Version      Architecture Description
+++-==================-============-============-=================================
ii  python3-catkin-pkg 0.5.2-100    all          catkin package library
```

ruturajsambhusvt, 2023-06-08:
Hello, the play with humble and iron might be unintended error, I didn't even realize that.

This is the output of - sudo apt-get install python3-catkin-pkg-modules python3-rospkg-modules
[catkin_out.txt](https://github.com/ros2/ros2/files/11696860/catkin_out.txt)

I do get that the latest one is vailable at the ros2 repo, great if you can let me know how to update to that. Sorry, I am not very familiar with ros.

ruturajsambhusvt, 2023-06-08:
Hello, the play with humble and iron might be unintended error, I didn't even realize that.

This is the output of - sudo apt-get install python3-catkin-pkg-modules python3-rospkg-modules
[catkin_out.txt](https://github.com/ros2/ros2/files/11696860/catkin_out.txt)

I do get that the latest one is vailable at the ros2 repo, great if you can let me know how to update to that. 
"sudo apt install python3-catkin-package=0.5.2" does not work and throws this error->

Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
E: Unable to locate package python3-catkin-package

mw46d, 2023-06-09:
What does `sudo apt install python3-catkin-pkg` do? ( -pkg instead of your -package ;-)

ruturajsambhusvt, 2023-06-09:
Hello,

This is the output of the `sudo apt install python3-catkin-pkg` ->

``` 
(base) trec@TREC-Rome:~$ sudo apt install python3-catkin-pkg
[sudo] password for trec: 
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
You might want to run 'apt --fix-broken install' to correct these.
The following packages have unmet dependencies:
 python3-catkin-pkg : Depends: python3-catkin-pkg-modules (>= 0.5.2) but it is not going to be installed
 python3-rosdistro-modules : Depends: python3-catkin-pkg-modules but it is not going to be installed
                             Depends: python3-rospkg-modules but it is not going to be installed
 ros-humble-ament-cmake-core : Depends: python3-catkin-pkg-modules but it is not going to be installed
 ros-humble-qt-gui : Depends: python3-catkin-pkg-modules but it is not going to be installed
 ros-humble-ros2doctor : Depends: python3-catkin-pkg-modules but it is not going to be installed
 ros-humble-ros2pkg : Depends: python3-catkin-pkg-modules but it is not going to be installed
 ros-humble-rqt-gui : Depends: python3-catkin-pkg-modules but it is not going to be installed
 ros-humble-rqt-msg : Depends: python3-catkin-pkg-modules but it is not going to be installed
 ros-humble-rqt-plot : Depends: python3-catkin-pkg-modules but it is not going to be installed
 ros-humble-rqt-publisher : Depends: python3-catkin-pkg-modules but it is not going to be installed
 ros-humble-rqt-shell : Depends: python3-catkin-pkg-modules but it is not going to be installed
E: Unmet dependencies. Try 'apt --fix-broken install' with no packages (or specify a solution).
```

This is the output of `sudo apt update && sudo apt upgrade` ->

```
(base) trec@TREC-Rome:~$ sudo apt update && sudo apt upgrade
Get:1 file:/var/cuda-repo-ubuntu2204-11-7-local  InRelease [1,575 B]
Get:1 file:/var/cuda-repo-ubuntu2204-11-7-local  InRelease [1,575 B]
Hit:2 http://packages.microsoft.com/repos/code stable InRelease                
Hit:3 http://us.archive.ubuntu.com/ubuntu jammy InRelease                      
Hit:4 https://download.docker.com/linux/ubuntu jammy InRelease                 
Get:5 http://security.ubuntu.com/ubuntu jammy-security InRelease [110 kB]      
Hit:6 https://linux.teamviewer.com/deb stable InRelease                        
Hit:7 https://nvidia.github.io/libnvidia-container/experimental/ubuntu18.04/amd64  InRelease
Hit:8 https://dl.google.com/linux/chrome/deb stable InRelease                  
Get:9 http://us.archive.ubuntu.com/ubuntu jammy-updates InRelease [119 kB]     
Hit:10 https://packages.microsoft.com/repos/vscode stable InRelease            
Hit:11 https://nvidia.github.io/libnvidia-container/stable/ubuntu18.04/amd64  InRelease
Get:12 http://us.archive.ubuntu.com/ubuntu jammy-backports InRelease [108 kB]  
Ign:13 http://packages.ros.org/ros/ubuntu jammy InRelease                      
Hit:14 https://ppa.launchpadcontent.net/deadsnakes/ppa/ubuntu jammy InRelease
Hit:15 http://packages.ros.org/ros2/ubuntu jammy InRelease                     
Err:16 http://packages.ros.org/ros/ubuntu jammy Release                        
  404  Not Found [IP: 140.211.166.134 80]
Hit:17 https://ppa.launchpadcontent.net/kisak/kisak-mesa/ubuntu jammy InRelease
Hit:18 https://ppa.launchpadcontent.net/obsproject/obs-studio/ubuntu jammy InRelease
Reading package lists... Done                               
E: The repository 'http://packages.ros.org/ros/ubuntu jammy Release' does not have a Release file.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.
```

ruturajsambhusvt, 2023-06-09:
Okay, I commented out the line from these two files - `ros-latest.list` and `ros-latest.list.save`.

The output of `sudo apt install python3-catkin-pkg python3-catkin-pkg-modules python3-rospkg-modules` after commenting those lines->

```
(base) trec@TREC-Rome:~$ sudo apt install python3-catkin-pkg python3-catkin-pkg-modules python3-rospkg-modules
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
python3-catkin-pkg is already the newest version (0.5.2-100).
python3-catkin-pkg-modules is already the newest version (0.5.2-1).
The following packages were automatically installed and are no longer required:
  cmake-extras docker-scan-plugin libgl1-amber-dri libgnome-bg-4-1 libllvm13
  libllvm13:i386 libntfs-3g89 libvo-amrwbenc0 libvulkan1:i386
  libwayland-client0:i386 mesa-vulkan-drivers:i386 python3-catkin
  ros-environment
Use 'sudo apt autoremove' to remove them.
The following NEW packages will be installed:
  python3-rospkg-modules
0 upgraded, 1 newly installed, 0 to remove and 5 not upgraded.
716 not fully installed or removed.
Need to get 0 B/24.2 kB of archives.
After this operation, 135 kB of additional disk space will be used.
Do you want to continue? [Y/n] Y
(Reading database ... 306924 files and directories currently installed.)
Preparing to unpack .../python3-rospkg-modules_1.5.0-1_all.deb ...
Unpacking python3-rospkg-modules (1.5.0-1) ...
dpkg: error processing archive /var/cache/apt/archives/python3-rospkg-modules_1.
5.0-1_all.deb (--unpack):
 trying to overwrite '/usr/lib/python3/dist-packages/rospkg/__init__.py', which 
is also in package python3-rospkg 1.3.0-1
Errors were encountered while processing:
 /var/cache/apt/archives/python3-rospkg-modules_1.5.0-1_all.deb
E: Sub-process /usr/bin/dpkg returned an error code (1)

```

This doesn't change the situation for all

mw46d, 2023-06-09:
It kind of does:-( I don't know, why you have all those old packages around. But you now found the next one:-( I actually don't have 'python3-rospkg' installed at all. But there is also a version 1.5 in the ros2 repos.
```
marcow@feather6:/tmp$ apt policy python3-rospkg
python3-rospkg:
  Installed: (none)
  Candidate: 1.5.0-100
  Version table:
     1.5.0-100 500
        500 http://packages.ros.org/ros2/ubuntu jammy/main amd64 Packages
     1.3.0-1 500
        500 http://us.archive.ubuntu.com/ubuntu jammy/universe amd64 Packages
        500 http://us.archive.ubuntu.com/ubuntu jammy/universe i386 Packages
```
so you should try to add it to the install list ( `sudo apt install python3-catkin-pkg python3-catkin-pkg-modules python3-rospkg-modules python3-rospkg` )

ruturajsambhusvt, 2023-06-09:
Hello, 
Weird, running `sudo apt install ros-iron-desktop` worked, but ROS2 Humble was installed instead of Iron. The Hello World example works though. Thank you very much for your help! :-)

mw46d, 2023-06-10:
You might have both now;-)  You can check with `ls /opt/ros/` and something like `env | egrep '^ROS_'` before and after a reboot (log-out & log-back-in) ;-)

Jaykumaran, 2025-04-14:
I tried with 

```
sudo apt install aptitude
sudo aptitude install ros-humble-desktop
```

ros2 installed.. but unsure about what other conflicts it can bring in.

As of now there is no problem.