ros2/ros2 #1275: Python package not compiling due to some stderr

Kunal-Attri, 2022-05-25:
## Bug report

**Required Info:**

- Operating System: Pop OS 22.04 [based on Ubuntu 22.04]
- Installation type: Debian via apt [Humble]
- Version or commit hash: NA
- DDS implementation: NA
- Client library (if applicable): NA

#### Steps to reproduce issue
1. Pop OS have Python 3.10 by default.
2. Installing ROS 2 Humble.
3. After that, create a custom Pyhton package and try to build it using colcon, it gives the error.

#### Expected behavior
Build package successfully.

#### Actual behavior
Starting >>> py_pubsub
--- stderr: py_pubsub                   
/home/kunal/.local/lib/python3.10/site-packages/setuptools/command/install.py:34: SetuptoolsDeprecationWarning: setup.py install is deprecated. Use build and pip and other standards-based tools.
  warnings.warn(
---
Finished <<< py_pubsub [3.76s]

Summary: 1 package finished [4.02s]
  1 package had stderr output: py_pubsub

#### Additional information
This error is more or less because of Python 3.10. COuld you help me figure out how to correct it.

aprotyas, 2022-05-25:
This is just a warning. Your package should still be built and installed correctly. Can you verify that?

Kunal-Attri, 2022-05-26:
Yes, I also expected but behaviour. Although, there is a install and build directories formed, but ros2 cannot detect any packages.[Attached screenshot.]
I also tried it with a cpp build, that is working fine.

[image]

nuclearsandwich, 2022-05-26:
Without being able to see the contents of the py_pubsub package, this looks like an issue with with the package configuration and not related to the deprecation warning, which is shown for every Python package built with colcon since they all use the deprecated setup.py installation method.

If this package is in a public repository on GitHub could you link to it?

If not, the first thing I would check is what the recursive contents of the package install directory are (`install/py_pubsub`) since that will help us to determine whether the package is configured correctly for `ros2 run`

Kunal-Attri, 2022-05-26:
I am referring to this tutorial: https://docs.ros.org/en/humble/Tutorials/Writing-A-Simple-Py-Publisher-And-Subscriber.html

Also the recursive contents of install/py_pubsub are as in screenshot:
[image]

Kunal-Attri, 2022-05-26:
Also, refer following repo, whereI have put up all the workspace files:
https://github.com/Kunal-Attri/dev_ws

aprotyas, 2022-05-26:
Have you sourced your workspace's `setup.bash` file? At any rate, I would suggest moving this conversation to answers.ros.org.

Kunal-Attri, 2022-05-26:
https://github.com/ros2/ros2/issues/1275#issuecomment-1138063118

In the screenshot visible, I have sourced the setup.bash file.

clalancette, 2022-05-26:
So the problem here is that you are using too new of a version of setuptools.  You can see that from this:

```
/home/kunal/.local/lib/python3.10/site-packages/setuptools/command/install.py
```

What I'll suggest is that you `pip3 uninstall setuptools`, which should remove the custom version and then use  the packaged version.  Then clean your workspace and rebuild.  See the known issues about Humble: https://docs.ros.org/en/humble/Releases/Release-Humble-Hawksbill.html#known-issues

Kunal-Attri, 2022-05-26:
Thanks a lot for your support. The error has been sorted out.
Kudos to you!