ros2/ros2 #1277: Not able to move joints using Joint State Publisher with ROS2 Humble

fvatansever, 2022-06-06:
Hello,

After launching robots on RViz2 with Joint State Publisher on ROS2 Humble Binary Install, I have noticed that the GUI does not include any joints to move. I have tried this with 3 other robots and different packages, the result did not change. When these packages are launched with ROS2 Foxy, everything seems to be working fine. 

The Joint State Publisher GUI output on ROS2 Humble is given below.

[image]

Also, when I echo the `/joint_states` I can see the joints in the output as below.

> name:
> - right_gripper_finger_joint
> - left_gripper_finger_joint
> - joint_1
> - joint_2
> - joint_3
> - joint_4
> - joint_5
> - joint_6
> position:
> - 0.0
> - 0.0
> - 0.0
> - 0.0
> - 0.0
> - 0.0
> - 0.0
> - 0.0

clalancette, 2022-06-06:
Can you give a bit more details here?  In particular, what version of `ros-humble-joint-state-publisher` are you using, which robots are you trying to launch, and how are you launching everything?

I just tried out the tutorial at https://docs.ros.org/en/humble/Tutorials/URDF/Building-a-Movable-Robot-Model-with-URDF.html with `ros-humble-joint-state-publisher` 2.3.0, and it seems to be working as expected for me there.

fvatansever, 2022-06-07:
The version of the `ros-humble-joint-state-publisher` that I use is also 2.3.0. I have tried launching generally custom robots, besides Kuka KR210 and UR5. All these robots work fine with ROS2 Foxy. Also, I tried the tutorial that you mentioned, it is the same.

[image]

Here is a simple launch file that I use,

      
      from launch import LaunchDescription
      from launch_ros.actions import Node
      from ament_index_python.packages import get_package_share_directory
      import os
      
      
      def generate_launch_description():
          
          pkg_description = get_package_share_directory('kuka_kr210')
          urdf_file = os.path.join(pkg_description, "urdf", 'kr210.urdf')   
      
          joint_state_publisher_node = Node(
              package="joint_state_publisher_gui",
              executable="joint_state_publisher_gui",
          )
          robot_state_publisher_node = Node(
              package="robot_state_publisher",
              executable="robot_state_publisher",
              output="both",
              arguments=[urdf_file],
          )
          rviz_node = Node(
              package="rviz2",
              executable="rviz2",
              name="rviz2",
              output="log",
          )
      
          nodes_to_run = [
              joint_state_publisher_node,
              robot_state_publisher_node,
              rviz_node
          ]
      
          return LaunchDescription(nodes_to_run)

fvatansever, 2022-06-07:
Okay, after some debugging, I have managed to fix this problem. The logs hadthe error below;

      [joint_state_publisher_gui-1]   File "/opt/ros/humble/lib/python3.10/site-packages/joint_state_publisher_gui/joint_state_publisher_gui.py", line 88, in __init__
      [joint_state_publisher_gui-1]     self.slider.setValue(RANGE/2)
      [joint_state_publisher_gui-1] TypeError: setValue(self, int): argument 1 has unexpected type 'float'

After converting the parameter of the `setValue(self, int)` functions to integer in the  file `/opt/ros/humble/lib/python3.10/site-packages/joint_state/publisher_gui/joint_state_publisher_gui.py`, the problem resolved.

I just changed all the `setValue(VALUE)` into `setValue(int(VALUE))`. After that, everything seems to be working as they are expected.

clalancette, 2022-06-07:
> I just changed all the `setValue(VALUE)` into `setValue(int(VALUE))`. After that, everything seems to be working as they are expected.

Yes, version 2.3.0 of joint_state_publisher_gui should have had that.  This is still in testing in Humble, but it should be in the next sync.

In any case, glad you have this fixed, and it should be fixed for everyone soon.