ros2/ros2 #1279: Cancel Goal with a Description in Action Server

ravijo, 2022-06-11:
This may not be an appropriate place to post my issue. Therefore, I apologize beforehand.  [The issue is posted in ROS answers](https://answers.ros.org/question/401851/).

**Required Info:**
- Operating System:
  - Ubuntu 20.04
- Installation type:
  - binaries
- ROS version:
  - foxy

While writing an action server, I could not find a way to provide a message to the client upon canceling a goal. [Please see the following code snippet](https://github.com/ros2/demos/blob/b0c1389ae61172788cccf4eb4e34b3b84c408a9d/action_tutorials/action_tutorials_cpp/src/fibonacci_action_server.cpp#L60-L63):
```
if (goal->order > 9000) {
  return rclcpp_action::GoalResponse::REJECT;
}
```

A better approach could be providing a reason along with a message, such as shown in the following pseudo code:
```
if (goal->order > 9000) {
  return rclcpp_action::GoalResponse::REJECT ("Order, i.e., " << goal->order << " is more than allowed value");
}
```

Please consider a situation wherein the goal has multiple parameters to check, as shown below:
```
if (goal->param_a < 0) {
  return rclcpp_action::GoalResponse::REJECT;
}
if (goal->param_b == true) {
  return rclcpp_action::GoalResponse::REJECT;
}
if (goal->param_c.size()  > 10) {
  return rclcpp_action::GoalResponse::REJECT;
}
```

In the above case, I believe it is essential for the client to know why the request got rejected.

sloretz, 2022-06-30:
> In the above case, I believe it is essential for the client to know why the request got rejected.

What is the client going to do when a goal it requests is rejected? If all the client gets is a human readable string from the action server, probably all it's going to do is log it. I'd recommend making the action server log it instead.

```c++
if (goal->order > 9000) {
  RCLCPP_WARN(this->get_logger(), "Order, i.e., %d is more than allowed value", goal->order);
  return rclcpp_action::GoalResponse::REJECT;
}
```

gavanderhoorn, 2022-06-30:
I'd also like it very much if we could somehow arrive at a solution where action servers can return reasons for why goals were rejected.

For now I've worked around it by accepting a goal -- even though I already know I won't do anything with it -- and then immediately cancelling it.

The action definition then includes both a human readable error message as well as a numeric error / status code in the result, which makes automated handling of such cases feasible.

I don't really like this work-around though, and I'd like to understand the rationale for not allowing something like what the ROS 1 implementation supported (specific to `FollowJointTrajectory`, but it illustrates the idea):

```c++
...
control_msgs::FollowJointTrajectoryResult rslt;
rslt.error_code = control_msgs::FollowJointTrajectoryResult::INVALID_GOAL;
rslt.error_string = "Rejected because ...";
gh.setRejected(rslt, reject_msg);
```

> I'd recommend making the action server log it instead.

this is not really an appropriate solution I believe. The action server could be (very) remote to the client, and this leaves the client completely in the dark as to why it receives a rejection.

gavanderhoorn, 2022-07-06:
@fujitatomoya: I'm not sure I understand what you're saying.

I have plenty of applications that interpret numeric return/error/status codes embedded in Action goal results.

State machines make this very natural.

> atm, i think general numeric error / status code would be enough with logging information on action server why rejected in details.

logging a message on the side of the server is definitely not sufficient.

How would an action client know *why* its goal was rejected? Say a server just needs to communicate something like a `EAGAIN` (but then the ROS equivalent). The only thing we can do right now is reject (without any indication why) and log on the server side "please client, if you read this, or your human, try again later" ...

I really don't understand what the argument is for not allowing servers to communicate a reason for rejection.

fujitatomoya, 2022-07-06:
@gavanderhoorn sorry for the confusion.

> How would an action client know why its goal was rejected?

adding implementation specific messages could help, at least we can log the reason on client side, but i was just thinking that this is not really good interface.
i am not against adding the reason message, but i think sufficient(probably reconsideration required) `numeric error / status code` as common interface would help the application more about `what needs to be done next.`.

ravijo, 2022-07-07:
Thanks, @gavanderhoorn for elaborating on the issue precisely.

Vicidel, 2024-02-16:
Hello guys, I agree that this would be nice to have. Any simple solution to make this work?

KyleJosling, 2024-05-14:
Bumping this again, this would be really nice to have.

salogranada, 2024-06-12:
Agree; this would be nice to have. Would it be possible to be considered for inclusion in the future?

severin-lemaignan, 2024-09-18:
Another +1, with a specific example: we have an action server that expect a user-provided string + some optional markup  (think a string for speech-to-text for instance). If the markup syntax is invalid, we would like to reject the goal, providing a useful error message (eg, position of the syntax error in the input), so that we can provide helpful feedback to the user.