ros2/ros2 #1280: /duration.hpp(104,3): error C2059: syntax error: '('

arielc-brillianetor, 2022-06-12:
## Bug report
Hi,
I'm trying to build a project using ROS2 and boost, on Windows. When I build the project only with other packages which call boost but no use of ROS2 libraries, the project build successfully. If a build a clean project using the ROS2 libraries but without any boost libraries the project builds successfully.  But when I combine the two in the same CMakeLists.txt file and use both libraries, the boost and the ROS2 it fails.
I am quit new at reporting issues and ROS2 in general, so if I forgot something please let me know.
Thanks a lot for your help,
Ariel

- Operating System:
I'm using Windows 10

- These are the libraries I want to use:
rclcpp, rclcpp_action, Boost::system, Boost::random, Boost::date_time, Boost::regex

#### Additional information
Besides boost, one of the projects which I call using find_package() uses jsoncpp and websocketpp, which I installed using vcpkg as well as the boost libraries.

This is the output of my console window:
C:\ros_ws\create3_examples_ws>colcon build --merge-install --event-handlers console_direct+ --packages-select create3_demo
WARNING: Package name "brill_interfaceAI_cpp" does not follow the naming conventions. It should start with a lower case letter and only contain lower case letters, digits, underscores, and dashes.
WARNING: Package name "brill_messageFactory_cpp" does not follow the naming conventions. It should start with a lower case letter and only contain lower case letters, digits, underscores, and dashes.
Starting >>> create3_demo
-- Selecting Windows SDK version 10.0.22000.0 to target Windows 10.0.19044.
-- Found ament_cmake: 1.4.0 (C:/ros2_humble/share/ament_cmake/cmake)
-- Found rclcpp: 16.2.0 (C:/ros2_humble/share/rclcpp/cmake)
-- Found rosidl_generator_c: 3.2.0 (C:/ros2_humble/share/rosidl_generator_c/cmake)
-- Found rosidl_adapter: 3.2.0 (C:/ros2_humble/share/rosidl_adapter/cmake)
-- Found rosidl_generator_cpp: 3.2.0 (C:/ros2_humble/share/rosidl_generator_cpp/cmake)
-- Using all available rosidl_typesupport_c: rosidl_typesupport_fastrtps_c;rosidl_typesupport_introspection_c
-- Using all available rosidl_typesupport_cpp: rosidl_typesupport_fastrtps_cpp;rosidl_typesupport_introspection_cpp
-- Found rmw_implementation_cmake: 6.3.0 (C:/ros2_humble/share/rmw_implementation_cmake/cmake)
-- Found rmw_fastrtps_cpp: 6.3.0 (C:/ros2_humble/share/rmw_fastrtps_cpp/cmake)
-- Using RMW implementation 'rmw_fastrtps_cpp' as default
-- Found rclcpp_action: 16.2.0 (C:/ros2_humble/share/rclcpp_action/cmake)
-- Found std_msgs: 4.5.0 (C:/ros2_humble/share/std_msgs/cmake)
-- Found brill_interfaceAI_cpp: 0.0.0 (C:/ros_ws/create3_examples_ws/install/share/brill_interfaceAI_cpp/cmake)
-- Found brill_messageFactory_cpp: 0.0.0 (C:/ros_ws/create3_examples_ws/install/share/brill_messageFactory_cpp/cmake)
-- Found brill_api_cpp: 0.0.0 (C:/ros_ws/create3_examples_ws/install/share/brill_api_cpp/cmake)
-- Configuring done
-- Generating done
-- Build files have been written to: C:/ros_ws/create3_examples_ws/build/create3_demo
Microsoft (R) Build Engine version 16.11.2+f32259642 for .NET Framework
Copyright (C) Microsoft Corporation. All rights reserved.

  Checking Build System
  Building Custom Rule C:/ros_ws/create3_examples_ws/src/brillianetor/create3_demo/CMakeLists.txt
  Create3Bot.cpp
  Create3BotData.cpp
  Dock.cpp
  MoveToCoordinates.cpp
  Undock.cpp
  Create3API.cpp
C:\Projects\vcpkg\installed\x64-windows\include\websocketpp/frame.hpp(834,25): warning C4267: '=': conversion from 'size_t' to 'uint32_t', possible loss of data (compiling source file C:\ros_ws\create3_examples_ws\src\brillianetor\create3_demo\src\Create3API.cpp) [C:\ros_ws\create3_examples_ws\build\create3_demo\Create3API.vcxproj]
C:\Projects\vcpkg\installed\x64-windows\include\websocketpp/frame.hpp(834,25): warning C4267: '=': conversion from 'size_t' to 'uint32_t', possible loss of data (compiling source file C:\ros_ws\create3_examples_ws\src\brillianetor\create3_demo\src\Create3Bot.cpp) [C:\ros_ws\create3_examples_ws\build\create3_demo\Create3API.vcxproj]
C:\Projects\vcpkg\installed\x64-windows\include\websocketpp/common/md5.hpp(367,34): warning C4267: '+=': conversion from 'size_t' to 'websocketpp::md5::md5_word_t', possible loss of data (compiling source file C:\ros_ws\create3_examples_ws\src\brillianetor\create3_demo\src\Create3API.cpp) [C:\ros_ws\create3_examples_ws\build\create3_demo\Create3API.vcxproj]
C:\Projects\vcpkg\installed\x64-windows\include\websocketpp/sha1/sha1.hpp(176,28): warning C4267: '=': conversion from 'size_t' to 'unsigned int', possible loss of data (compiling source file C:\ros_ws\create3_examples_ws\src\brillianetor\create3_demo\src\Create3API.cpp) [C:\ros_ws\create3_examples_ws\build\create3_demo\Create3API.vcxproj]
C:\Projects\vcpkg\installed\x64-windows\include\websocketpp/common/md5.hpp(367,34): warning C4267: '+=': conversion from 'size_t' to 'websocketpp::md5::md5_word_t', possible loss of data (compiling source file C:\ros_ws\create3_examples_ws\src\brillianetor\create3_demo\src\Create3Bot.cpp) [C:\ros_ws\create3_examples_ws\build\create3_demo\Create3API.vcxproj]
C:\Projects\vcpkg\installed\x64-windows\include\websocketpp/sha1/sha1.hpp(176,28): warning C4267: '=': conversion from 'size_t' to 'unsigned int', possible loss of data (compiling source file C:\ros_ws\create3_examples_ws\src\brillianetor\create3_demo\src\Create3Bot.cpp) [C:\ros_ws\create3_examples_ws\build\create3_demo\Create3API.vcxproj]
C:\ros2_humble\include\rclcpp\rclcpp/utilities.hpp(250,51): warning C4003: not enough arguments for function-like macro invocation 'max' (compiling source file C:\ros_ws\create3_examples_ws\src\brillianetor\create3_demo\src\Create3Bot.cpp) [C:\ros_ws\create3_examples_ws\build\create3_demo\Create3API.vcxproj]
C:\ros2_humble\include\rclcpp\rclcpp/utilities.hpp(267,51): warning C4003: not enough arguments for function-like macro invocation 'min' (compiling source file C:\ros_ws\create3_examples_ws\src\brillianetor\create3_demo\src\Create3Bot.cpp) [C:\ros_ws\create3_examples_ws\build\create3_demo\Create3API.vcxproj]
C:\ros2_humble\include\rclcpp\rclcpp/utilities.hpp(284,51): warning C4003: not enough arguments for function-like macro invocation 'max' (compiling source file C:\ros_ws\create3_examples_ws\src\brillianetor\create3_demo\src\Create3Bot.cpp) [C:\ros_ws\create3_examples_ws\build\create3_demo\Create3API.vcxproj]
C:\ros2_humble\include\rclcpp\rclcpp/utilities.hpp(301,51): warning C4003: not enough arguments for function-like macro invocation 'min' (compiling source file C:\ros_ws\create3_examples_ws\src\brillianetor\create3_demo\src\Create3Bot.cpp) [C:\ros_ws\create3_examples_ws\build\create3_demo\Create3API.vcxproj]
C:\ros2_humble\include\rclcpp\rclcpp/duration.hpp(104,3): warning C4003: not enough arguments for function-like macro invocation 'max' (compiling source file C:\ros_ws\create3_examples_ws\src\brillianetor\create3_demo\src\Create3Bot.cpp) [C:\ros_ws\create3_examples_ws\build\create3_demo\Create3API.vcxproj]
C:\ros2_humble\include\rclcpp\rclcpp/duration.hpp(104,3): error C2059: syntax error: '(' (compiling source file C:\ros_ws\create3_examples_ws\src\brillianetor\create3_demo\src\Create3Bot.cpp) [C:\ros_ws\create3_examples_ws\build\create3_demo\Create3API.vcxproj]
C:\ros2_humble\include\rclcpp\rclcpp/duration.hpp(104,3): error C2143: syntax error: missing ',' before '?' (compiling source file C:\ros_ws\create3_examples_ws\src\brillianetor\create3_demo\src\Create3Bot.cpp) [C:\ros_ws\create3_examples_ws\build\create3_demo\Create3API.vcxproj]
C:\ros2_humble\include\rclcpp\rclcpp/duration.hpp(104,3): error C2059: syntax error: '?' (compiling source file C:\ros_ws\create3_examples_ws\src\brillianetor\create3_demo\src\Create3Bot.cpp) [C:\ros_ws\create3_examples_ws\build\create3_demo\Create3API.vcxproj]
C:\ros2_humble\include\rclcpp\rclcpp/duration.hpp(151,11): error C2535: 'rclcpp::Duration::Duration(void)': member function already defined or declared (compiling source file C:\ros_ws\create3_examples_ws\src\brillianetor\create3_demo\src\Create3Bot.cpp) [C:\ros_ws\create3_examples_ws\build\create3_demo\Create3API.vcxproj]
C:\ros2_humble\include\rclcpp\rclcpp/duration.hpp(102): message : see declaration of 'rclcpp::Duration::Duration' (compiling source file C:\ros_ws\create3_examples_ws\src\brillianetor\create3_demo\src\Create3Bot.cpp) [C:\ros_ws\create3_examples_ws\build\create3_demo\Create3API.vcxproj]
[... cut ...] _ws\create3_examples_ws\build\create3_demo\Create3API.vcxproj]
C:\ros2_humble\include\rclcpp\rclcpp/create_timer.hpp(112,31): warning C4003: not enough arguments for function-like macro invocation 'max' (compiling source file C:\ros_ws\create3_examples_ws\src\brillianetor\create3_demo\src\Create3Bot.cpp) [C:\ros_ws\create3_examples_ws\build\create3_demo\Create3API.vcxproj]
Failed   <<< create3_demo [6.14s, exited with code 1]

Summary: 0 packages finished [6.38s]
  1 package failed: create3_demo

clalancette, 2022-06-20:
I'm not 100% sure, but I think this may be a problem with the `min()`/`max()` macros that `#include <windows.h>` provides (and which is the cause of lots of problems).

In short, you should probably try to add `-DNOMINMAX` to your command-line arguments, which will avoid this problematic behavior.  See https://stackoverflow.com/questions/13416418/define-nominmax-using-stdmin-max for an example.  I *think* that will fix your problem.

arielc-brillianetor, 2022-06-21:
I think I found a solution, but I'm not sure what the wider implications are on other aspects of the project.
I added the following to all the headers which called headers from boost or windows:

#pragma push_macro("max")
#pragma push_macro("min")
#undef max
#undef min
#define _USE_MATH_DEFINES

Again, I don't really understand what each part does but like @clalancette said it had to do with min/max definitions.
Thanks a lot for helping!

arielc-brillianetor, 2022-06-21:
So I found a solution, even though I'm not sure what the wider implications are on other aspects of the project.
I added the following to all the headers which called headers tied to boost or to windows:

#pragma push_macro("max")
#pragma push_macro("min")
#undef max
#undef min
#define _USE_MATH_DEFINES
@clalancette thanks a lot for helping out

closed 2022-06-21 by arielc-brillianetor