ros2/ros2 #1283: The function IncludeLaunchDescription in ROS Humble run launch file twice!

BlueBirdHouse, 2022-06-17:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu Ubuntu 22.04 LTS
- Installation type:
  - binaries
- Client library (if applicable):
  - IncludeLaunchDescription of launch.actions 

#### Steps to reproduce issue

``` test.launch.py ``` -->
```
import os
from ament_index_python.packages import get_package_share_directory
from launch import LaunchDescription

from launch.actions import IncludeLaunchDescription
from launch.launch_description_sources import PythonLaunchDescriptionSource

def generate_launch_description():
    # Create the launch description and populate
    ld = LaunchDescription()

    default_launch_file_path = os.path.join(get_package_share_directory('robot_bringup'), 'launch')

    start_test2_cmd = IncludeLaunchDescription(
        PythonLaunchDescriptionSource(
            os.path.join(default_launch_file_path, 'test2.launch.py')
        )
    )
    ld.add_action(start_test2_cmd)

    return ld
```

``` test2.launch.py ``` -->
```
import os
from launch import LaunchDescription

def generate_launch_description():
    # Create the launch description and populate
    ld = LaunchDescription()

    print("I only need to run once.")

    return ld
```

#### Expected behavior

bluebird@bluebird-Jammy:~/Documents/InClassProjects/turtle_bot_ws$ ros2 launch robot_bringup test.launch.py 
[INFO] [launch]: All log files can be found below /home/bluebird/.ros/log/2022-06-17-10-17-38-337096-bluebird-Jammy-22777
[INFO] [launch]: Default logging verbosity is set to INFO
I only need to run once.

#### Actual behavior

bluebird@bluebird-Jammy:~/Documents/InClassProjects/turtle_bot_ws$ ros2 launch robot_bringup test.launch.py 
[INFO] [launch]: All log files can be found below /home/bluebird/.ros/log/2022-06-17-10-17-38-337096-bluebird-Jammy-22777
[INFO] [launch]: Default logging verbosity is set to INFO
I only need to run once.
I only need to run once.

#### Additional information

BlueBirdHouse, 2022-06-18:
My process reminds me that the following codes seem run only once.

import os
from launch import LaunchDescription
import launch

`test2.launch.py` -->
```
def generate_launch_description():

    ld = LaunchDescription()

    print_cmd = launch.actions.ExecuteProcess(
        cmd=['echo', 'I only need to run once.'], output='screen', shell=True)

    ld.add_action(print_cmd)

    return ld
```

We have the following results.

[image]

That is true, but why does the print function run twice?

adityapande-1995, 2022-07-08:
Investigation so far : 
* This issue exists in rolling as well, as expected
* This seems to be the line that is generating the prints : as it is running the function ``generate_launch_description()`` : https://github.com/ros2/launch/blob/6a2bf20fb952da15002b2bca61f56feeb6c0f46f/launch/launch/launch_description_sources/python_launch_file_utilities.py#L68

adityapande-1995, 2022-08-19:
The multiple prints are just due to the fact that the launch descriptions are processed twice because of the include tag, but the actual processes are run only once. Closing this issue as this is not a bug as such.

closed 2022-08-19 by adityapande-1995