ros2/ros2 #1287: "Error, pkgProblemResolver” for ROS Humble Installation on Ubuntu 22.04

Bon-1900, 2022-06-23:
After a fresh install of Ubuntu Server 22.04 on a Raspberry Pi 3 B+, I am still getting this error when following the installation instructions from here: https://docs.ros.org/en/humble/Installation/Ubuntu-Install-Debians.html

I might have made an error during the installation and start with a fresh installation again soon. 

```
ubuntu@ubuntu:~$ sudo apt install ros-humble-ros-base
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 apt : Depends: libapt-pkg6.0 (>= 2.4.5) but it is not going to be installed
       Depends: libsystemd0 but it is not installable
 bsdutils : PreDepends: libsystemd0 but it is not installable
 init : PreDepends: systemd-sysv but it is not going to be installed
 libprocps8 : Depends: libsystemd0 (>= 209) but it is not installable
 util-linux : PreDepends: libsystemd0 but it is not installable
              PreDepends: libudev1 (>= 183) but it is not installable
E: Error, pkgProblemResolver::Resolve generated breaks, this may be caused by held packages. 
```

_Originally posted by @maxpolzin in https://github.com/ros2/ros2/issues/1272#issuecomment-1163562269_

I have encountered the same issue when installing ROS Humble on a Parallels-based Ubuntu 22.04 VM. The full error message is as below:

```
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies.
 apt : Depends: libapt-pkg6.0 (>= 2.4.5) but it is not going to be installed
       Depends: libsystemd0 but it is not installable
 bsdutils : PreDepends: libsystemd0 but it is not installable
 dconf-service : Depends: default-dbus-session-bus or
                          dbus-session-bus
 init : PreDepends: systemd-sysv but it is not going to be installed
 libcolord2 : Depends: libudev1 (>= 196) but it is not installable
 libdbus-1-3 : Depends: libsystemd0 but it is not installable
               Recommends: dbus but it is not going to be installed
 libgudev-1.0-0 : Depends: libudev1 (>= 199) but it is not installable
 libhwloc15 : Depends: libudev1 (>= 183) but it is not installable
 libinput-bin : Depends: libudev1 (>= 183) but it is not installable
 libinput10 : Depends: libudev1 (>= 183) but it is not installable
 libprocps8 : Depends: libsystemd0 (>= 209) but it is not installable
 libpulse0 : Depends: libsystemd0 but it is not installable
 libqt5gui5 : Depends: libudev1 (>= 183) but it is not installable
 libsdl2-dev : Depends: libsdl2-2.0-0 (= 2.0.20+dfsg-2build1) but 2.0.20+dfsg-2ubuntu1.22.04.1 is to be installed
               Depends: libudev-dev but it is not going to be installed
 libusb-1.0-0 : Depends: libudev1 (>= 183) but it is not installable
 ros-humble-pcl-conversions : Depends: libpcl-dev but it is not going to be installed
 util-linux : PreDepends: libsystemd0 but it is not installable
              PreDepends: libudev1 (>= 183) but it is not installable
E: Error, pkgProblemResolver::Resolve generated breaks, this may be caused by held packages.

```

airbutt, 2022-06-25:
I have exact the same issue today. I tryed it a couple of times making the sudo apt update and sudo apt upgrade beforehand. 
Made always a fresh installation of Ubuntu Server 22.04 (64Bit)  with Raspbery PI imager.

Raspberry is a 3b+ 
during apt upgrade I got message that Kernel 5.15.0-1005-raspi must be updated to 5.15.0-1011-raspi and system to be reboot manually and I did so .
I checked also that the kernel is ok after reboot and it was.

Tried it with desktop and ros-base version.

Interestingly 3 Weeks ago everything works fine on same device and SD card. So something happend in between with the packages.

Cheers

Airbutt

christianrauch, 2022-06-25:
I am getting this issue on a fresh install (+ updates) of Ubuntu 22.04:

```
$ sudo apt install ros-humble-ros-base 
Paketlisten werden gelesen… Fertig
Abhängigkeitsbaum wird aufgebaut… Fertig
Statusinformationen werden eingelesen… Fertig
Einige Pakete konnten nicht installiert werden. Das kann bedeuten, dass
Sie eine unmögliche Situation angefordert haben oder, wenn Sie die
Unstable-Distribution verwenden, dass einige erforderliche Pakete noch
nicht erstellt wurden oder Incoming noch nicht verlassen haben.
Die folgenden Informationen helfen Ihnen vielleicht, die Situation zu lösen:

Die folgenden Pakete haben unerfüllte Abhängigkeiten:
 apt : Hängt ab von: libapt-pkg6.0 (>= 2.4.5) soll aber nicht installiert werden
       Hängt ab von: libsystemd0 ist aber nicht installierbar
 bsdutils : Hängt ab von (vorher): libsystemd0 ist aber nicht installierbar
 init : Hängt ab von (vorher): systemd-sysv
 ros-humble-ament-cmake : Hängt ab von: cmake
 ros-humble-ament-cmake-core : Hängt ab von: cmake
 shim-signed : Hängt ab von: grub-efi-amd64-signed soll aber nicht installiert werden oder
                              grub-efi-arm64-signed ist aber nicht installierbar
               Hängt ab von: grub2-common (>= 2.04-1ubuntu24)
 util-linux : Hängt ab von (vorher): libsystemd0 ist aber nicht installierbar
              Hängt ab von (vorher): libudev1 (>= 183) ist aber nicht installierbar
E: Fehler: Unterbrechungen durch pkgProblemResolver::Resolve hervorgerufen; dies könnte durch zurückgehaltene Pakete verursacht worden sein.
```

I don't get where the dependency on systemd comes from, but all those packages (libsystemd0, libudev1, util-linux, shim-signed, apt, bsdutils, cmake) are installed and updated. Can you please remove the dependencies on irrelevant system packages

giovanidiniz, 2022-06-26:
Having the same problem on a fresh 22.04 install. Is building from the source the way to go for now?

FarisHamdi, 2022-06-26:
Similar Issue here with a fresh ubuntu 22.04 install. I made sure to update and upgrade system packages prior to installation

`Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 apt : Depends: libapt-pkg6.0 (>= 2.4.5) but it is not going to be installed
       Depends: libsystemd0 but it is not installable
 bsdutils : PreDepends: libsystemd0 but it is not installable
 dconf-gsettings-backend : Depends: dconf-service (< 0.40.0-3.1~)
                           Depends: dconf-service (>= 0.40.0-3)
 init : PreDepends: systemd-sysv
 libcolord2 : Depends: libudev1 (>= 196) but it is not installable
 libdbus-1-3 : Depends: libsystemd0 but it is not installable
               Recommends: dbus
 libhwloc15 : Depends: libudev1 (>= 183) but it is not installable
 libignition-cmake2-dev : Depends: cmake
 libopenmpi-dev : Depends: openmpi-bin (>= 3.0.0-1)
                  Recommends: libcoarrays-openmpi-dev but it is not going to be installed
 libopenni2-0 : Depends: libudev1 (>= 183) but it is not installable
 libpulse0 : Depends: libsystemd0 but it is not installable
 libqt5designer5 : Depends: libqt5gui5 (>= 5.14.1) but it is not installable or
                            libqt5gui5-gles (>= 5.14.1) but it is not installable
 libqt5designercomponents5 : Depends: libqt5gui5 (>= 5.14.1) but it is not installable or
                                      libqt5gui5-gles (>= 5.14.1) but it is not installable
 libqt5help5 : Depends: libqt5gui5 (>= 5.0.2) but it is not installable or
                        libqt5gui5-gles (>= 5.0.2) but it is not installable
 libqt5printsupport5 : Depends: libqt5gui5 (>= 5.14.1) but it is not installable or
                                libqt5gui5-gles (>= 5.14.1) but it is not installable
 libqt5quickshapes5 : Depends: libqt5gui5 (>= 5.14.1) but it is not installable or
                               libqt5gui5-gles (>= 5.14.1) but it is not installable
                      Depends: libqt5quick5 (>= 5.14.1) but it is not installable or
                               libqt5quick5-gles (>= 5.14.1) but it is not installable
 libqt5quicktest5 : Depends: libqt5gui5 (>= 5.12.2) but it is not installable or
                             libqt5gui5-gles (>= 5.12.2) but it is not installable
                    Depends: libqt5quick5 (>= 5.0.2) but it is not installable or
                             libqt5quick5-gles (>= 5.0.2) but it is not installable
 libqt5quickwidgets5 : Depends: libqt5gui5 (>= 5.10.0) but it is not installable or
                                libqt5gui5-gles (>= 5.10.0) but it is not installable
                       Depends: libqt5quick5 (>= 5.15.1) but it is not installable or
                                libqt5quick5-gles (>= 5.15.1) but it is not installable
 libqt5svg5 : Depends: libqt5gui5 (>= 5.15.3+dfsg~) but it is not installable or
                       libqt5gui5-gles (>= 5.15.3+dfsg~) but it is not installable
 libqt5webkit5 : Depends: libqt5gui5 (>= 5.14.1) but it is not installable or
                          libqt5gui5-gles (>= 5.14.1) but it is not installable
                 Depends: libqt5quick5 (>= 5.15.3+dfsg~) but it is not installable or
                          libqt5quick5-gles (>= 5.15.3+dfsg~) but it is not installable
 libqt5widgets5 : Depends: libqt5gui5 (>= 5.15.1) but it is not installable or
                           libqt5gui5-gles (>= 5.15.1) but it is not installable
 libudev-dev : Depends: libudev1 (= 249.11-0ubuntu3) but it is not installable
 libusb-1.0-0 : Depends: libudev1 (>= 183) but it is not installable
 libvtk9-qt-dev : Depends: libqt5opengl5-dev but it is not installable
 libvtk9.1-qt : Depends: libqt5gui5 (>= 5.11.0~rc1) but it is not installable or
                         libqt5gui5-gles (>= 5.11.0~rc1) but it is not installable
 mpi-default-bin : Depends: openmpi-bin
 python3-pyqt5 : Depends: libqt5gui5 (>= 5.1.0) but it is not installable
                 Depends: libqt5gui5 (>= 5.15.1) but it is not installable or
                          libqt5gui5-gles (>= 5.15.1) but it is not installable
 python3-pyqt5.qtsvg : Depends: libqt5gui5 (>= 5.0.2) but it is not installable or
                                libqt5gui5-gles (>= 5.0.2) but it is not installable
 python3-pyside2.qtgui : Depends: libqt5gui5 (>= 5.15) but it is not installable
                         Depends: libqt5gui5 (>= 5.15.1) but it is not installable or
                                  libqt5gui5-gles (>= 5.15.1) but it is not installable
 python3-pyside2.qtsvg : Depends: libqt5gui5 (>= 5.15) but it is not installable or
[... 17 lines cut ...]
 ros-humble-foonathan-memory-vendor : Depends: cmake
 ros-humble-rqt-image-view : Depends: libqt5gui5 (>= 5.14.1) but it is not installable or
                                      libqt5gui5-gles (>= 5.14.1) but it is not installable
 ros-humble-rviz-common : Depends: libqt5gui5 (>= 5.2.0) but it is not installable or
                                   libqt5gui5-gles (>= 5.2.0) but it is not installable
                          Depends: libqt5gui5 but it is not installable
[... cut ...]
                        Depends: libqt5gui5 but it is not installable
 shim-signed : Depends: grub-efi-amd64-signed but it is not going to be installed or
                        grub-efi-arm64-signed but it is not installable
               Depends: grub2-common (>= 2.04-1ubuntu24)
 util-linux : PreDepends: libsystemd0 but it is not installable
              PreDepends: libudev1 (>= 183) but it is not installable
E: Error, pkgProblemResolver::Resolve generated breaks, this may be caused by held packages.
`

jakkaj, 2022-06-26:
I am seeing this too now in an Ubuntu-22.04 based VS Code Dev Container.

LanderU, 2022-06-26:
We're having the same issue, in our case, it affects ROS 2 rolling and humble. Is there any workaround for this issue?

y-kitamu, 2022-06-26:
I have the same issue with ubntu22.04 docker.
For a temporal solution, I hold the `libsystemd0` version before `apt-get upgrade` and unhold it just before `apt-get install ros-humble-desktop`:

```Dockerfile
FROM ubuntu:22.04

RUN apt-mark hold libsystemd0

RUN apt-get update && apt-get install locales -y
RUN locale-gen en_US en_US.UTF-8
RUN update-locale LC_ALL=en_US.UTF-8 LANG=en_US.UTF-8
RUN LANG=en_US.UTF-8

RUN apt-get update && apt-get install curl gnupg lsb-release -y
RUN curl -sSL https://raw.githubusercontent.com/ros/rosdistro/master/ros.key -o /usr/share/keyrings/ros-archive-keyring.gpg
RUN /bin/bash -c 'echo "deb [arch=$(dpkg --print-architecture) signed-by=/usr/share/keyrings/ros-archive-keyring.gpg] http://packages.ros.org/ros2/ubuntu $(source /etc/os-release && echo $UBUNTU_CODENAME) main"' | tee /etc/apt/sources.list.d/ros2.list > /dev/null

RUN apt-get update && apt-get upgrade -y
RUN apt-mark unhold libsystemd0
RUN apt-get install ros-humble-desktop -y
```

I'm not sure if this is a good solution, but for me, it works.

gavanderhoorn, 2022-06-26:
> Can you please remove the dependencies on irrelevant system packages?

it's very likely this is a symptom, not the cause.

Asking for symptoms to be hidden is not constructive.

christianrauch, 2022-06-26:
> > Can you please remove the dependencies on irrelevant system packages?
> 
> it's very likely this is a symptom, not the cause.
> 
> Asking for symptoms to be hidden is not very constructive.

I don't have a full grasp of what is happening with the package dependencies here, but I *think* incompatible package versions are causing this. ROS is not the first project encountering this. I've seen these package issues before in other projects.

> Asking for symptoms to be hidden is not very constructive.

You may think that the proper solution is to fix the package dependencies for this specific dependency issue. But it will leave the general problem open to occur again in future. I think this could be prevented in future by removing direct or indirect dependencies on unrelated system packages.

I am trying to stay constructive (while every new user of Ubuntu cannot install ROS anymore) :-) Just correct me if my assumptions about package dependencies are wrong.

just-think, 2022-06-26:
> 

Can you please specify what packages to remove? I have fresh installed ubunty 22.04 with the same problem. What to try first?

christianrauch, 2022-06-26:
@gavanderhoorn To be a bit more constructive here, can it be that some ROS packages depend on 32bit packages that collide with their corresponding 64bit versions?

It seems weird to me that `ros-humble-ament-cmake` would depend on `cmake:i386`:
```
$ apt-cache depends ros-humble-ament-cmake
ros-humble-ament-cmake
  Hängt ab von: cmake
    cmake:i386
  Hängt ab von: ros-humble-ament-cmake-core
  Hängt ab von: ros-humble-ament-cmake-export-definitions
  Hängt ab von: ros-humble-ament-cmake-export-dependencies
  Hängt ab von: ros-humble-ament-cmake-export-include-directories
  Hängt ab von: ros-humble-ament-cmake-export-interfaces
  Hängt ab von: ros-humble-ament-cmake-export-libraries
  Hängt ab von: ros-humble-ament-cmake-export-link-flags
  Hängt ab von: ros-humble-ament-cmake-export-targets
  Hängt ab von: ros-humble-ament-cmake-gen-version-h
  Hängt ab von: ros-humble-ament-cmake-libraries
  Hängt ab von: ros-humble-ament-cmake-python
  Hängt ab von: ros-humble-ament-cmake-target-dependencies
  Hängt ab von: ros-humble-ament-cmake-test
  Hängt ab von: ros-humble-ament-cmake-version
  Hängt ab von: ros-humble-ros-workspace
```

giovanidiniz, 2022-06-26:
Would it be possible to revert libsystemd0 to a previous version, install ros-humble-* packages and then upgrade it back? 
just thinking of a way to get folks going until we can figure out a proper solution

giovanidiniz, 2022-06-26:
I'm reporting things I'm trying in hopes it helps with troubleshooting.

I tried @y-kitamu apt hold solution with no success (bare metal).

I'm investigating the multi-arch hypothesis but not very knowledgeable in this.

christianrauch, 2022-06-27:
Ubuntu released updates today to several of the packages listed in the error messages. Just update all packages and this problem should be solved.

airbutt, 2022-06-28:
Yes, same for me, after apt update and apt upgrade the apt install ros-humble-desktop runs perfectly through with out issues. Question: what was finally the issue and have we to fear it comes back again with next package update or so. I have to give some insides in my report to my teacher.

Bon-1900, 2022-06-28:
The recent update seems to solve the pkgProblemResolver issue, but a new emerges. Potentialy for ARM-based system only. 

`$ sudo apt install ros-humble-desktop
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies.
 libsdl2-dev : Depends: libsdl2-2.0-0 (= 2.0.20+dfsg-2build1) but 2.0.20+dfsg-2ubuntu1.22.04.1 is to be installed
E: Unable to correct problems, you have held broken packages.
`

Bon-1900, 2022-06-29:
After installing the packages mentioned from source ( [libsdl2.0.20+dfsg-2ubuntu1.22.04.1](https://launchpad.net/ubuntu/+source/libsdl2/2.0.20+dfsg-2ubuntu1.22.04.1) & [libsdl2-dev 2.0.20+dfsg-2ubuntu1.22.04.1 (arm64 binary)](https://launchpad.net/ubuntu/jammy/arm64/libsdl2-dev/2.0.20+dfsg-2ubuntu1.22.04.1) ), ROS 2 was installed successfully. The 2 packages are both proposed version, hence the dependency issue.

closed 2022-06-29 by Bon-1900

rajeshpachaikani, 2022-07-06:
> Similar Issue here with a fresh ubuntu 22.04 install. I made sure to update and upgrade system packages prior to installation
>
> ```
> Reading package lists... Done
> Building dependency tree... Done [...]

Apt is installing other packages without any errors and there is no broken packages but I am still having this issue. I request this issue to be reopened and resolved for amd64 targets also.

Bon-1900, 2022-07-06:
Hopefully, [my closing comment above](https://github.com/ros2/ros2/issues/1287#issuecomment-1170028168) will be of help.  In any case, Humble appears to depend on many cutting-edge packages. Some of which may require installation from the source.  Alternatively, you could stay tuned for the official Ubuntu releases and bug fixes. 

> Hi everyone,
> Is there a fix with this yet? Following ros2 Humble Hawksbill Ubuntu installation I still manage to reach pkgProblemResolver Error at the "sudo apt install ros-humble-desktop" step.

ProSerb, 2022-07-06:
> Hopefully, [my closing comment above](https://github.com/ros2/ros2/issues/1287#issuecomment-1170028168) will be of help. In any case, Humble appears to depend on many cutting-edge packages. Some of [...]

I’m new to this, forgive me but how do I go about installing the packages from source? I install the tar.xs file then extract it but then what?

rajeshpachaikani, 2022-07-06:
Thank you for your reply. While waiting for Canonical to release cutting edge packages into the lts version of ubuntu, can we get a glimpse of what one could do to install humble? Is installing from source the only way or can we do anything else to get the package manager to work?

christianrauch, 2022-07-06:
Well, you can [install ROS2 on RHEL](https://docs.ros.org/en/humble/Installation/RHEL-Install-RPMs.html) (Red Hat Enterprise Linux). This should save you from any Ubuntu apt troubles.

Nils-Holgerson, 2022-07-07:
I am having the same problem (Ubuntu 22.04):

`ubun@ubun-VirtualBox:~$ sudo apt install ros-humble-desktop
[sudo] password for ubun: 
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 apt : Depends: libapt-pkg6.0 (>= 2.4.5) but it is not going to be installed
       Depends: libsystemd0 but it is not installable
 bsdutils : PreDepends: libsystemd0 but it is not installable
 dconf-gsettings-backend : Depends: dconf-service (< 0.40.0-3.1~)
                           Depends: dconf-service (>= 0.40.0-3)
 init : PreDepends: systemd-sysv
 libcolord2 : Depends: libudev1 (>= 196) but it is not installable
 libdbus-1-3 : Depends: libsystemd0 but it is not installable
               Recommends: dbus
 libgudev-1.0-0 : Depends: libudev1 (>= 199) but it is not installable
 libhwloc15 : Depends: libudev1 (>= 183) but it is not installable
 libignition-cmake2-dev : Depends: cmake
 libinput10 : Depends: libudev1 (>= 183) but it is not installable
              Depends: libinput-bin (>= 1.20.0-1ubuntu0.1)
 libopenmpi-dev : Depends: openmpi-bin (>= 3.0.0-1)
                  Recommends: libcoarrays-openmpi-dev but it is not going to be installed
 libopenni2-0 : Depends: libudev1 (>= 183) but it is not installable
 libpulse0 : Depends: libsystemd0 but it is not installable
 libqt5gui5 : Depends: libudev1 (>= 183) but it is not installable
 libudev-dev : Depends: libudev1 (= 249.11-0ubuntu3) but it is not installable
 libusb-1.0-0 : Depends: libudev1 (>= 183) but it is not installable
 mpi-default-bin : Depends: openmpi-bin
 ros-humble-ament-cmake : Depends: cmake
 ros-humble-ament-cmake-core : Depends: cmake
 ros-humble-foonathan-memory-vendor : Depends: cmake
 shim-signed : Depends: grub-efi-amd64-signed but it is not going to be installed or
                        grub-efi-arm64-signed but it is not installable
               Depends: grub2-common (>= 2.04-1ubuntu24)
 util-linux : PreDepends: libsystemd0 but it is not installable
              PreDepends: libudev1 (>= 183) but it is not installable
E: Error, pkgProblemResolver::Resolve generated breaks, this may be caused by held packages.
`

kenji-miyake, 2022-08-16:
This problem seems to be occurring again today. (Yesterday, there was no error.)
For detailed information, please refer to the issue in my project: https://github.com/autowarefoundation/autoware/issues/2797

I'll write simple steps to reproduce the problem below.

Please let me know if this should be reported elsewhere.

## How to reproduce

```docker
# From https://docs.ros.org/en/humble/Installation/Ubuntu-Install-Debians.html
from ubuntu:22.04
SHELL ["/bin/bash", "-o", "pipefail", "-c"]

RUN apt-get -y update && \
  apt-get -y install software-properties-common && \
  add-apt-repository universe

RUN apt-get -y update && apt-get -y install curl gnupg lsb-release && \
  curl -sSL https://raw.githubusercontent.com/ros/rosdistro/master/ros.key -o /usr/share/keyrings/ros-archive-keyring.gpg

RUN echo "deb [arch=$(dpkg --print-architecture) signed-by=/usr/share/keyrings/ros-archive-keyring.gpg] http://packages.ros.org/ros2/ubuntu $(source /etc/os-release && echo $UBUNTU_CODENAME) main" | tee /etc/apt/sources.list.d/ros2.list > /dev/null

RUN apt-get -y update && apt-get -y upgrade

RUN apt-get -y install ros-humble-desktop
```

```console
$ mkdir /tmp/humble
$ vim /tmp/humble/Dockerfile # Copy the above file content
$ docker build /tmp/humble
Sending build context to Docker daemon   2.56kB
Step 1/7 : from ubuntu:22.04
 ---> 3f4714ee068a
Step 2/7 : SHELL ["/bin/bash", "-o", "pipefail", "-c"]
 ---> Using cache
 ---> 344e16a7354c
Step 3/7 : RUN apt-get -y update &&   apt-get -y install software-properties-common &&   add-apt-repository universe
 ---> Using cache
 ---> fe73405d6f7f
Step 4/7 : RUN apt-get -y update && apt-get -y install curl gnupg lsb-release &&   curl -sSL https://raw.githubusercontent.com/ros/rosdistro/master/ros.key -o /usr/share/keyrings/ros-archive-keyring.gpg
 ---> Using cache
 ---> ca1f4f569ed5
Step 5/7 : RUN echo "deb [arch=$(dpkg --print-architecture) signed-by=/usr/share/keyrings/ros-archive-keyring.gpg] http://packages.ros.org/ros2/ubuntu $(source /etc/os-release && echo $UBUNTU_CODENAME) main" | tee /etc/apt/sources.list.d/ros2.list > /dev/null
 ---> Using cache
 ---> 5b6bcedb0d28
Step 6/7 : RUN apt-get -y update && apt-get -y upgrade
 ---> Using cache
 ---> 342b44be972e
Step 7/7 : RUN apt-get install ros-humble-desktop
 ---> Running in 4506af2a133d
Reading package lists...
Building dependency tree...
Reading state information...
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 libignition-math6-dev : Depends: libignition-cmake2-dev (>= 2.13.0) but 2.12.1-2~jammy is to be installed
E: Unable to correct problems, you have held broken packages.
The command '/bin/bash -o pipefail -c apt-get install ros-humble-desktop' returned a non-zero code: 100
```

clalancette, 2022-08-16:
> This problem seems to be occurring again today.

The issue today was a completely separate issue, and happened because of a problem importing gazebo packages into the ROS buildfarm.  It should be resolved now.

tonypascal, 2022-11-30:
I am still having this issue today after a fresh install...

tonypascal, 2022-12-01:
> After installing the packages mentioned from source ( [libsdl2.0.20+dfsg-2ubuntu1.22.04.1](https://launchpad.net/ubuntu/+source/libsdl2/2.0.20+dfsg-2ubuntu1.22.04.1) & [libsdl2-dev [...]

how do you do that please ?

aaronmaynard, 2023-01-01:
I have a resolution that worked for me (having the below error)
```
amaynard@CPR02067L:~$ sudo apt install ros-humble-desktop
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help resolve the situation:

The following packages have unmet dependencies:
 libignition-math6-dev : Depends: libignition-math6 (= 6.12.0-3~jammy) but 6.13.0-1~focal is to be installed
E: Unable to correct problems, you have held broken packages.

```
Following the steps [here](https://gazebosim.org/api/math/6.1/install.html), you can update the package manually.
I was then able to install Humble Desktop