ros2/ros2 #1289: On-Going Issues with Large Tickets

SteveMacenski, 2022-07-01:
Hi,

Since the ~drawn of man~ start of ROS 2, working with large topics has been an on-going problem that is still very significant today. The [ROS 2 description paper](https://www.science.org/doi/10.1126/scirobotics.abm6074) I wrote with Brian, Tully, Chris, and William benchmarked just as much, showing that messages larger than ~1MB really start to cause problems (Figure 3). 

The DDS vendors have long hand-waved these away as "edge cases" but this is not an edge case in robotics. Pretty much every robotic user is going to have at least one of the following: (1) camera (2) depth camera (3) 3D lidar or similar. These generate massive amounts of data, relative to DDS' current use, that are not being handled appropriately for ROS 2 to be used in typical robot applications. There's been discussions about potential ways to improve this or tune DDS settings to better handle for more than 2 years now, but alas we're now on the 2nd LTS distribution of ROS 2, and I still can't publish a 30 fps RGBD camera feed to Nav2. 

A low resolution camera at 640x320 RGB camera image is roughly 5MB in size. So I think the need is clear that we need to drastically improve the performance of ROS 2 w.r.t. larger topics. 

Other examples of where this is problematic:
- Within Samsung, we have teams that want to use cameras and depth sensors with manipulators for various activities in `rclpy`. They've found that they fundamentally cannot create sensor processing pipelines without manually composing nodes into the same process, and even then struggle to get full data rates. Further, since there is no dynamic composition in `rclpy`, any such activities would have to throw everything into the same `main()` no longer making a ROS-based system modular. 
- I have not even bothered looking at or porting `pcl_ros` to ROS 2 due to the lack of performant handling of large messages (even within the same process, even using intraprocess communication) due to the CPU/resource overhead and the impossibility of making functional pipelines without explicit use of composition

I bring up this topic as an entry point to trying to find solutions to this so that handling large topics are not being ignored moving forward since this is a substantive issue. Most robots need cameras or depth sensing and it seems at the moment that's not being seriously considered. 

Potential actions:
- If simply reconfiguring DDS profiles is the answer - doing that as the default profile
- If on the same CPU, using zero-copy communication that I believe Cyclone and Fast-DDS have implemented automatically without user modification
- Other performance improvements

Pleune, 2024-06-06:
Has there been any further discussion on this? I think it would be very reasonable to change the default fastdds profile to have a much larger shared pool to alleviate most folk's issues. We have just started porting over to ROS2, and this issues has taken up multiple days of our time to root cause and figure out how to work around. I think is unacceptable for ROS2 to have these issues out of the box, where simple thinks like viewing a low resolution (500x500) video in rqt does not work.

ciandonovan, 2024-06-06:
@Pleune the rmw_zenoh is the future IMO, but only released unstable as of now, need to compile ROS from source with support for it. Have used it extensively for streaming video, lidar, pointclouds, joystick teleop etc. over LAN/WiFi as well as across the WAN in a Wireguard tunnel with excellent throughput and latency, something just not practicable with DDS.

The rmw_zenoh also has support for POSIX shared memory, unsure if it's zero copy or not, but should be more than capable for what you describe. Have not tested that part personally though.

At the moment we still use CycloneDDS over the local loopback, which isn't the most efficient approach but it does work reliably, can recommend that for the moment until rmw_zenoh is released as stable with binaries.

IamPhytan, 2024-06-07:
> the rmw_zenoh is the future IMO, but only released unstable as of now, need to compile ROS from source with support for it.

As explained in [this webinar](https://www.youtube.com/live/m5x7QABH1xI?t=3628) about the Zenoh ROS 2 RMW, you don't need to compile ROS from source to use rmw_zenoh. The rmw_zenoh can be used with a regular installation of ROS 2 Jazzy. On the other side, yes, the middleware must be built from source in a colcon workspace.

EduPonz, 2024-06-07:
Starting from ROS 2 Jazzy, it is possible to configure Fast DDS to run in _LARGE_DATA_ mode. This is just a matter of exporting an env var `export FASTDDS_BUILTIN_TRANSPORTS=LARGE_DATA`, you can check the docs [here](https://fast-dds.docs.eprosima.com/en/latest/fastdds/use_cases/tcp/tcp_with_multicast_discovery.html). This automatically configures the Fast DDS builtin transport to a shared memory one (for inter host comms), and a TCP one (for exchanging data). In addition, a UDP transport is used for the multicast part of discovery so you still get automatic discovery of nodes across the network. You can check a short video demo [here](https://www.youtube.com/watch?v=_C-F7YRjbvk).

Pleune, 2024-06-07:
Thanks all for some very fast responses.

I did experiment with fastdds's large data mode, but I find that rclpy subscribers have horrible throughput of non POD messages unless the shared memory allocation is larger than the serialized message size. This can be fixed with the xml configuration file, but it is very hard to find the information to do that correctly. This is a file that works for me:
<details>
```
<?xml version="1.0" encoding="UTF-8"?>
<dds
	xmlns="http://www.eprosima.com/XMLSchemas/fastRTPS_Profiles">
	<profiles>
		<transport_descriptors>
			<!-- Create a descriptor for the new transport -->
			<transport_descriptor>
				<transport_id>shm_transport</transport_id>
				<type>SHM</type>
				<segment_size>10485760</segment_size>
			</transport_descriptor>
		</transport_descriptors>
		<!-- Default publisher profile -->
		<publisher profile_name="default publisher profile" is_default_profile="true">
			<qos>
				<publishMode>
					<kind>SYNCHRONOUS</kind>
				</publishMode>
				<data_sharing>
					<kind>AUTOMATIC</kind>
				</data_sharing>
			</qos>
			<historyMemoryPolicy>DYAMIC</historyMemoryPolicy>
		</publisher>
		<!-- Default subscriber profile -->
		<subscriber profile_name="default subscriber profile" is_default_profile="true">
			<qos>
				<data_sharing>
					<kind>AUTOMATIC</kind>
				</data_sharing>
			</qos>
			<historyMemoryPolicy>DYNAMIC</historyMemoryPolicy>
		</subscriber>
		<participant profile_name="SHMParticipant" is_default_profile="true">
			<rtps>
				<userTransports>
					<transport_id>shm_transport</transport_id>
				</userTransports>
				<useBuiltinTransports>false</useBuiltinTransports>
			</rtps>
		</participant>
	</profiles>
</dds>
```
</details>

Now, after seeing these responses I have tried switching to cyclonedds, which for these large image messages, works way way way better, going from 1-2fps in rqt, to full 60 without an issue. I would love to see an obvious note in the wiki suggesting switching to cyclone as a very easy workaround if working with images/pointclouds, particularly in python nodes.

Pleune, 2024-06-07:
I have proposed some changed to the wiki page that may have saved me a lot of time

https://github.com/ros2/ros2_documentation/pull/4522

iluetkeb, 2025-03-19:
@EduPonz What is the situation with LARGE_DATA on humble? This seems to have been available already in fastdds 2.6, which AFAIK is used in humble, but you mention it only for jazzy and later? We have some system still on humble and one thing I noticed is that the discovery server doesn't like it if you set that env variable.
@Pleune your suggested use of synchronous publishing on FastDDS has some drawbacks, like that publish takes longer, which is also shared with the default cyclone dds configuration. I'm wondering why you found this necessary, it does not match our experience. Could it be that you are using very small publish buffers? In that situation, synchronous publishing can sometimes be helpful to avoid data loss because it slows down the publishers, but I would suggest going for larger publish buffers instead, which should have the same effect while not slowing down the publisher.

Pleune, 2025-03-20:
So our usecase that lead me down this road was trying to send 5mp image messages at 60fps. latency was actually not a concern, I just needed the messages to flow reliably. I do understand that this is a extremely large amount of throughput to be handled by ros on a single topic... but that is what we need for our application. We really were not able to get FastRTPS to work for us in an acceptable way. And it worked without any issue or extra configuration in ROS1.

We did switch to cyclone for a little while, which worked "OK" for a time. But the (de-)serialization of dynamic length messages in python with cyclone is still not good. It was using up the majority of our system's CPU time whenever any python nodes were spun up, even with rqt. At least I think this is what was happening... I am certainly not an expert here.

In the end, since this discussion started, We discovered rmw_zenoh. Actually it's mention in this thread above was the first time we even came across any mention of it. Since then Our team ended up switching 100% to zenoh, and the ONLY issues we have had are when someone forgets and runs a non-zenoh command and it starts up a non-zenoh ros2 daemon that confuses everything. There is a note about this in the zenoh readme, so that is not a big deal. Aside from this we have had zero issues with zenoh over the last 8 months using the default configurations. It seems like even loaned messages are working? Either that or it is just otherwise efficient enough as is. We have not really had the time to dig into it.

I do still think that until one RMW implementation can cover all ros use-cases, there should be a pretty in depth comparison of all of the options in the wiki. Some of the details are quite hard to find without blindly searching through the GH issues. There is no mention of configuring zero-copy for fastrtps with > 500KB messages. And even now, there is zero mention at all of Zenoh on the wiki page. I get that it is still in development, and that a influx of support issues about it until things are released. But at least for my (fairly common?) usecase of streaming full resolution images it works by far the best with the least amount of setup hassle.

As it stands now, for a team like us to picking up ROS2 without access to any people with prior experience, it can be extremely difficult to even find all of the information needed to get something as simple as a high-resolution webcam open in rqt.

iluetkeb, 2025-03-24:
@Pleune Thanks for your comments! btw, I would appreciate a pointer to the "configuring zero-copy for fastrtps with > 500KB messages" that you mentioned.
Just one more remark regarding ROS1: Some things were "less hassle" with ROS1, but ROS1 also hid many issues, most notably large queue sizes. As long as the image was smooth in rviz, people were happy, but latency _is_ also important, and with ROS1 you could fluctuate from low to high latency without people noticing, when the change happened slowly and the image was still smooth (meaning data rate was kept up). This could cause problem with sluggish system reaction, however.
Similarly, TCP's automatic window sizing avoided many network related errors (such as different topics crowding each other out), but it could also lead to hard-to-find errors. So, in some sense, while I understand that it can be very frustrating, sometimes DDS can highlight hidden issues. That said, we're also exploring zenoh, for precisely the same reasons as you mentioned here.

Pleune, 2025-03-24:
I do appreciate the more robust engineering going into ros2. It is honestly a huge improvement, just a few speed bumps like this along the way.

Here is where I got most of my information with my 500k comment https://github.com/ros2/rmw_fastrtps/issues/739. And if you look at my previous comment in this thread, I have posted the xml settings I had to change.

My understanding is that the default shared memory size is just very small compared to image messages. Then I think if the image message does not fit into SHM, then it must be sent over tcp loopback instead, and then I guess the (de)serialization of very large dynamic sized messages is not very performant? I'm reaching here, but that was my takeaway anyways.

With our image messages being >10MB getting passed through sometimes 5 nodes, it really is a difficult problem to solve. With ROS1 we certainly did struggle with extremely high latency and buffer sizes. So in that way dealing with this has already been worth it. When ros2 gets to a point where all of this "just works" it will be amazing.