ros2/ros2 #1290: moveit-setup-assisant

andy2990, 2022-07-04:
## Bug report

**Required Info:**

- Operating System:ubuntu22.04
  - 
- Installation type:binaries
  - 
- Version or commit hash:humble
  - 
- DDS implementation:
  - 
- Client library (if applicable):
  - 

#### Steps to reproduce issue

```
i have installed the ros-humble-moveit-setup-assisant using "sudo apt install ros-humble-moveit-setup-assisant".
then i ran the command "ros2 launch moveit-setup-assistant setup-assistan.launch"
```

#### Expected behavior
show the moveit-setup-assistant interface
#### Actual behavior
[ERROR] [launch]: Caught exception in launch (see debug for traceback): Caught exception when trying to load file of format [launch]: Error trying to process rule "substitution":
Unknown substitution: arg
#### Additional information

----
## Feature request

#### Feature description

#### Implementation considerations

fujitatomoya, 2022-07-06:
I think this issue should be moved to https://github.com/ros-planning/moveit/tree/master/moveit_setup_assistant.

fujitatomoya, 2022-07-06:
CC: @clalancette

clalancette, 2022-07-06:
Yes, agreed.  Please open an issue on https://github.com/ros-planning/moveit .  Closing this out.

closed 2022-07-06 by clalancette