ros2/ros2 #1291: install ros2 in virtual machine environment

hopechenmin, 2022-07-07:
## Bug report

**Required Info:**

- Operating System:
  - 
- Installation type:
  - 
- Version or commit hash:
  - 
- DDS implementation:
  - 
- Client library (if applicable):
  - 

#### Steps to reproduce issue

```

```

#### Expected behavior

#### Actual behavior

#### Additional information

----
## Feature request

#### Feature description

#### Implementation considerations

Nils-Holgerson, 2022-07-07:
I am experiencing a similar problem with a Ubuntu 22.04 virtual machine and ROS2 Humble installation.

```
ubun@ubun-VirtualBox:~$ sudo apt install ros-humble-desktop
[sudo] password for ubun: 
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 apt : Depends: libapt-pkg6.0 (>= 2.4.5) but it is not going to be installed
       Depends: libsystemd0 but it is not installable
 bsdutils : PreDepends: libsystemd0 but it is not installable
 dconf-gsettings-backend : Depends: dconf-service (< 0.40.0-3.1~)
                           Depends: dconf-service (>= 0.40.0-3)
 init : PreDepends: systemd-sysv
 libcolord2 : Depends: libudev1 (>= 196) but it is not installable
 libdbus-1-3 : Depends: libsystemd0 but it is not installable
               Recommends: dbus
 libgudev-1.0-0 : Depends: libudev1 (>= 199) but it is not installable
 libhwloc15 : Depends: libudev1 (>= 183) but it is not installable
 libignition-cmake2-dev : Depends: cmake
 libinput10 : Depends: libudev1 (>= 183) but it is not installable
              Depends: libinput-bin (>= 1.20.0-1ubuntu0.1)
 libopenmpi-dev : Depends: openmpi-bin (>= 3.0.0-1)
                  Recommends: libcoarrays-openmpi-dev but it is not going to be installed
 libopenni2-0 : Depends: libudev1 (>= 183) but it is not installable
 libpulse0 : Depends: libsystemd0 but it is not installable
 libqt5gui5 : Depends: libudev1 (>= 183) but it is not installable
 libudev-dev : Depends: libudev1 (= 249.11-0ubuntu3) but it is not installable
 libusb-1.0-0 : Depends: libudev1 (>= 183) but it is not installable
 mpi-default-bin : Depends: openmpi-bin
 ros-humble-ament-cmake : Depends: cmake
 ros-humble-ament-cmake-core : Depends: cmake
 ros-humble-foonathan-memory-vendor : Depends: cmake
 shim-signed : Depends: grub-efi-amd64-signed but it is not going to be installed or
                        grub-efi-arm64-signed but it is not installable
               Depends: grub2-common (>= 2.04-1ubuntu24)
 util-linux : PreDepends: libsystemd0 but it is not installable
              PreDepends: libudev1 (>= 183) but it is not installable
E: Error, pkgProblemResolver::Resolve generated breaks, this may be caused by held packages.
```

hopechenmin, 2022-07-08:
>I had tried to install ros2 Humble in a Ubuntu22.04 virtual machine in VMware Platform and get the same issue as yours.

clalancette, 2022-07-28:
This is a duplicate of https://github.com/ros2/ros2/issues/1272 .  While we've fixed it as far as we can on the ROS 2 side, this is a problem that goes deeper with Ubuntu itself (and apt).  See https://salsa.debian.org/apt-team/apt/-/merge_requests/248

Given that there isn't anything we can do here, closing this out.

closed 2022-07-28 by clalancette