ros2/ros2 #1294: Issue listening to more than one topic at once.

Brac24, 2022-07-08:
## Bug report

**Required Info:**

- Operating System:
  Laptop: Ubuntu 20.04
  Desktop: Ubuntu 20.04
  Jetson Nano: Ubuntu 18.04
- Installation type:
  Laptop: ROS 2 Foxy Binaries
  Desktop: ROS 2 Foxy in Docker Container from following repo: https://github.com/NVIDIA-ISAAC-ROS/isaac_ros_common
  Jetson Nano: ROS 2 Foxy in Docker Container with the following image built from the following dockerfile: https://github.com/dusty-nv/jetson-containers/blob/master/Dockerfile.ros.slam

- Version or commit hash:
  - 
- DDS implementation:
  Jetson Nano (from dockerfile): rmw_cyclonedds_cpp
  Desktop: rmw_fastrtps_cpp
  Laptop: rmw_fastrtps_cpp
(I did test changing laptop DDS implementation to rmw_cyclonedds_cpp but problem persisted)

I have been trying to figure out this issue for a while now. I assume it is maybe a configuration change but no idea what it would be.
I have an intel realsense D455 that I am using and generating depth and color images and wanting to publish them to remote devices in my local network. At first I thought it was a bandwidth issue and I needed to configure my router but it works in one case which makes me believe it is not a bandwidth issue. In  the image below I am showing the devices I have. The arrows indicate the data flow of the intel realsense topics from one device to the remote device. Red arrow means I can't seem to receive data from the publisher but I can confirm that it is indeed at least publishing locally. Yellow arrow means that it works if I listen to a single topic but as soon as I listen to 2 topics the messages completely stop coming in. For example simply running ros2 topic hz realsenseTopic1 works. If I open a second terminal and run: ros2 topic hz realsenseTopic2 the terminals just freeze and no longer publish the topic rate. If I stop one of them eventually it will start working again.

[image]

What is interesting though is that listening to 2 topics on the jetson nano coming from my desktop it works perfectly fine. Both depth and color topics come in at close to 30 fps (~29.3fps). For the jetson nano I am using the following docker container for ROS 2 Foxy https://github.com/dusty-nv/jetson-containers/blob/master/Dockerfile.ros.slam. On my desktop I am using a docker container from nvidia. Specifically I run the run_dev.sh script from here https://github.com/NVIDIA-ISAAC-ROS/isaac_ros_common/blob/main/scripts/run_dev.sh to run the container. On my laptop I have a native install of ROS 2 Foxy using the debian distribution.

In the end the most important thing for me is to be able to receive camera data from the jetson nano  on to my laptop because the jetson nano will be on a mobile robot and I would want to be receiving data remotely and do some processing on the laptop.

Brac24, 2022-07-08:
So I was able to listen to two topic simultaneously using ros2 topic hz. What I did was changed the dds implementation of the jetson nano to rmw_fastrtps_cpp and the laptop to rmw_cyclonedds_cpp. So this mimcs the working setup of publishing from the Desktop to the Jetson Nano which is the configuration that works. 

But why did that fix it?

So now the yellow arrow going from Jetson Nano to Laptop I thought would now be green since the topic rate for each topic (I am listening to 2 topics) is 30 fps for each. But now the issue I have is when I visualize both the color and depth images in rviz2, after about 5 seconds both streams will freeze.

sloretz, 2024-02-23:
Thank you for opening an issue. I'm glad you found a workaround. Since Foxy has reached end of life I'll close this issue.

closed 2024-02-23 by sloretz