ros2/ros2 #1295: ros2 commands hang on first invocation

adamlm, 2022-07-14:
## Bug report

The ROS 2 `node list` and `topic list` commands hang on first invocation.

**Required Info:**

- Operating System:
  - Ubuntu 18.04

- Installation type:
  - binaries (installed through `apt`)

- Version or commit hash:
  - ROS: galactic
  - ros-base: 0.9.3
  - ros-core: 0.9.3
  - ros2cli: 0.13.3
 
- DDS implementation:
  - Fast-DDS
  - default configuration

- Client library (if applicable):
  - N/A

#### Steps to reproduce issue

1. Begin with no ROS processes running on the local machine or any other machine on the network.
2. Launch a new terminal session (no ROS installation sourced).
3. Source the ROS installation.
   ```shell
   user@host:~$ source /opt/ros/galactic/setup.bash
   ```
4. Open System Monitor (or equivalent) and apply the filter `ros`.
5. Confirm that no ROS-related processes are running.
6. In the terminal session, attempt to list ROS nodes.
   ```shell
   user@host:~$ ros2 node list
   ```
7. Observe that two ROS-related processes appear in System Monitor:
   - `ros2`
   - `_ros2_daemon`
8. Observe that the `ros2 node list` command has not returned.

#### Expected behavior

The `ros2 node list` command returns quickly and displays no discovered nodes.

#### Actual behavior

The `ros2 node list` command hangs indefinitely.

#### Additional information

The above process also applies to `ros2 topic list`.

If I leave the first terminal session alone and repeat the process in a second terminal session, both `node list` and `topic list` commands return as expected. This leads me to believe there is an issue with spawning the ROS 2 daemon.

The above issue does not seem to affect the Foxy distribution. I followed the listed steps but replaced `galactic` with `foxy`. When invoking the `ros2 node list` command, I saw the `ros2` process appear in System Monitor, but it completed, and only the `_ros2_daemon` process remained.

Neonardo1702, 2022-07-19:
I am experiencing similar issues in my setup
- ROS Galactic installed using apt
- Ubuntu 20.04
Points worth noting is that some nodes such as demo_nodes talker works properly, but listener hangs indefinitely. Seems like node that includes topic subscription is not able to run correctly. Reinstalling ROS2 and rebooting several times did not help with the issue.

adamlm, 2022-07-20:
I did some testing and collected more information.

I have a firewall enabled with the following defaults:
* deny (incoming)
* allow (outgoing)
* deny (routed)

and the following rules:
* `239.255.0.1 7400:7650/udp on wlp2s0 ALLOW IN Anywhere` (ROS 2 multicast - Domain 0)
* `224.0.0.7 10317:10318/udp on wlp2s0 ALLOW IN Anywhere` (Ignition Gazebo multicast - legacy)
* `239.255.0.7 10317:10318/udp on wlp2s0 ALLOW IN Anywhere` (Ignition Gazebo multicast)

I have two network interfaces:
* `wlp2s0`
* `virbr0`

I allow multicast traffic on `wlp2s0` but not on `virbr0`.

When I run the `talker`/`listener` demo nodes example on Galactic using **Cyclone DDS** (default for Galactic), I see that `ufw` blocks the multicast traffic on `virbr0`, which I expect. The `talker` node works fine, but the `listener` node never receives messages. Steps to reproduce (separate terminal sessions):

```shell
$ RMW_IMPLEMENTATION=rmw_cyclonedds_cpp ros2 run demo_nodes_cpp talker
```

```shell
$ RMW_IMPLEMENTATION=rmw_cyclonedds_cpp ros2 run demo_nodes_cpp listener
```

If I allow ROS 2 multicast traffic on `virbr0` or disable the firewall entirely, the `listener` node works as expected.

When I run the `talker`/`listener` demo nodes example on Galactic using **Fast DDS**, both nodes work as expected, and `ufw` does not report blocking traffic on `virbr0` (from what I can see in `journalctl`). Steps to reproduce (separate terminal sessions):

```shell
$ RMW_IMPLEMENTATION=rmw_fastrtps_cpp ros2 run demo_nodes_cpp talker
```

```shell
$ RMW_IMPLEMENTATION=rmw_fastrtps_cpp ros2 run demo_nodes_cpp listener
```

You may need to install the Fast DDS middleware. Follow the tutorial [here](https://docs.ros.org/en/galactic/Installation/DDS-Implementations.html).

Based on the above, I'm not sure if this issue is related to ROS 2 or Cyclone DDS. If it specific to Cyclone DDS, I will close this issue and open a new one with them.

adamlm, 2022-07-20:
After looking around in the Cyclone DDS issues, I can confirm it's not a ROS 2 issue. Issue eclipse-cyclonedds/cyclonedds#1337 appears to be related to this one.

The problem seems to be that Cyclone DDS arbitrarily chooses a network interface to use. In my case, the options were `virbr0` and `wlp2s0`. It happened to choose the `virbr0`, on which interface I block multicast traffic.

The solution I found was to manually specify which interface to use in the `cyclonedds.xml` file. Here is what I used, which I  modified from Cyclone DDS's [`README.md`](https://github.com/eclipse-cyclonedds/cyclonedds#run-time-configuration) and the DDS-Demonstrators [documentation](https://dds-demonstrators.readthedocs.io/en/latest/Teams/1.Hurricane/setupCycloneDDS.html#notes) that I stumbled upon:

```xml
<?xml version="1.0" encoding="UTF-8" ?>
<CycloneDDS xmlns="https://cdds.io/config" xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance" xsi:schemaLocation="https://cdds.io/config https://raw.githubusercontent.com/eclipse-cyclonedds/cyclonedds/master/etc/cyclonedds.xsd">
    <Domain Id="any">
        <General>
            <NetworkInterfaceAddress>wlp2s0</NetworkInterfaceAddress>
            <AllowMulticast>default</AllowMulticast>
            <MaxMessageSize>65500B</MaxMessageSize>
        </General>
    </Domain>
</CycloneDDS>
```

Save the above contents into `cyclonedds.xml`. From here, you can run the `talker`/`listener` examples on Galactic using Cyclone DDS as follows (separate terminal sessions):

```shell
$ CYCLONEDDS_URI=file://$PWD/cyclonedds.xml RMW_IMPLEMENTATION=rmw_cyclonedds_cpp ros2 run demo_nodes_cpp talker
```

```shell
$ CYCLONEDDS_URI=file://$PWD/cyclonedds.xml RMW_IMPLEMENTATION=rmw_cyclonedds_cpp ros2 run demo_nodes_cpp listener
```

Make sure you're in the same directory as the `cyclonedds.xml` file.

Closing this issue as it does not seem to be a bug, at least from the ROS 2 perspective.

closed 2022-07-20 by adamlm