ros2/ros2 #1297: ros2 debug build windows fails

michaelaeriksen, 2022-07-19:
## Bug report

**Required Info:**

- Operating System:
  - Windows 11, Visual Studio 2022
- Installation type:
  - from source 

I'm trying to build ros2 from source in debug mode on Windows 11 using Visual Studio 2022. But I get this when installing the python_d packages
```
C:\Python310\lib\site-packages\pip\_internal\locations\_distutils.py:9: DeprecationWarning: The distutils package is deprecated and slated for removal in Python 3.12. Use setuptools or check PEP 632 for potential alternatives
  from distutils.cmd import Command as DistutilsCommand
C:\Python310\lib\distutils\command\install.py:13: DeprecationWarning: The distutils.sysconfig module is deprecated, use sysconfig instead
  from distutils.sysconfig import get_config_vars
ERROR: numpy-1.18.4-cp38-cp38d-win_amd64.whl is not a supported wheel on this platform.
```

I then tried to follow the Build Instruction posted here https://github.com/ros2/ros2/tree/numpy-archives and that also generates lots of errors. 

For reference everything builds fine in release build, but I need a debug build for my development.

clalancette, 2022-07-26:
It looks like we would probably need an updated numpy package for this to work.  Since Windows 11 with Visual Studio 2022 isn't currently a supported platform, we may not do this anytime soon.  But if you do figure out how to get it to work, we'd be happy to hear about how to do it and document it for the future.

songyuc, 2023-07-10:
Does ROS2 support Win11, currently?

clalancette, 2023-07-10:
> Does ROS2 support Win11, currently?

Not officially, no.  Unofficially I know people have been making it work, but there may be some patching needed.

michaelaeriksen, 2023-07-10:
ROS2 builds fine on Win11 with a few mods. I use `vcpkg` for all dependencies instead of `Chocolatey` and it works great.

songyuc, 2023-07-12:
OK, I will try learning to build it on Win11 later.

clalancette, 2026-10-26:
We remove the debug builds, and updates Lyrical and Rolling to use Windows 11.  We've also switched to pixi.  Thus I think the spirit of this is done, so closing.  If you are still having Windows issues, please feel free to open a new issue.

closed 2026-10-26 by clalancette