ros2/ros2 #1301: [rclpy] node having many services is not responding

ravijo, 2022-07-26:
## Bug report

One of my nodes has many services, publishers, subscribers, and action servers. To be precise, there are 17 service servers, 1 publisher, 1 subscriber, and 1 action sever. This node is not responding. However, the node works when I remove any one item: publisher, subscriber, service server, or action server.

**Required Info:**

- Operating System:
  - Ubuntu 20.04.4 LTS
- Installation type:
  - binaries
- Version or commit hash:
  - ROS Foxy
- DDS implementation:
  - Default
- Client library:
  - rclpy

#### Steps to reproduce issue

**publisher_member_function.py**
```
from action_tutorials_interfaces.action import Fibonacci
import rclpy
from rclpy.action import ActionServer
from rclpy.node import Node
from std_msgs.msg import String
from std_srvs.srv import SetBool, Trigger

class MinimalPublisher(Node):

    def __init__(self):
        super().__init__('minimal_publisher')
        self.pub = self.create_publisher(String, 'pub', 10)
        self.sub = self.create_subscription(String, 'sub', self.listener_cb, 10)
        self.srv_1 = self.create_service(SetBool, 'srv_1', self.set_bool_cb)
        self.srv_2 = self.create_service(SetBool, 'srv_2', self.set_bool_cb)
        self.srv_3 = self.create_service(SetBool, 'srv_3', self.set_bool_cb)
        self.srv_4 = self.create_service(SetBool, 'srv_4', self.set_bool_cb)
        self.srv_5 = self.create_service(SetBool, 'srv_5', self.set_bool_cb)
        self.srv_6 = self.create_service(SetBool, 'srv_6', self.set_bool_cb)
        self.srv_7 = self.create_service(SetBool, 'srv_7', self.set_bool_cb)
        self.srv_8 = self.create_service(SetBool, 'srv_8', self.set_bool_cb)
        self.srv_9 = self.create_service(SetBool, 'srv_9', self.set_bool_cb)
        self.srv_10 = self.create_service(Trigger, 'srv_10', self.trigger_cb)
        self.srv_11 = self.create_service(Trigger, 'srv_11', self.trigger_cb)
        self.srv_12 = self.create_service(Trigger, 'srv_12', self.trigger_cb)
        self.srv_13 = self.create_service(Trigger, 'srv_13', self.trigger_cb)
        self.srv_14 = self.create_service(Trigger, 'srv_14', self.trigger_cb)
        self.srv_15 = self.create_service(Trigger, 'srv_15', self.trigger_cb)
        self.srv_16 = self.create_service(Trigger, 'srv_16', self.trigger_cb)
        self.srv_17 = self.create_service(Trigger, 'srv_17', self.trigger_cb)
        self.ac_srv = ActionServer(self, Fibonacci, 'fib', self.execute_cb)
        self.timer = self.create_timer(1, self.timer_cb)

    def listener_cb(self, msg):
        self.get_logger().info(msg.data)

    def execute_cb(self, goal_handle):
        fb = Fibonacci.Feedback()
        fb.partial_sequence = [0, 1]
        for i in range(1, goal_handle.request.order):
            fb.partial_sequence.append(fb.partial_sequence[i] + fb.partial_sequence[i - 1])
            goal_handle.publish_feedback(fb)
        goal_handle.succeed()
        result = Fibonacci.Result()
        result.sequence = fb.partial_sequence
        return result

    def timer_cb(self):
        msg = String(data='Hello from Publisher')
        self.pub.publish(msg)

    def set_bool_cb(self, req, res):
        res.message = 'Hello from SetBool Service'
        return res

    def trigger_cb(self, req, res):
        res.message = 'Hello from Trigger Service'
        return res

def main(args=None):
    rclpy.init(args=args)

    minimal_publisher = MinimalPublisher()

    rclpy.spin(minimal_publisher)
    rclpy.shutdown()

if __name__ == '__main__':
    main()
```

#### Actual behavior

Upon calling services or echoing on topic, the node is not saying anything, as shown below:
```console
ravi@dell:~/ros_ws$ ros2 service call /srv_1 std_srvs/srv/SetBool "{data: True}"
waiting for service to become available...
^C
ravi@dell:~/ros_ws$ ros2 topic echo /pub
^Cravi@dell:~/ros_ws$ 
```

#### Additional Info
The node works when I remove any one item: publisher, subscriber, service server, or action server.

ravijo, 2022-07-27:
My apologies. It has been cross-posted. Please see the following URL:

https://answers.ros.org/question/404288/rclpy-node-having-many-services-is-not-responding/

fujitatomoya, 2022-07-28:
This does not happen with https://github.com/ros2/ros2/commit/5a35507c37f3b0449e31cb4073cf0352bfd52b0b

```bash
root@tomoyafujita:~/ros2_ws/colcon_ws# ros2 node info /minimal_publisher
/minimal_publisher
  Subscribers:
    /sub: std_msgs/msg/String
  Publishers:
    /parameter_events: rcl_interfaces/msg/ParameterEvent
    /pub: std_msgs/msg/String
    /rosout: rcl_interfaces/msg/Log
  Service Servers:
    /minimal_publisher/describe_parameters: rcl_interfaces/srv/DescribeParameters
    /minimal_publisher/get_parameter_types: rcl_interfaces/srv/GetParameterTypes
    /minimal_publisher/get_parameters: rcl_interfaces/srv/GetParameters
    /minimal_publisher/list_parameters: rcl_interfaces/srv/ListParameters
    /minimal_publisher/set_parameters: rcl_interfaces/srv/SetParameters
    /minimal_publisher/set_parameters_atomically: rcl_interfaces/srv/SetParametersAtomically
    /srv_1: std_srvs/srv/SetBool
    /srv_10: std_srvs/srv/Trigger
    /srv_11: std_srvs/srv/Trigger
    /srv_12: std_srvs/srv/Trigger
    /srv_13: std_srvs/srv/Trigger
    /srv_14: std_srvs/srv/Trigger
    /srv_15: std_srvs/srv/Trigger
    /srv_16: std_srvs/srv/Trigger
    /srv_17: std_srvs/srv/Trigger
    /srv_2: std_srvs/srv/SetBool
    /srv_3: std_srvs/srv/SetBool
    /srv_4: std_srvs/srv/SetBool
    /srv_5: std_srvs/srv/SetBool
    /srv_6: std_srvs/srv/SetBool
    /srv_7: std_srvs/srv/SetBool
    /srv_8: std_srvs/srv/SetBool
    /srv_9: std_srvs/srv/SetBool
  Service Clients:

  Action Servers:
    /fib: action_tutorials_interfaces/action/Fibonacci
  Action Clients:

root@tomoyafujita:~/ros2_ws/colcon_ws# ros2 topic echo /pub
data: Hello from Publisher
---
data: Hello from Publisher
---
data: Hello from Publisher
---
data: Hello from Publisher
---
data: Hello from Publisher
---
^Croot@tomoyafujita:~/ros2_ws/colcon_wsros2 service call /srv_1 std_srvs/srv/SetBool "{data: True}" "
requester: making request: std_srvs.srv.SetBool_Request(data=True)

response:
std_srvs.srv.SetBool_Response(success=False, message='Hello from SetBool Service')
```

@ravijo it would be useful to see the debug information for the node with `ros2 run xxx --ros-args --log-level debug`. that could tell us more information. (like, service request came in or not? or does not even receive service request and so on.)

ravijo, 2022-07-28:
@fujitatomoya Thank you very much. Please see below for requested information:

```
$ ros2 run examples_rclpy_minimal_publisher publisher_member_function --ros-args --log-level debug
[DEBUG] [1658991674.201703102] [rcl]: Initializing node 'minimal_publisher' in namespace ''
[DEBUG] [1658991674.201825717] [rcl]: Using domain ID of '0'
[DEBUG] [1658991674.206718990] [rcl]: Initializing publisher for topic name '/rosout'
[DEBUG] [1658991674.206800134] [rcl]: Expanded topic name '/rosout'
[DEBUG] [1658991674.207169810] [rcl]: Publisher initialized
[DEBUG] [1658991674.207233769] [rcl]: Node initialized
[DEBUG] [1658991674.236790995] [rcl]: Initializing publisher for topic name '/parameter_events'
[DEBUG] [1658991674.236811904] [rcl]: Expanded topic name '/parameter_events'
[DEBUG] [1658991674.237031757] [rcl]: Publisher initialized
[DEBUG] [1658991674.237591051] [rcl]: Initializing service for service name 'minimal_publisher/describe_parameters'
[DEBUG] [1658991674.237602229] [rcl]: Expanded service name '/minimal_publisher/describe_parameters'
[DEBUG] [1658991674.237656464] [rmw_fastrtps_cpp]: ************ Service Details *********
[DEBUG] [1658991674.237663233] [rmw_fastrtps_cpp]: Sub Topic rq/minimal_publisher/describe_parametersRequest
[DEBUG] [1658991674.237666770] [rmw_fastrtps_cpp]: Pub Topic rr/minimal_publisher/describe_parametersReply
[DEBUG] [1658991674.237670352] [rmw_fastrtps_cpp]: ***********
[DEBUG] [1658991674.237879420] [rcl]: Service initialized
[DEBUG] [1658991674.237967576] [rcl]: Initializing service for service name 'minimal_publisher/get_parameters'
[DEBUG] [1658991674.237978925] [rcl]: Expanded service name '/minimal_publisher/get_parameters'
[DEBUG] [1658991674.238011496] [rmw_fastrtps_cpp]: ************ Service Details *********
[DEBUG] [1658991674.238019045] [rmw_fastrtps_cpp]: Sub Topic rq/minimal_publisher/get_parametersRequest
[DEBUG] [1658991674.238024700] [rmw_fastrtps_cpp]: Pub Topic rr/minimal_publisher/get_parametersReply
[DEBUG] [1658991674.238028843] [rmw_fastrtps_cpp]: ***********
[DEBUG] [1658991674.238256217] [rcl]: Service initialized
[DEBUG] [1658991674.238365463] [rcl]: Initializing service for service name 'minimal_publisher/get_parameter_types'
[DEBUG] [1658991674.238374163] [rcl]: Expanded service name '/minimal_publisher/get_parameter_types'
[DEBUG] [1658991674.238404784] [rmw_fastrtps_cpp]: ************ Service Details *********
[DEBUG] [1658991674.238410804] [rmw_fastrtps_cpp]: Sub Topic rq/minimal_publisher/get_parameter_typesRequest
[DEBUG] [1658991674.238414220] [rmw_fastrtps_cpp]: Pub Topic rr/minimal_publisher/get_parameter_typesReply
[DEBUG] [1658991674.238417271] [rmw_fastrtps_cpp]: ***********
[DEBUG] [1658991674.238662644] [rcl]: Service initialized
[DEBUG] [1658991674.238798365] [rcl]: Initializing service for service name 'minimal_publisher/list_parameters'
[DEBUG] [1658991674.238808482] [rcl]: Expanded service name '/minimal_publisher/list_parameters'
[DEBUG] [1658991674.238876534] [rmw_fastrtps_cpp]: ************ Service Details *********
[DEBUG] [1658991674.238883794] [rmw_fastrtps_cpp]: Sub Topic rq/minimal_publisher/list_parametersRequest
[DEBUG] [1658991674.238889109] [rmw_fastrtps_cpp]: Pub Topic rr/minimal_publisher/list_parametersReply
[DEBUG] [1658991674.238894076] [rmw_fastrtps_cpp]: ***********
[DEBUG] [1658991674.239112348] [rcl]: Service initialized
[DEBUG] [1658991674.239242505] [rcl]: Initializing service for service name 'minimal_publisher/set_parameters'
[DEBUG] [1658991674.239252815] [rcl]: Expanded service name '/minimal_publisher/set_parameters'
[DEBUG] [1658991674.239298005] [rmw_fastrtps_cpp]: ************ Service Details *********
[DEBUG] [1658991674.239305153] [rmw_fastrtps_cpp]: Sub Topic rq/minimal_publisher/set_parametersRequest
[DEBUG] [1658991674.239309968] [rmw_fastrtps_cpp]: Pub Topic rr/minimal_publisher/set_parametersReply
[DEBUG] [1658991674.239315211] [rmw_fastrtps_cpp]: ***********
[DEBUG] [1658991674.239479885] [rcl]: Service initialized
[DEBUG] [1658991674.239558381] [rcl]: Initializing service for service name 'minimal_publisher/set_parameters_atomically'
[DEBUG] [1658991674.239568681] [rcl]: Expanded service name '/minimal_publisher/set_parameters_atomically'
[DEBUG] [1658991674.239600191] [rmw_fastrtps_cpp]: ************ Service Details *********
[DEBUG] [1658991674.239637050] [rmw_fastrtps_cpp]: Sub Topic rq/minimal_publisher/set_parameters_atomicallyRequest
[DEBUG] [1658991674.239642664] [rmw_fastrtps_cpp]: Pub Topic rr/minimal_publisher/set_parameters_atomicallyReply
[DEBUG] [1658991674.239647508] [rmw_fastrtps_cpp]: ***********
[DEBUG] [1658991674.239814934] [rcl]: Service initialized
[DEBUG] [1658991674.240188784] [rcl]: Initializing publisher for topic name 'pub'
[DEBUG] [1658991674.240202222] [rcl]: Expanded topic name '/pub'
[DEBUG] [1658991674.240356972] [rcl]: Publisher initialized
[DEBUG] [1658991674.240416129] [rcl]: Initializing subscription for topic name 'sub'
[DEBUG] [1658991674.240426063] [rcl]: Expanded topic name '/sub'
[DEBUG] [1658991674.240510925] [rcl]: Subscription initialized
[DEBUG] [1658991674.240926774] [rcl]: Initializing service for service name 'srv_1'
[DEBUG] [1658991674.240939566] [rcl]: Expanded service name '/srv_1'
[DEBUG] [1658991674.241016108] [rmw_fastrtps_cpp]: ************ Service Details *********
[DEBUG] [1658991674.241024649] [rmw_fastrtps_cpp]: Sub Topic rq/srv_1Request
[DEBUG] [1658991674.241029935] [rmw_fastrtps_cpp]: Pub Topic rr/srv_1Reply
[... cut ...]
[DEBUG] [1658991700.766374088] [rcl]: Finalizing service
[DEBUG] [1658991700.766481068] [rcl]: Service finalized
[DEBUG] [1658991700.766488690] [rcl]: Finalizing node
[DEBUG] [1658991700.766493662] [rcl]: Finalizing publisher
[DEBUG] [1658991700.766583884] [rcl]: Publisher finalized
[DEBUG] [1658991700.767276771] [rcl]: Node finalized
[DEBUG] [1658991700.767300592] [rcl]: Timer canceled
[DEBUG] [1658991700.767308122] [rcl]: Shutting down ROS client library, for context at address: 0x7fd20389ca30
```

ravijo, 2022-07-28:
Here is an interesting analysis. In one terminal, I kept the service call command running as shown below:

```console
$ ros2 service call /srv_1 std_srvs/srv/SetBool "{data: True}"
waiting for service to become available...

```
Note that, this terminal is left as it is. In another terminal, I run the node as shown below:
```
$ ros2 run examples_rclpy_minimal_publisher publisher_member_function --ros-args --log-level debug
[DEBUG] [1658992156.103584899] [rcl]: Initializing node 'minimal_publisher' in namespace ''
[DEBUG] [1658992156.103614175] [rcl]: Using domain ID of '0'
[DEBUG] [1658992156.107387580] [rcl]: Initializing publisher for topic name '/rosout'
[DEBUG] [1658992156.107402981] [rcl]: Expanded topic name '/rosout'
[DEBUG] [1658992156.107617280] [rcl]: Publisher initialized
[DEBUG] [1658992156.107627194] [rcl]: Node initialized
[DEBUG] [1658992156.129684079] [rcl]: Initializing publisher for topic name '/parameter_events'
[DEBUG] [1658992156.129707632] [rcl]: Expanded topic name '/parameter_events'
[DEBUG] [1658992156.129908397] [rcl]: Publisher initialized
[DEBUG] [1658992156.130514629] [rcl]: Initializing service for service name 'minimal_publisher/describe_parameters'
[DEBUG] [1658992156.130531907] [rcl]: Expanded service name '/minimal_publisher/describe_parameters'
[DEBUG] [1658992156.130585571] [rmw_fastrtps_cpp]: ************ Service Details *********
[DEBUG] [1658992156.130594934] [rmw_fastrtps_cpp]: Sub Topic rq/minimal_publisher/describe_parametersRequest
[DEBUG] [1658992156.130599858] [rmw_fastrtps_cpp]: Pub Topic rr/minimal_publisher/describe_parametersReply
[DEBUG] [1658992156.130604170] [rmw_fastrtps_cpp]: ***********
[DEBUG] [1658992156.130842496] [rcl]: Service initialized
[DEBUG] [1658992156.130965988] [rcl]: Initializing service for service name 'minimal_publisher/get_parameters'
[DEBUG] [1658992156.130978590] [rcl]: Expanded service name '/minimal_publisher/get_parameters'
[DEBUG] [1658992156.131015938] [rmw_fastrtps_cpp]: ************ Service Details *********
[DEBUG] [1658992156.131023095] [rmw_fastrtps_cpp]: Sub Topic rq/minimal_publisher/get_parametersRequest
[DEBUG] [1658992156.131026152] [rmw_fastrtps_cpp]: Pub Topic rr/minimal_publisher/get_parametersReply
[DEBUG] [1658992156.131028741] [rmw_fastrtps_cpp]: ***********
[DEBUG] [1658992156.131199731] [rcl]: Service initialized
[DEBUG] [1658992156.131284403] [rcl]: Initializing service for service name 'minimal_publisher/get_parameter_types'
[DEBUG] [1658992156.131293182] [rcl]: Expanded service name '/minimal_publisher/get_parameter_types'
[DEBUG] [1658992156.131325800] [rmw_fastrtps_cpp]: ************ Service Details *********
[DEBUG] [1658992156.131332237] [rmw_fastrtps_cpp]: Sub Topic rq/minimal_publisher/get_parameter_typesRequest
[DEBUG] [1658992156.131335162] [rmw_fastrtps_cpp]: Pub Topic rr/minimal_publisher/get_parameter_typesReply
[DEBUG] [1658992156.131337768] [rmw_fastrtps_cpp]: ***********
[DEBUG] [1658992156.131497742] [rcl]: Service initialized
[DEBUG] [1658992156.131600254] [rcl]: Initializing service for service name 'minimal_publisher/list_parameters'
[DEBUG] [1658992156.131609757] [rcl]: Expanded service name '/minimal_publisher/list_parameters'
[DEBUG] [1658992156.131642812] [rmw_fastrtps_cpp]: ************ Service Details *********
[DEBUG] [1658992156.131649321] [rmw_fastrtps_cpp]: Sub Topic rq/minimal_publisher/list_parametersRequest
[DEBUG] [1658992156.131652847] [rmw_fastrtps_cpp]: Pub Topic rr/minimal_publisher/list_parametersReply
[DEBUG] [1658992156.131656254] [rmw_fastrtps_cpp]: ***********
[DEBUG] [1658992156.131819254] [rcl]: Service initialized
[DEBUG] [1658992156.131925336] [rcl]: Initializing service for service name 'minimal_publisher/set_parameters'
[DEBUG] [1658992156.131934641] [rcl]: Expanded service name '/minimal_publisher/set_parameters'
[DEBUG] [1658992156.131967481] [rmw_fastrtps_cpp]: ************ Service Details *********
[DEBUG] [1658992156.131973988] [rmw_fastrtps_cpp]: Sub Topic rq/minimal_publisher/set_parametersRequest
[DEBUG] [1658992156.131977731] [rmw_fastrtps_cpp]: Pub Topic rr/minimal_publisher/set_parametersReply
[DEBUG] [1658992156.131981299] [rmw_fastrtps_cpp]: ***********
[DEBUG] [1658992156.132166315] [rcl]: Service initialized
[DEBUG] [1658992156.132255527] [rcl]: Initializing service for service name 'minimal_publisher/set_parameters_atomically'
[DEBUG] [1658992156.132265298] [rcl]: Expanded service name '/minimal_publisher/set_parameters_atomically'
[DEBUG] [1658992156.132298757] [rmw_fastrtps_cpp]: ************ Service Details *********
[DEBUG] [1658992156.132321943] [rmw_fastrtps_cpp]: Sub Topic rq/minimal_publisher/set_parameters_atomicallyRequest
[DEBUG] [1658992156.132326681] [rmw_fastrtps_cpp]: Pub Topic rr/minimal_publisher/set_parameters_atomicallyReply
[DEBUG] [1658992156.132329609] [rmw_fastrtps_cpp]: ***********
[DEBUG] [1658992156.132493532] [rcl]: Service initialized
[DEBUG] [1658992156.132916834] [rcl]: Initializing publisher for topic name 'pub'
[DEBUG] [1658992156.132934162] [rcl]: Expanded topic name '/pub'
[DEBUG] [1658992156.133117212] [rcl]: Publisher initialized
[DEBUG] [1658992156.133178870] [rcl]: Initializing subscription for topic name 'sub'
[DEBUG] [1658992156.133187179] [rcl]: Expanded topic name '/sub'
[DEBUG] [1658992156.133277972] [rcl]: Subscription initialized
[DEBUG] [1658992156.133607113] [rcl]: Initializing service for service name 'srv_1'
[DEBUG] [1658992156.133619072] [rcl]: Expanded service name '/srv_1'
[... cut ...]
I immediately see service response on first terminal. However, when I try again, it does not work. Please see below:
```
$ ros2 service call /srv_1 std_srvs/srv/SetBool "{data: True}"
waiting for service to become available...
requester: making request: std_srvs.srv.SetBool_Request(data=True)

response:
std_srvs.srv.SetBool_Response(success=False, message='Hello from SetBool Service')

ravi@tm:~/del_ws$ ros2 service call /srv_1 std_srvs/srv/SetBool "{data: True}"
waiting for service to become available...

```

ravijo, 2022-07-28:
Update
-----------
It works with This does not happen with ROS Galactic.

fujitatomoya, 2022-07-31:
I confirmed that this cannot be observed in humble or rolling, and https://github.com/ros2/ros2/issues/1301#issuecomment-1197864161 makes that this is the problem for foxy only.

> I immediately see service response on first terminal. However, when I try again, it does not work. Please see below:

are you using the same host to issue the 2nd request? it seems to be another host?

with your environment, with `export RMW_IMPLEMENTATION=rmw_cyclonedds_cpp` can solve the problem? i am not sure the requirement but this would be nice to try.

ravijo, 2022-07-31:
> are you using the same host to issue the 2nd request? it seems to be another host?

Same host. The service calling command on terminal 1 is left to wait. However, when on terminal 2, `publisher_member_function` is executed, I can see the response on terminal 1. But trying again on terminal 1 shows "waiting for service to become available..." message.

> with your environment, with export RMW_IMPLEMENTATION=rmw_cyclonedds_cpp can solve the problem? i am not sure the requirement but this would be nice to try.

It works after installing `ros-foxy-rmw-cyclonedds-cpp` package.

fujitatomoya, 2022-07-31:
Thanks for checking. I think there have been many fixes during galactic for `rmw_fastrtps`.

@MiguelCompany @iuhilnehc-ynos any idea what could be related to this issue?

MiguelCompany, 2022-08-01:
@fujitatomoya @ravijo Does this work with foxy built from sources?

We released [Fast DDS v2.1.2](https://github.com/eProsima/Fast-DDS/releases/tag/v2.1.2) last week, but we had [an issue](https://github.com/ros2-gbp/fastrtps-release/issues/24) generating the binary package

ravijo, 2022-08-01:
> Does this work with foxy built from sources?

@MiguelCompany : I am sorry, I installed ROS Foxy from binaries using `apt-get`.

iuhilnehc-ynos, 2022-08-01:
@MiguelCompany 

> Does this work with foxy built from sources?

I am afraid not. I can use the latest ros2.repos on [foxy](https://github.com/ros2/ros2/blob/foxy/ros2.repos) to reproduce this issue, and also tried using the Fast-DDS with `tags/v2.1.2`, but the phenomenon is the same.

<details><summary> NG log </summary>
<p>

```shell
chenlh ros2-foxy $ RMW_IMPLEMENTATION=rmw_fastrtps_cpp ros2 run examples_rclpy_minimal_publisher publisher_member_function

$ cat /proc/821585/maps | grep libfastrtps
7f0c9521d000-7f0c95a0d000 r--p 00000000 08:12 39101099                   /home/chenlh/Projects/ROS2/ros2-foxy/install/lib/libfastrtps.so.2.1.2
7f0c95a0d000-7f0c9626c000 r-xp 007f0000 08:12 39101099                   /home/chenlh/Projects/ROS2/ros2-foxy/install/lib/libfastrtps.so.2.1.2
7f0c9626c000-7f0c9650b000 r--p 0104f000 08:12 39101099                   /home/chenlh/Projects/ROS2/ros2-foxy/install/lib/libfastrtps.so.2.1.2
7f0c9650b000-7f0c9651d000 r--p 012ed000 08:12 39101099                   /home/chenlh/Projects/ROS2/ros2-foxy/install/lib/libfastrtps.so.2.1.2
7f0c9651d000-7f0c9656b000 rw-p 012ff000 08:12 39101099                   /home/chenlh/Projects/ROS2/ros2-foxy/install/lib/libfastrtps.so.2.1.2

chenlh ros2-foxy $ RMW_IMPLEMENTATION=rmw_fastrtps_cpp ros2 topic echo /pub
<nothing received>

```

</p>
</details>

iuhilnehc-ynos, 2022-08-05:
@MiguelCompany @fujitatomoya 

I see that you have backported the [commit](https://github.com/eProsima/Fast-DDS/commit/a00453fc41a0f62b609d64642b4c4dca700e51c7) in 2.1.x, but it seems not to fix this issue. (Perhaps you intended to use it to fix another problem.).

Could you backport this [one](https://github.com/eProsima/Fast-DDS/pull/2079) into 2.1.x? I think it can fix this issue.

fujitatomoya, 2022-08-08:
Just FYI, https://github.com/eProsima/Fast-DDS/pull/2906 backport is under review.

EduPonz, 2022-08-09:
> Just FYI, [eProsima/Fast-DDS#2906](https://github.com/eProsima/Fast-DDS/pull/2906) backport is under review.

I just merged it

fujitatomoya, 2023-03-21:
This has been addressed. @clalancette can you close this?