ros2/ros2 #1310: tf2_kdl: Cannot locate rosdep definition for [orocos_kdl]

samuk, 2022-08-05:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 22.04 / Humble
- Installation type: https://github.com/linorobot/ros2me

#### Steps to reproduce issue

```
Install Humble/ 22:04 https://github.com/linorobot/ros2me
Follow instructions here: https://github.com/Agroecology-Lab/visual-multi-crop-row-navigation/tree/ROS2#dependencies

```

#### Expected behavior

Rosdep finds dependencies

#### Actual behavior

ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
tf2_kdl: Cannot locate rosdep definition for [orocos_kdl]
tf2_geometry_msgs: Cannot locate rosdep definition for [orocos_kdl]
tf2_eigen_kdl: Cannot locate rosdep definition for [orocos_kdl]

clalancette, 2022-08-05:
I don't know what a `ros2me` installation is, but this is correct behavior for Humble.  We removed `orocos_kdl` in favor of `orocos_kdl_vendor`.  That means that `orocos_kdl` is no longer released in Humble.  My guess is that your source code is out of date (you need to be on the `humble` branch), but I'm not sure.

closed 2022-08-05 by clalancette

samuk, 2022-08-05:
How should I use orocos_kdl_vendor in Humble please?

I found this https://github.com/ros2/orocos_kdl_vendor/tree/humble but I'm unclear how to use it?

ijnek, 2022-08-09:
I feel that this is now becoming more suited for a post on ROS Answers, but I think the issue you're facing is from the instructions of compiling tf2_geometry_msgs from source as instructed in https://github.com/Agroecology-Lab/visual-multi-crop-row-navigation/tree/ROS2#dependencies

You should be able to just skip the following lines from the instructions there:

```sh
git clone https://github.com/ros2/geometry2.git
cd geometry2
git fetch origin pull/533/head:pr-533
git checkout pr-533
```

and things should work.