ros2/ros2 #1312: Remapping action names does not work

Bjoernolav, 2022-08-11:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 22.04
- Installation type:
  - Docker container based on `ros:humble`

#### Steps to reproduce issue
Run the following launch file to launch the `fibonacci` action client from `action_tutorials_py` with remapping of the action name:
```python
from launch import LaunchDescription
from launch_ros.actions import Node

def generate_launch_description():
    node = Node(
        package='action_tutorials_py',
        executable='fibonacci_action_server',
        remappings=[
            ('fibonacci', '/remapped_fibonacci'),
            ('~/list_parameters', '/fibonacci_list_parameters'),
        ]
    )
    return LaunchDescription([
        node
    ])

```

#### Expected behavior
Launching the `fibonnaci_action_server` node running with the `/remapped_fibonacci` action server, and also exposing the `/fibonacci_list_parameters` service instead of the `/fibonacci_action_server/fibonacci_list_parameters` service.

#### Actual behavior
The service is remapped correctly, but the action is not.

Output from `ros2 action list`
```
➜  zb_ws ros2 action list
/fibonacci
```

Output from `ros2 service list`
```
➜  zb_ws ros2 service list
/fibonacci_action_server/describe_parameters
/fibonacci_action_server/get_parameter_types
/fibonacci_action_server/get_parameters
/fibonacci_action_server/set_parameters
/fibonacci_action_server/set_parameters_atomically
/fibonacci_list_parameters
```

clalancette, 2022-09-01:
Yeah, this is a feature for actions that we never implemented.  It is probably possible to implement it relatively straightforwardly in https://github.com/ros2/rcl/tree/rolling/rcl_action , but it just isn't done so currently.

There is a workaround that can be used, which is to remap all of the topics and services that are associated with an action.  For the `fibonacci` example, that would look something like:

```
ros2 run action_tutorials_py fibonacci_action_server --ros-args -r /fibonacci/_action/feedback:=/myout/_action/feedback -r /fibonacci/_action/status:=/myout/_action/status -r /fibonacci/_action/cancel_goal:=/myout/_action/cancel_goal -r /fibonacci/_action/get_result:=/myout/_action/get_result -r /fibonacci/_action/send_goal:=/myout/_action/send_goal
```

And for the client:
```
ros2 run action_tutorials_py fibonacci_action_client  --ros-args -r /fibonacci/_action/feedback:=/myout/_action/feedback -r /fibonacci/_action/status:=/myout/_action/status -r /fibonacci/_action/cancel_goal:=/myout/_action/cancel_goal -r /fibonacci/_action/get_result:=/myout/_action/get_result -r /fibonacci/_action/send_goal:=/myout/_action/send_goal
```

I don't know if we are going to have time to look at this soon, but we'll keep it on the list to eventually fix.

closed 2025-01-24 by fujitatomoya

christophebedard, 2025-03-25:
> Thanks! Will this be backported?

See https://github.com/ros2/rcl/pull/1220 for a Jazzy backport. It seems pretty straightforward, but it would be helpful if someone could build and test it just to validate it.