ros2/ros2 #1317: YAML config: Substitution happens for comments

akorb, 2022-08-19:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 20.04 LTS
- Installation type:
  - Binary
- Version or commit hash:
  - galactic
- DDS implementation:
  - Eclipse Cyclone DDS

#### Steps to reproduce issue
Add a comment like this to any configuration yaml file that is used:
```yaml
    # $(env NON_EXISTING_ENV_VAR)
```

and add `allow_subst="true"` to its referencing `param` tag:
```xml
    <param from="my_config.yaml" allow_substs="true"/>
```

#### Expected behavior

The comment should be ignored and it should work as usually.

#### Actual behavior

Ros tries to resolve the environment variable and crashes if it does not exist, even though it is commented out.

clalancette, 2022-08-19:
Can you give a complete example of the problem?

From the short snippet you've provided, it looks like you are using XML launch files.  In XML `#` is not the comment character; `

akorb, 2022-08-19:
I see. The comment is meant to be put into a yaml configuration file (`my_config.yaml`). For example:

```yaml
/**:
  ros__parameters:
    parameterA: 5

    # $(env NON_EXISTING_ENV_VAR)

    parameterB: true
```

And in the launch file `allow_substs` has to be set to true, otherwise the `$(env ...)` is not resolved:
```xml
<launch>
  <node pkg="my_node" exec="my_node" name="my_node">
    <param from="my_config.yaml" allow_substs="true"/>
  </node>
</launch>

```

clalancette, 2022-08-19:
OK, I see.  Thanks for the updated example, that helps a lot.

This is indeed a bug, and I can reproduce it locally on Rolling with the following:

turtlesim.yaml:
```
/turtlesim:
  ros__parameters:

    # $(env NON_EXISTING_ENV_VAR)

    background_b: 0
    background_g: 86
    background_r: 69
    holonomic: false
    qos_overrides:
      /parameter_events:
        publisher:
          depth: 1000
          durability: volatile
          history: keep_last
          reliability: reliable
    use_sim_time: false
```

turtlesim-launch.xml
```
<launch>
  <node pkg="turtlesim" exec="turtlesim_node" name="turtlesim">
    <param from="turtlesim.yaml" allow_substs="true"/>
  </node>
</launch>
```

@ivanpauno Since you did the initial substitution work, can you take a look?

ivanpauno, 2022-08-19:
Yes, I can take a look, it seems a bit tricky ...