ros2/ros2 #1318: Unable to Communicate between Ubuntu 20.04 Container and Ubuntu 20.04 Host

cardboardcode, 2022-08-20:
## Bug Report :bug: 

This bug is in regards to **the need to containerize the setup of ROS2 packages** in order to **avoid tedious environment conflict resolutions**. 

However, the bug described in this issue thread demonstrates that **dockerized ROS2 packages can only communicate with ROS2 packages actively running in other docker containers** and unable to communicate with ROS2 nodes that are running in native **Host**.

This may potentially highlight a need to have **excessively large memory capacity** to host multiple actively-running ROS2 Docker Containers, rather than be able to instantiate more memory-demanding ROS2 packages in native **Host** while still avoiding environment conflicts.

Eg. `osrf/ros:foxy-desktop` Docker Image requires at least **3.11GB**.

**Required Info:**
- Operating System:
  - **Ubuntu 20.04**
- Installation type:
  - Binaries for **Host** and **Container**
- Version or commit hash:
  - **ROS2 Foxy Fitzroy**
- DDS implementation:
  - **Fast-DDS** (Formerly-known as FAST-RTPS)
- Client library (if applicable):
  - `rclcpp`

#### Steps to reproduce issue

The command below instantiates a ROS2 node in a Docker Container and publishes `std_msgs::msg::String` on a ROS2 topic labelled, `/chatter`.

##### Talker [ Container ]
```bash
# Open new terminal
docker run -ti --rm --ipc host --net host -v /dev/shm:/dev/shm --user root osrf/ros:foxy-desktop ros2 run demo_nodes_cpp talker
```
##### Listener [ Host ]
```bash
# Open new terminal
source /opt/ros/foxy/setup.bash
ros2 run demo_nodes_cpp listener
```
#### Expected behavior :heavy_check_mark: 
**Talker** in **Container** should be able to publish onto `/chatter` ROS2 topic. **Listener** in **Host** should be able to subscribe to the same topic and output to the terminal the following:

##### Talker [ Container ]
```bash
[INFO] [1660974428.078278078] [talker]: Publishing: 'Hello World: 4'
[INFO] [1660974429.078255106] [talker]: Publishing: 'Hello World: 5'
[INFO] [1660974430.078242799] [talker]: Publishing: 'Hello World: 6'
[INFO] [1660974431.078225721] [talker]: Publishing: 'Hello World: 7'
[INFO] [1660974432.078207510] [talker]: Publishing: 'Hello World: 8'
[INFO] [1660974433.078177587] [talker]: Publishing: 'Hello World: 9'
[INFO] [1660974434.078181532] [talker]: Publishing: 'Hello World: 10'
[INFO] [1660974435.078144220] [talker]: Publishing: 'Hello World: 11'
```

##### Listener [ Host ]
```bash
[INFO] [1660974428.078838800] [listener]: I heard: [Hello World: 4]
[INFO] [1660974429.078711720] [listener]: I heard: [Hello World: 5]
[INFO] [1660974430.078709416] [listener]: I heard: [Hello World: 6]
[INFO] [1660974431.078692205] [listener]: I heard: [Hello World: 7]
[INFO] [1660974432.078687710] [listener]: I heard: [Hello World: 8]
[INFO] [1660974433.078352958] [listener]: I heard: [Hello World: 9]
[INFO] [1660974434.078650348] [listener]: I heard: [Hello World: 10]
[INFO] [1660974435.078621701] [listener]: I heard: [Hello World: 11]
```

#### Actual behavior :red_circle:  

The actual behaviour is, while **Talker** in **Container** is verified to be actively running, **Listener** in **Host** is unresponsive as shown below:
 
[image]

#### Additional information :book: 

The purpose of this issue thread is **to track the progress of resolving this problem by me**. 

In additional, it is **also a request for help from the wider ROS2 community**. 

## :warning: Help Needed

**Attempted Solutions** which I have tried thus far will be posted shortly as subsequent comments in great detail as efforts to debug this continues.

cardboardcode, 2022-08-20:
## Communication between Ubuntu 20.04 **Container** and Ubuntu 20.04 **Container**

As mentioned under **Bug Report** 2nd paragraph, **Talker** in **Container** is able to interact via its ROS2 topic `/chatter` with **Listener** in **Host** from within different docker containers.

The following is a minimal example to prove it:

#### Steps to Reproduce

Run **Talker** from within **Docker Container**.

```bash
# Open a new terminal
docker run -it --rm osrf/ros:foxy-desktop ros2 run demo_nodes_cpp talker
```
Run **Listener** from within **Docker Container**.

```bash
# Open a new terminal
docker run -it --rm osrf/ros:foxy-desktop ros2 run demo_nodes_cpp listener
```

#### Verify
You should see a similar terminal output:

[image]

If you do, you have just proven to yourself that **Container to Container** ROS2 interaction is possible when using docker containerization.

## :warning:

The issue, however, is proving whether **Container to Host** ROS2 interaction is possible when using docker containerization. At least, this minimal example shows the ROS2 works with Docker and that ROS2 communication is able to be established outside of a container.

cardboardcode, 2022-08-20:
## Attempted Solution 1

Googling around, I found a few solutions, notably many of which are from concerns richiware's solutions concerning adjusting `FASTRTPS_DEFAULT_PROFILES_FILE` `.xml` file configuration and/or setting the right **Docker Run Command** configurations.

This solution looks at simply setting the recommended **Docker Run Command** configurations like so:

```bash
docker run -ti \
--rm \
--ipc host \
--net host \
-v /dev/shm:/dev/shm \
--user root osrf/ros:foxy-desktop \
ros2 run demo_nodes_cpp talker
```

## :warning: Conclusion
No changes observed. Issue persists.

## References

1. https://answers.ros.org/question/370595/ros2-foxy-nodes-cant-communicate-through-docker-container-border/?answer=371655#post-id-371655

cardboardcode, 2022-08-20:
## Attempted Solution 2 :red_circle: 

Suspicions went to **hidden pre-existing environmental variables** that could be interfering with ROS2 communications between **Container** and **Host**.

As mentioned by Cam Salzburger under a MatLab Answers forum thread [1], there are the following variables that determine what is and what isn't within a ROS2 network.

1. `ROS_DOMAIN_ID` 
2. IP Subnet

This solution looks at ensuring that `ROS_DOMAIN_ID` has been set correctly for both **Container** and **Host**:

```bash
# Open new terminal
docker run -ti \
--rm \
--ipc host \
--net host \
-e ROS_DOMAIN_ID=2 \
-v /dev/shm:/dev/shm \
--user root osrf/ros:foxy-desktop \
ros2 run demo_nodes_cpp talker
```

```bash
# Open new terminal
source /opt/ros/foxy/setup.bash
export ROS_DOMAIN_ID=2
ros2 run demo_nodes_cpp listener
```

## :warning: Conclusion
No changes observed. Issue persists.

Running `sudo netstat -unlp | grep talker` reveals that `ROS_DOMAIN_ID` works as described in [2], setting the IP address port at **79XX**. 

```bash
udp        0      0 0.0.0.0:49087           0.0.0.0:*                           26615/talker        
udp        0      0 0.0.0.0:7900            0.0.0.0:*                           26615/talker        
udp        0      0 0.0.0.0:7912            0.0.0.0:*                           26615/talker        
udp        0      0 0.0.0.0:7913            0.0.0.0:*                           26615/talker        
udp        0      0 127.0.0.1:44244         0.0.0.0:*                           26615/talker
```
Running `sudo netstat -unlp | grep listener` reveals that the **IP Subnet** for both **Listener** and **Talker** are the same.

```bash
udp        0      0 0.0.0.0:37535           0.0.0.0:*                           26877/listener      
udp        0      0 127.0.0.1:39204         0.0.0.0:*                           26877/listener      
udp        0      0 0.0.0.0:7900            0.0.0.0:*                           26877/listener      
udp        0      0 0.0.0.0:7914            0.0.0.0:*                           26877/listener      
udp        0      0 0.0.0.0:7915            0.0.0.0:*                           26877/listener
``

However, **Talker** in **Container** is still not communicating with **Listener** in **Host**. Issue thus persists, even with `ROS_DOMAIN_ID`  and **IP Subnet** verified to be the same.

## References

1. https://www.mathworks.com/matlabcentral/answers/497662-trying-to-connect-to-an-active-node-ros2#answer_408571
2. https://docs.ros.org/en/foxy/Concepts/About-Domain-ID.html

cardboardcode, 2022-08-20:
## Attempted Solution 3 :red_circle: 

Further googling reveals this short comment by **wl** in [1]:
[image]
This seems to suggest switching to *Cyclone DDS* as the default RMW would resolve this issue.

```bash
Results pending... To be updated once done.
```

## References
1. https://answers.ros.org/question/370595/ros2-foxy-nodes-cant-communicate-through-docker-container-border/?answer=371655#post-id-371655

cardboardcode, 2022-08-20:
## Attempted Solution 4 :wave: 

The `.xml` solution, suggested by richiware in [1], has yet to be explored due to the unclear instructions left regarding how to properly set a given ROS2 node as a Super Client under **Fast-DDS** framework. 

```bash
Results pending... To be updated once done.
```

## References
1. https://answers.ros.org/question/370595/ros2-foxy-nodes-cant-communicate-through-docker-container-border/?answer=371655#post-id-371655

ijnek, 2022-08-20:
I might be missing what you're trying to achieve, but removing ``--net host`` from the command in your intial post makes the communication work for me:

```
docker run -ti --rm --ipc host -v /dev/shm:/dev/shm --user root osrf/ros:foxy-desktop ros2 run demo_nodes_cpp talker
```

In fact, removing all the flags, the following works for me:

```
docker run -ti osrf/ros:foxy-desktop ros2 run demo_nodes_cpp talker
```

cardboardcode, 2022-08-20:
Hi @ijnek, noted. 

Will give it a try without the flags for `docker run command`. Will update below the results.

cardboardcode, 2022-08-20:
## Communication between Ubuntu 20.04 Container and Ubuntu 20.04 Host :heavy_check_mark: 

This is a minimal reproducible example to show **Container to Host** ROS2 interaction:

1. Start with a fresh install of **Ubuntu 20.04** and **ROS2 Foxy Fitzroy** in a **Virtual Machine Ubuntu 20.04**.
> I am using KVM. Here's a link to get started installing KVM on Ubuntu 20.04:
> https://www.tecmint.com/install-kvm-on-ubuntu/

2. Run **Talker** in **Container**.

```bash
# Open new terminal
docker run -ti osrf/ros:foxy-desktop ros2 run demo_nodes_cpp talker
```
3. Run **Listener** in **Host**.
```bash
# Open new terminal
source /opt/ros/foxy/setup.bash && \
ros2 run demo_nodes_cpp listener
```

## Verify

**Talker** in **Container**
```bash
rosi@rosi:~$ sudo docker run -ti osrf/ros:foxy-desktop ros2 run demo_nodes_cpp talker
[INFO] [1661008721.904435594] [talker]: Publishing: 'Hello World: 1'
[INFO] [1661008722.903522586] [talker]: Publishing: 'Hello World: 2'
[INFO] [1661008723.902581547] [talker]: Publishing: 'Hello World: 3'
[INFO] [1661008724.916828799] [talker]: Publishing: 'Hello World: 4'
[INFO] [1661008725.931438781] [talker]: Publishing: 'Hello World: 5'
[INFO] [1661008726.944363299] [talker]: Publishing: 'Hello World: 6'
[INFO] [1661008727.955637402] [talker]: Publishing: 'Hello World: 7'
[INFO] [1661008728.965586427] [talker]: Publishing: 'Hello World: 8'
[INFO] [1661008729.974221471] [talker]: Publishing: 'Hello World: 9'
[INFO] [1661008730.981918545] [talker]: Publishing: 'Hello World: 10'
[INFO] [1661008731.988699018] [talker]: Publishing: 'Hello World: 11'
[INFO] [1661008732.994716465] [talker]: Publishing: 'Hello World: 12'

```

**Listener** in **Host**
```bash
rosi@rosi:~$ source /opt/ros/foxy/setup.bash && \
> ros2 run demo_nodes_cpp listener
[INFO] [1661008730.983271364] [listener]: I heard: [Hello World: 10]
[INFO] [1661008731.989979161] [listener]: I heard: [Hello World: 11]
[INFO] [1661008732.995790257] [listener]: I heard: [Hello World: 12]
...
```

## Others :warning: 
However, for a persisting unknown reason, I am still unable to replicate in my current native **Ubuntu 20.04**.

Further effort will be dedicated to narrow down suspected environmental variables and/or dependency discrepancies.

@ijnek Thank you for the help. :smile:

iuhilnehc-ynos, 2022-08-22:
@cardboardcode 

https://github.com/ros2/ros2/issues/1318#issue-1345067845
Have you tried using the `root` user to run the listener application?

Barry-Xu-2018, 2022-08-22:
> Running --net host docker run command flag seems to force the Talker to use Shared Memory (SHM), instead of UDP. UDP is required for the desirable ROS2 interaction between Container and Host.m   

Used message is `std_msgs::msg::String` which is non-pod message. So SHM should be not used by Fast-DDS.

cardboardcode, 2022-08-22:
Hi @iuhilnehc-ynos, I don't think I have. Thanks for the suggestion.

Will probably try it out tomorrow (23rd Aug) and update the results here in this comment.

cardboardcode, 2022-08-23:
## Rough Workaround :adhesive_bandage: :confused: 
For now, if you are encountering the same issue as me on getting **Container to Host** ROS2 interactions to work, the short-term (and sadly destructive) solution is to: 

### Reinstall Native Host **Ubuntu 20.04** Operating System. 

## Verify :1st_place_medal: 
Once reinstalled, follow the instructions under  the minimal example **Communication between Ubuntu 20.04 Container and Ubuntu 20.04 Host** as shown above.

If you get similar results, the issues is resolved.

cardboardcode, 2022-08-23:
## Solution :partying_face: 
Ensure that **UiD namespace** is the same between **Ubuntu 20.04 Container** and **Ubuntu 20.04 Host**.

Closing.

closed 2022-08-23 by cardboardcode

ruffsl, 2022-09-20:
I admittedly skimmed this ticket when following the link from the TSC meeting minutes, but would the need for ensuring a shared UiD namespace between containers be ultimately due in part in avoiding duplicate or colliding GUIDs for DDS participants?

See relevant links:

- https://discourse.ros.org/t/ros-cross-distribution-communication/27335/9

- https://answers.ros.org/question/296828/ros2-connectivity-across-docker-containers-via-host-driver/?answer=298320#post-id-298320