ros2/ros2 #1322: Namespace is not reliably added to topics of a node

mrmara, 2022-09-13:
## Bug report

**Required Info:**

- Operating System:
    ```
    Distributor ID:	Ubuntu
    Description:	Ubuntu 20.04.4 LTS
    Release:	20.04
    Codename:	focal
    ```
- Installation type:
  - Binaries
- Version or commit hash:
  - Galactic
- DDS implementation:
  - Bug was reproduced on both fatrtps and cyclonedds
- Client library (if applicable):
  - rclcpp

#### Steps to reproduce issue
Add a namespace to a node through the launch file with the ```namespce``` attribute. For example:
```
ROS_NAMESPACE_ = os.environ['ROS_NAMESPACE_']
node = launch_ros.actions.Node(package=this_package,
                                executable=this_executable,
                                output='screen',
                                emulate_tty=True,
                                parameters=[specific_file,config_file],
                                name=name,
                                namespace=ROS_NAMESPACE_,
                                )
```

## Expected behavior
Topics, services, node name should be namespaced:

node_name/service_name --> namespace/node_name/service_name
topic_name --> namespace/topic_name
node_name --> namespace/node_name
## Actual behavior
Services and node name gets correclty namespaced, but topics are not namespaced.
## Additional information
This is something that does not always happen. In some cases everything is correctly namespaced, but in other cases only the services and node name are namespaced. I have not been able to find a pattern to reproduce this behavior. Anyway for a specific node and a specific launch file, the behavior is always the same.
Nodes are written in c++.

roncapat, 2022-12-09:
From [this design article](https://design.ros2.org/articles/topic_and_service_names.html):

> Relative names are appended to the namespace of the node which creates them. For example, if the node is put into a namespace /ping/pong and the topic or service name is foo/bar, then the absolute name will become /ping/pong/foo/bar. However, if the topic or service name is /foo/bar, then the absolute name will stay just /foo/bar, ignoring the node’s namespace.

So maybe what you experience is that some nodes specify topic names as relative and some absolute. Am I right? Otherwise tell me and I try to dig more into your problem to come up with a better explanation/solution.

fujitatomoya, 2022-12-09:
@mrmara as @roncapat mentioned, it would be really helpful that if you can share self-contained reproducible example. and this issue is related to https://github.com/ros2/rclcpp? if that so, probably we could move this issue to appropriate repository to investigate and address fix.

mrmara, 2022-12-13:
Thanks for the reply @roncapat and @fujitatomoya and sorry for the late reply. 
Yes, you are totally right. I discovered the concept of relative and absolute topic names months ago and forgot to close the issue.

Thank you

closed 2022-12-13 by mrmara