ros2/ros2 #1325: issue in installing humble RPM packages in RHEL 8.6

cwrx777, 2022-09-17:
## Bug report

**Required Info:**

- Operating System:
  - RHEL 8.6
- Installation type:
  - [RPM](https://docs.ros.org/en/humble/Installation/RHEL-Install-RPMs.html#install-ros-2-packages)

#### Steps to reproduce issue
```bash
sudo dnf install ros-humble-desktop
```

#### Expected behavior
Successful installation.

#### Actual behavior
```
Error:
 Problem: package ros-humble-desktop-0.10.0-1.el8.20220808.171503.x86_64 requires ros-humble-pcl-conversions, but none of the providers can be installed
  - conflicting requests
  - nothing provides pcl-devel needed by ros-humble-pcl-conversions-2.4.0-4.el8.20220729.172910.x86_64
```

closed 2022-10-06 by clalancette

cottsay, 2022-10-06:
All of the missing packages listed here are part of the EPEL repository.

Please ensure that epel and CRB/PowerTools are enabled on your system, as described int he installation instructions: https://docs.ros.org/en/humble/Installation/RHEL-Install-RPMs.html#setup-sources

cwrx777, 2022-10-07:
I was not able to install power tools by following this instruction:
```
$ sudo dnf install 'dnf-command(config-manager)' epel-release -y
No match for argument: epel-release
```

and i followed [this ](https://access.redhat.com/discussions/5417621#comment-1986291)to install codeready.

cwrx777, 2022-10-07:
update:
I managed to install epel-release by following [this](https://www.cyberciti.biz/faq/install-epel-repo-on-an-rhel-8-x/).
however, power-tools installation failed
```
$ sudo dnf config-manager --set-enabled powertools
Error: No matching repo to modify: powertools.
```

cottsay, 2022-10-07:
We're tracking an update to the RHEL installation instructions already with this PR: ros2/ros2_documentation#2954