ros2/ros2 #1330: High compile times for interface generation and template instantiation

SegfaultCreator, 2022-10-07:
Hello everyone,

we are currently changing our messaging architecture from an internally developed library to ros2.

For this, we had to move over 450 different messages. These messages are now interfaces saved in a single library built with ament/cmake. (We tried separating the interfaces into multiple libraries but we found out that using interfaces of another library in an interface was not possible while building without colcon through conflicts in ament_cmake).

We noticed that we have a very high compilation time for the interfaces library (2m 16s on eight threads) and the library responsible for creating publishers and subscribers of these interfaces (200/1000 ms for each template instantiation respectively) . In summary, our code now builds 9m 43s as compared to 3 min 35 s before the change. While we already performed some tricks to reduce the compile time (splitting up the instantiations of templates into separate compilation units and using precompiled headers), we still seek to find ways to reduce this compile-time onto reasonable levels.

Does anyone have an idea on how to achieve this? We already thought about:

    Improving the effectiveness of the generation process for interfaces. We already changed the idl_generator in order to only rebuild the interfaces that were changed instead of the whole interface library. However, our approach seemed to have some side effects which is why we abandoned this idea.
    Finding ways to reduce the time to instantiate a single template. 1.2 s for each subscriber/publisher pair sums up to an impressive amount of time for 450+ messages. Would a Generic Publisher help? 

Maybe some of you can provide us answers on the above ideas or have additional thoughts on how we can save compile-time in ros2.

Our Specs are: Ros2 - Galactic Ubuntu 20.04 Clang-11

Thanks in Advance

Segfault Creator

gavanderhoorn, 2022-10-09:
> We tried separating the interfaces into multiple libraries but we found out that using interfaces of another library in an interface was not possible while building without colcon through conflicts in ament_cmake

this seems like it's the real issue here?

SegfaultCreator, 2022-10-09:
First of all: Thanks for the fast reply!

With interfaces, i mean the [message (.msg)](https://docs.ros.org/en/galactic/Concepts/About-ROS-Interfaces.html) interfaces. 
So what we tried here is to separate the different messages in a similar way as the data is separated, which is represented by the interfaces. Let me provide you a short example:

=== Position.msg === (Represents class "Position" from Lib A)
float64 lat
float64 lon
float64 alt

=== Object.msg ===  (Represents class "Object" from Lib B)
Position pos

When we want to use the Position in the Object.msg, we were not able to separate the files into different libraries (cause of the mentioned problem with ament_cmake), which means that we now have one crowded messaging library. Using colcon seems to be no option, as compile times are even higher then. (And i guess we have to restructure our whole project cause colcon needs libraries to be on the same folder level?).

But besides this, the template instantiation also seems to be a big issue. Yesterday i measured the time to instantiate a simple subscriber for a std_msgs::msg::string (~ 530 ms). It seems that it doesnt scale well if i need to be able to instantiate a pub/sub template for each of my message types. One solution we thought of, was to provide only one subscriber template, i.e. that for std::string and serialize each message before using the publisher. But that looks like a dirty walkaround, because i think rqt and other tools are intended to see the actual message type.

clalancette, 2022-10-09:
> When we want to use the Position in the Object.msg, we were not able to separate the files into different libraries (cause of the mentioned problem with ament_cmake), which means that we now have [...]

None of this should be true.  The entirety of ROS 2 is built on composable message types, separated into separate packages.  See for instance https://github.com/ros-planning/navigation_msgs/blob/rolling/map_msgs/msg/OccupancyGridUpdate.msg , which depends on `std_msgs/Header` available in https://github.com/ros2/common_interfaces/blob/rolling/std_msgs/msg/Header.msg .

Also, I think you may have a fundamental misunderstanding of what colcon is.  colcon is strictly a tool that topologically orders packages based on dependencies (provided in the package.xml files), and then invokes the build system on each.  As such, it imposes no overhead for compilation, and doesn't have any opinions on how libraries are structured.

You may be referring to `ament_cmake`, which is indeed involved in setting up the compilation process.  But it also imposes no compilation overhead, and it also doesn't impose any restrictions on how libraries are structured.

What I think would help here is if you did a couple of different things:
1. Went back through the basic tutorials at https://docs.ros.org/en/rolling/Tutorials.html , and see how things are put together.  I think it would help to start with how we recommend things are done before charting off into unknown territory.
2. Provide a concrete example project where you are having problems.  It may be easier for us to see what your problem is then.

effakcuL, 2022-10-24:
Hello,

I am a different member of the same group as SegfaultCreator.

Thank you for your detailed answer, which took quite some time for us to digest.
You were spot on with pointing us to the tutorials, because it helped us figure out what we did wrong (I think, we forgot the packageName of the other package in our .msg files)

With that knowledge (and some time) we managed to get a basic example to build in both versions (with and without colcon). However, we noticed that already in our simple example the build times are vastly different between both versions. With colcon our build is at about 20 seconds (real    0m20.274s; user    0m37.133s; sys     0m9.536s) while the build without colcon takes 4 seconds for cmake (real    0m3.760s;user    0m2.515s;sys     0m1.231s) and 6 seconds for make -j9 (real    0m5.903s;user    0m24.186s;sys     0m5.508s ). I uploaded this simple example to [https://github.com/effakcuL/ros_colcon_issue](https://github.com/effakcuL/ros_colcon_issue) for you to verify (maybe we don't use colcon properly). It should be compilable with colcon build in the "withColcon" directory and with creating a build directory and in there running cmake ../interface_test .

Since our main goal was to reduce compile times for our project this is a huge issue. We could not check the compile time effect of splitting the message libraries yet, since I wanted to give you an update as soon as I got the basic example running.

Are there hidden things happening in the colcon build process, which we just don't realize that we need them yet?
We saw the possibility to adapt the colcon process through additional settings... is that our way to go?
Is there a way to streamline our cmake process without colcon, as it is currently pretty involved (although it should easily be manageable with some cmake functions) to get the dependencies setup properly?
Is there a reason for you to not use the target properties to let cmake handle the dependencies or is this a relic of old cmake dependencies?

Also a slightly off-topic question, but very much connected... is there a way to not generate all headers and sources for a library once a single .msg file changes (comparing folders after changing a single message indicates, that all generated files are completely independent of other messages)?

Thank you very much!

gavanderhoorn, 2022-10-24:
Just a quick note (as I see this more often): Colcon is never required for anything. It's just convenience.

Your `withColcon` example builds just as well like this without any Colcon:

```shell
source /opt/ros/$ROS_DISTRO/setup.bash

cd /path/to/withColcon/basic_interface
mkdir build && cd build
cmake ..
# running as root in a Docker container, so this installs into /usr/local
make install
# add to environment
source /usr/local/share/basic_interface/local_setup.bash

cd /path/to/withColcon/advanced_interface
mkdir build && cd build
cmake ..
# running as root in a Docker container, so this installs into /usr/local
make install
# add to environment
source /usr/local/share/advanced_interface/local_setup.bash

cd /path/to/withColcon/interface_test
mkdir build && cd build
cmake ..
# running as root in a Docker container, so this installs into /usr/local
make install
# add to environment
source /usr/local/share/interface_test/local_setup.bash
```

the above is basically what Colcon does (ie: ordering your build, updating environment, tracking progress).

effakcuL, 2022-10-26:
From looking at it, it seems the multiple CMake runs in there and therefore all the testing for the compiler and the other things CMake does initially make up most of the additional 10 seconds in the comparison. However, incremental builds also take a significant amount longer, at least in our full example (in the provided one the build times are hard to compare).

Is there a way (probably some ament calls) to provide the later targets with the proper include_directories of already existing targets? With the target created by rosidl_generate_interfaces being a custom target it is kind of hard to track all the created targets and forwarding them to the next set of targets (So I guess pretty much what ament does). We tried to get it to work with ament_target_dependencies, but this does not seem to forward additional properties such as include directories.

For us this issue becomes even worse, since we have another application on windows, where we have to use a Visual Studio Solution, that is created by a third party software, so there we cannot even use CMake and have to properly understand the connection between all the target dependencies.